raspberrypi/rpi-eeprom #189: The stable bootloader pieeprom-2020-07-16.bin does not boot 2020-05-27-raspios-buster-armhf from USB

bhlevca, 2020-07-23:
The stable bootloader pieeprom-2020-07-16.bin does not boot 2020-05-27-raspios-buster-armhf or the 64 bit version from USB.

I switched back to the pieeprom-2020-06-15.bin bootloader and booting from USB was possible again. 

I suspect that there is some bug in the newest bootloader that prevents booting from the USB device.

timg236, 2020-07-23:
Works for me

closed 2020-07-23 by timg236

bhlevca, 2020-07-23:
It may work for you, but it does not for me!  Have you tried all the RaspberryPi 4 versions 2, 4, and 8 GB?

I have one with 8 GB and the 2020-07-16 bootloader did not work, while 2020-06-15 did.  
It is possible that I made a mistake, but the same could have happened to you, or perhaps the version of Pi you tried happened to work. If you did not test with an 8 GB Pi I suggest to check.

I cannot reopen it as github has not the best tools for this and I will not open another one. I did my job to report and, I found a solution for me. 

I feel from this and from other reports that bug reports are not welcome.  I hope that I am wrong

cleverca22, 2020-07-24:
have you enabled both BOOT_UART and uart_2ndstage and checked the logs?

timg236, 2020-07-24:
The bug report template provides a detailed set of questions for good reason. 

Your report ignores all of that and provides no information about why something doesn’t work in your specific setup. 

For general support enquires please use the forums where you will find people have used this software with different hardware versions.

timg236, 2020-07-24:
I’ve blocked the blank issue template. If you follow the steps in the bug template and can provide logs as suggested by cleverca22 then please crest a new issue