raspberrypi/rpi-eeprom #194: Usb 2.0 vs 3.0

malinduta, 2020-08-06:
**Describe the bug**
Have 2xhdd conected to pi 4b.
When the hdds are connected to usb 2.0 everything is working as expected.
When connected to usb 3.0, things are different. On shutdown, the hdds remains ON. On reboot the hdds are not seen, missing from Openmediavault.
Booting from SD card.

**To Reproduce**
Swap usb 2.0-3.0 and reboot/shutdown.

**Expected behaviour**
Usb 3.0 as 2.0

**Bootloader version and configuration**
vcgencmd bootloader_config
```
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
USB_MSD_PWR_OFF_TIME=5000
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
```
vcgencmd bootloader_version
```
Jul 16 2020 16:15:46
version 45291ce619884192a6622bef8948fb5151c2b456 (release)
timestamp 1594912546
```
**SD card boot (please complete the following information):**
 - OS Raspbian 2020-05-27

timg236, 2020-08-06:
The bug report is missing full details of the USB device and the symptoms that you are reporting will be specific to that USB device.

Please can you also try the latest bootloader beta release

malinduta, 2020-08-06:
Yes, sorry!
Is this the needed information?
lsusb -vvv
```
Bus 002 Device 003: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         9
  idVendor           0x152d JMicron Technology Corp. / JMicron USA Technology Corp.
  idProduct          0x0578 JMS567 SATA 6Gb/s bridge
  bcdDevice           32.02
  iManufacturer           1 JMicron
  iProduct                2 External Disk 3.0
  iSerial                 3 74D786119534764
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0079
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              896mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           4
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
[... 461 lines cut ...]
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0507 highspeed power suspend enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

```

timg236, 2020-08-06:
Thanks, that's useful. A link to the retail (e.g. amazon) product is also useful because there are a lot of products which use the same USB SATA adaptor.

The latest beta bootloader changes the USB port power initialization which I think should resolve that. It would also be useful to know the board revision (cat /proc/cpuinfo | grep -i revision)

malinduta, 2020-08-06:
Here is the [adaptor](https://www.ebay.co.uk/itm/303422332518). 
Forgot to mention, it is external powered, like [here](https://i.stack.imgur.com/YJyzB.png)

Will do the rest tonight, when arrive home.

malinduta, 2020-08-06:
cat /proc/cpuinfo | grep -i revision
```
CPU revision    : 3
CPU revision    : 3
CPU revision    : 3
CPU revision    : 3
Revision        : c03112
```

```
vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
```
```
vcgencmd bootloader_version
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019
```
When using the last beta pieeprom-2020-07-31.bin, first reboot (not power on) works (hdd are seen on OMV), second reboot does not work (hdd are missing in OMV), and when rebooting for the third time, the pi gets stuck on "Reached target reboot".
The same pattern happend every time I tested (5-6 times.)

timg236, 2020-08-06:
I've just noticed "Booting from SD card" so not a USB MSD boot thing. The Linux USB stack is separate and PCI is actually reset before Linux starts.

However, you could try adding USB_MSD_PWR_OFF_TIME=5000 which will cause the bootloader to turn off the USB port power for long enough that it should look like a power off to the HDD. It might be similar to an issue seen with some wireless keyboards which don't like something about the hardware power reset sequencing.

malinduta, 2020-08-06:
Added USB_MSD_PWR_OFF_TIME=5000.
Same behaviour as above. Disks are seen only on power up and first reboot on usb 3.0.
On usb 2.0 disks are seen on every reboot.

timg236, 2020-08-06:
You could try zero but this is starting to look like a Linux bug unless this is a regression compared to the latest default bootloader (ie latest critical update). Do the drives appear or you hotplug the device after boot?

malinduta, 2020-08-06:
No, I have to unplug the power to make them appear again.
But why is working as expected on 2.0?

timg236, 2020-08-07:
USB2.0 and USB3.0 are independent buses over the same wires, it's possible that it's talking to almost completely separate silicon on the SATA adapter.

lurch, 2020-08-07:
Based on the powering-diagram that @malinduta posted earlier, it makes sense why modifying USB_MSD_PWR_OFF_TIME makes no difference (the HDDs are externally powered, so powering off the USB ports at the Pi-side has no affect).

ping @P33M in case he has any ideas about if/how this can be resolved from the Linux-side.

malinduta, 2020-08-07:
Ok, but if the adaptor is "always" on, why is not seen on reboot? And why, on last beta, is seen only on first reboot?
It could be back powered even if +5v wire is disconected? 
[image]

timg236, 2020-08-07:
The bootloader switches off the USB port power to avoid any glitches around reboot. However, from the diagram it looks as though the adapter always provides power. If hotplugging the USB cable from Linux without disconnecting the power cable doesn't work then there's nothing the bootloader can do about this.

lurch, 2020-08-07:
> Ok, but if the adaptor is "always" on, why is not seen on reboot?

Based on similar issues in this repo (see e.g. #151 #180 #168 ), it sounds like some HDD->USB adaptors have "problems" (this isn't my area of expertise!) if the Pi is rebooted without the power also being removed from the adaptor - I believe that's the entire reason for the `USB_MSD_PWR_OFF_TIME ` option being added.

timg236, 2020-08-07:
Yes the problem is that the separate power supply is not really managed by the usb device. That’s not something the usb host can influence. Closing but because if hot plugging the usb in Linux doesn’t work without manually toggling the power supply then there’s nothing we can do about this

closed 2020-08-07 by timg236