raspberrypi/rpi-eeprom #197: Stable path to vl805 firmware bin

colemickens, 2020-08-11:
I'm packaging this repo ior nixpkgs (I'm building towards an automated, nix-driven rpi4 netboot cluster thing).

I need some guidance, how can I reliably know the path to the stable vl805 firmware whenever a new release happens? It looks like these paths have versions in their filenames:

* https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/stable/vl805-000137ad.bin
* https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/stable/vl805-000138a1.bin

Is it possibly this file: https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/vl805?

If not, can such a known path be declared so packagers can reliably pull out the vl805 firmware binary?

Also, are signatures ever published as well? Even if they're just sha256sums?

Thanks!

colemickens, 2020-08-11:
Actually, this applies to the firmware binaries as well, since `recovery.bin` is a separate file.

timg236, 2020-08-12:
The hexadecimal version number defined by VLI is part of the filename. If you look at the source for the rpi-eeprom-update script then the version number comparisons are there and you could re-use that portion of the script.

This file is a deprecated user-space tool for updating the VL805, the rpi-eeprom-update script now uses lspci instead
 https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/vl805

As with the GPU firmware we don't publish sha256sums via a separate channel (i.e. it would have to be something different to Github)

N.B. The separate VLI firmware is not relevant for the 8GB it's embedded in the bootloader EEPROM. It's possible that we'll change the bootloader to update the VLI from the bootloaders copy to simplify the distribution but that's pretty low priority right now.

For automation, you can use the normal Github automation hooks to trigger a script in whatever CI system you use.

closed 2020-08-12 by timg236

colemickens, 2020-08-12:
Thanks for the detailed explanation, I was just now starting to put together that the scripts are indeed built to expect directories full of specifically-named firmware. I guess I'll keep it this way in the resulting package.

timg236, 2020-08-12:
No problem, btw you can just do clone the repository and run rpi-eeprom-update directly from there. It already does all the calculation of current/latest versions and has various hooks and only depends on minimal dash script features. You might find it easier to tweak rpi-eeprom-update an automated git clone than start from scratch