raspberrypi/rpi-eeprom #198: Latest stable isn't tagged in git

colemickens, 2020-08-11:
The latest stable release isn't tagged in git. This makes it harder for packagers to keep this up to date.

colemickens, 2020-08-11:
Latest git tag `v2020.07.16-138a1`

despite this commit: https://github.com/raspberrypi/rpi-eeprom/commit/0e7da7ecd84cdf7b65d6fc8d43446aee665187dc : "Promote beta 2020-07-31 to stable".

colemickens, 2020-08-12:
actually, I can't even tell which vl805 firmware is latest... unless I'm supposed to go by sort of filename.

timg236, 2020-08-12:
If you are writing custom packages then I recommend using the files in the repository directly instead of the Github binary releases.
https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/stable

The Github binary releases are only there for the Raspberry Pi Imager which only requires the critical updates. 

There's been a few manually created updates in order to make USB MSD testing easier but update for 2020-07-31 (later today) will probably be the last one because there's no need to have two separate distribution mechanisms.

closed 2020-08-12 by timg236