raspberrypi/rpi-eeprom #201: USB Harddisk not available when plugged in before powering on or Rebooting

NA1144, 2020-08-12:
For general boot questions please check the read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) sticky post on the forums.  

N.B The bootloader does not persist in memory and if the rainbow splash screen has been displayed the issue is likely to be in the firmware or Linux. If so, it's better to target the bug in the [Firmware](https://github.com/raspberrypi/firmware/) or [Linux](https://github.com/raspberrypi/linux/) repositories first e.g. NFS, USB or dmesg logs would be Linux issues.

**Describe the bug**
A clear and concise description of what the bug is.

1: Update RPI-EEPROM Using Raspberry Pi OS to the latest Stable FW Version from 07/31/2020
2: Switch Back to my Ubuntu 20.04 64Bit OS SD Card  
3: Plug in USB Harddisk in one of the USB 3.0 Ports
4: Power on RPI and wait for Boot
5: Try to Access Harddisk with cd /media/HDD1, Harddisk was not mounted (was not showed up in Sudo blkid or ls -l /dev/sd*)
6: I restarted and then the hard drive was still not recognised and mounted

My Workaround was to plug in the Harddisk after Powering up and Booting. I had to mount my Harddisk Manualy in Ubuntu

**To Reproduce**
Steps to reproduce the behavior:

1: Just Plug in your Harddrive in one of the USB 3.0 Ports and Power on wait for Boot. 
2: Try to access your Drive after Booting up. 

**Expected behaviour**
A clear and concise description of what you expected to happen.

The hard drive should have been recognized when plugged in before Powering on as in the last stable FW version from 07/16/2020.

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration (vcgencmd bootloader_config) and version (vcgencmd bootloader_version)

nothing modified
Bootloader: last stable FW version from 07/16/2020 
vcgencmd bootloader_version doesnt work in Ubuntu :-(

**SD card boot (please complete the following information):**
 - OS e.g. Raspbian
 - SD card type
 - Partition information (fdisk -l) if you are able to obtain this from another computer.

Ubuntu Server 20.04 64Bit for RPI4
SD Card SDHC Kingston 32GB

**Additional context**
Add any other context about the problem here.

timg236, 2020-08-12:
You haven't specified what sort of USB disk you are using and without that, this bug will be automatically closed.

You can try increasing USB_MSD_PWR_OFF_TIME to 5000 in boot eeprom config and please also try to reproduce in Raspberry Pi OS.

NA1144, 2020-08-12:
As you suggested, I tried it on raspberry pi os and I can't find my hard drive with the sudo blkid command: After a restart nothing changes
Below are the entries from sudo blkid
/ dev / mmcblk0p1: LABEL_FATBOOT = "boot" LABEL = "boot" UUID = "6E1A-D477" TYPE = "vfat" PARTUUID = "56cd6262-01"
/ dev / mmcblk0p2: LABEL = "rootfs" UUID = "087331ff-7f16-4bca-93e2-b07bd0469ef3" TYPE = "ext4" PARTUUID = "56cd6262-02"
/ dev / mmcblk0: PTUUID = "56cd6262" PTTYPE = "dos"

If I plug in the hard drive after the boot process as described above, then I can find my hard drive
Below are the entries from sudo blkid
/ dev / mmcblk0p1: LABEL_FATBOOT = "boot" LABEL = "boot" UUID = "6E1A-D477" TYPE = "vfat" PARTUUID = "56cd6262-01"
/ dev / mmcblk0p2: LABEL = "rootfs" UUID = "087331ff-7f16-4bca-93e2-b07bd0469ef3" TYPE = "ext4" PARTUUID = "56cd6262-02"
/ dev / mmcblk0: PTUUID = "56cd6262" PTTYPE = "dos"
/ dev / sda1: UUID = "1906-B367" TYPE = "vfat" PARTUUID = "151d16ee-bff3-4f5b-8b9b-dc94bc518e6a"
/ dev / sda2: UUID = "48460141-542b-43b7-b7d5-47184a6a37e8" TYPE = "ext4" PARTUUID = "60a1502e-f250-49b8-9a0d-57d0b08ecc59"

Increasing USB_MSD_PWR_OFF_TIME to 5000 in boot eeprom config hasn't worked out for me :-(
I m using the following USB Harddisk : ST750LM0 22 HN-M-750MBB 750GB

timg236, 2020-08-12:
The USB SATA adapter and power supply is the significant bit

How about USB_MSD_PWR_OFF_TIME=0?

NA1144, 2020-08-12:
i have tried to set USB_MSD_PWR_OFF_TIME=0 and again it isnt working :-(
The Last stable FW version from 07/16/2020 was working fine for me maybe i just should flash the old firmware ;-)
My Current Firmware:
vcgencmd bootloader_version
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019
My Current Bootloader Config after your Sugesstion:
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
USB_MSD_PWR_OFF_TIME=0

timg236, 2020-08-12:
That's a bit weird because the latest release only changes the MSD_PWR_OFF time

Which USB SATA adapter are you using?

NA1144, 2020-08-12:
not sure how to determine the exact Name of the USB SATA Adapter but it was in an Silver cnmemory Zinc Case
I have a Raspberry Pi 4 Model B Rev 1.1 and i m using the Original Power Supply (the white one)

timg236, 2020-08-12:
From Linux run lsusb -vvv and an Amazon link to the product is ideal

How is the hard drive powered?

NA1144, 2020-08-12:
ASM1051E SATA 6Gb/s bridge, ASM1053E SATA 6Gb/s bridge, ASM1153 SATA 3Gb/s bridge, ASM1153E SATA 6Gb/s bridge
https://www.amazon.de/Memory-Zinc-externe-Festplattengehäuse-silber/dp/B007OYUY6K
The Drive is Powered with the USB-Port no external Power Supply

timg236, 2020-08-12:
"The Drive is Powered with the USB-Port no external Power Supply" 

It might be worth trying a powered USB hub if you have one, generally only flash drives will work reliably without an external power supply. It may be close to the edge of working but fails if port power goes off then on again.

NA1144, 2020-08-12:
it was working with the old firmware and i had no issues. i have tried it with an ssd and it is working. I was running my Harddisk since March this year and i had no issues until now with the new FW. I will try some workarounds and maybe buy an external Hub when nothing else Works. Thank you for Helping

closed 2020-08-12 by NA1144

timg236, 2020-08-12:
btw: I tried a couple of 2.5" HDDs with latest bootloader and latest test Raspberry Pi OS (http://downloads.raspberrypi.org/nightlies/). Both were detected by the bootloader and both worked when hot-plugged from the desktop.

However, I encountered a few problems where the drive either vanished or boot got stuck just before loading the desktop. That lines up with the most power hungry part of boot so I think the voltage dipped because of the extra load of the HDD.

However, connecting the HDD via a USB hub in my monitor resolved the issue.

If a powered HUB doesn't resolve the issue but reverting back to the old bootloader is the only way to make it work then please re-open this issue

satcom886, 2020-10-07:
I'd open this again. I have this issue since I updated.

HDD works fine until I reboot the Pi. Then I can't find the drive in `/dev`.

I'm also using a 4GB Pi4 (I think 1.2), but the SATA -> USB thing has its own power source.
I thought my dock was dying so I bought a new one, but the problem persists.

I also downgraded both the bootloader and the USB firmware to a previous version. I started by only downgrading the USB firmware, but nothing changed. Only after I downgraded pieeprom, the problem was solved.

satcom886, 2020-10-07:
This is on Arch (btw) and I haven't modified any EEPROM config values.

NA1144, 2020-10-08:
Hi
I still have the problem. A Bootloader downgrade worked for me too.
Please try to change the configuration values mentioned above.  Maybe it works for you 
Let me know before I reopen the issue or maybe you can reopen it.

satcom886, 2020-10-13:
I'm sorry, I thought I would be able to test this, but my Pi 4 with the USB thing is used as a server and I can't really afford being offline again. I hope you understand.