raspberrypi/rpi-eeprom #214: `pieeprom-2020-07-31.bin` does not boot from the USB SSD

ilg-ul, 2020-09-02:
**Describe the bug**

`pieeprom-2020-07-31.bin` does not boot from the USB SSD (RPi OS 64) created with `pieeprom-2020-06-15.bin`.

**To Reproduce**

On a Raspberry Pi 4 4 GB
- program `pieeprom-2020-06-15.bin` on a Raspberry Pi 4 4 GB
- install RPi OS 64 on an USB SSD

Raspberry Pi 4 8 GB
- boot RPi OS 32 from SD Card
- program `pieeprom-2020-07-31.bin`
- connect USB SSD from 4 GB Raspberry
- reboot

**Expected behaviour**

- the 8 GB system to boot

**Screenshots**

- no photo available, but the bootloader reported that no SD Card is available; there were no messages related to USB, as if it was not there; booting from SD showed the /dev/sd to be there, so the hardware is ok.

**Bootloader version and configuration**

```console
$ sudo vcgencmd bootloader_version
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019
```

**SD card boot:**
 - OS: Raspberry Pi OS 64
 - SD card type: SAMSUNG EVO Plus 64
 - USB SSD: 250 GB Sandisk Portable USB 3.1

**Network boot:**
- the Ethernet connection is ok

**USB boot:**

The same system, in the same configuration, with the same original power supply, boots from USB with pieeprom-2020-06-15.bin; the power should be fine.

**Additional context**

Since the beginning of the year I had a RPi 4, 4G, now running RPi SO 64 from USB SSD, and booting with `pieeprom-2020-06-15.bin`.

Today I got a new RPI 4, 8G. I started it with micro SD running RPi OS 32. I updated/upgraded it and got:

```console
$ ls -l /lib/firmware/raspberrypi/bootloader/stable/
total 2344
-rw-r--r-- 1 root root 524288 Apr 23 18:53 pieeprom-2020-04-16.bin
-rw-r--r-- 1 root root 524288 Jun 17 12:15 pieeprom-2020-06-15.bin
-rw-r--r-- 1 root root 524288 Jul 20 14:17 pieeprom-2020-07-16.bin
-rw-r--r-- 1 root root 524288 Aug 10 16:17 pieeprom-2020-07-31.bin
-rw-r--r-- 1 root root  98196 Aug 10 16:17 recovery.bin
-rw-r--r-- 1 root root  98904 Feb 28  2020 vl805-000137ad.bin
-rw-r--r-- 1 root root  99224 Jul 20 14:17 vl805-000138a1.bin

$ cat /etc/default/rpi-eeprom-update
FIRMWARE_RELEASE_STATUS="stable"

$ sudo rpi-eeprom-update -a
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTFS /boot
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: update available
CURRENT: Thu 16 Apr 2020 05:11:26 PM UTC (1587057086)
 LATEST: Fri 31 Jul 2020 01:43:39 PM UTC (1596203019)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000137ad
 LATEST: 000137ad
BOOTFS /boot
EEPROM updates pending. Please reboot to apply the update.

$ sudo reboot

$ sudo vcgencmd bootloader_version
Jul 31 2020 14:43:39
version f8d1dc69144b10b4bd78b4b6d08658ac3005726d (release)
timestamp 1596203019
```

After removing the micro SD and moving the SSD from the RPi 4G to the new RPi 8G, the system did not boot.

I booted from the micro SD again and reverted to `pieeprom-2020-06-15.bin`:

```
$ sudo rpi-eeprom-update -d -f /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-06-15.bin
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTFS /boot
*** INSTALLING /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-06-15.bin  ***
BOOTFS /boot
EEPROM update pending. Please reboot to apply the update.

$ sudo reboot

$ sudo vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
```

With this older eeprom, the same system was able to boot from the same USB SSD.

It is perfectly possible that I'm doing something stupid, but it is also possible that the `pieeprom-2020-07-31.bin` may have some issues with booting from USB SSD.

lurch, 2020-09-02:
It might be that the firmware files (`start.elf` etc.) on the SSD are new enough to support the Pi4 4G, but too old to support the Pi4 8G? Doing a `sudo apt update; sudo apt full-upgrade` when booted from the SSD (on the 4GB Pi4?) should get them up to date.

timg236, 2020-09-02:
Either bootloader logs or a screenshot showing USB MSD state is would be required to debug this

ilg-ul, 2020-09-02:
You were right, update & full-upgrade, followed by `sudo rpi-eeprom-update -d -f /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-07-31.bin` solved the problem, now the boot page displays July 31, the same USB SSD is seen and the system boots.

I also double checked, I copied `2020-05-27-raspios-buster-arm64.zip` on an empty USB stick, and the system was able to boot.

The problem was solved, but I still do not understand why after the first attempt to use `pieeprom-2020-07-31.bin`, the bootloader (which certainly displayed July 31) did not even mention USB, as it did not exist, and asked only for the SD Card.

> bootloader logs or a screenshot showing USB MSD state is would be required to debug this

ok, if I'll encounter such cases in the future I'll take screenshots.

timg236, 2020-09-02:
Best guess is that the BOOT_ORDER was changed to something other than USB, but if it happens again feel free to re-open with a screenshot.
Glad to hear it's working for you now !

closed 2020-09-02 by timg236

ilg-ul, 2020-09-02:
> Best guess is that the BOOT_ORDER was changed to something other than USB

The first time when I issued `rpi-eeprom-update` I did not use `-d`. I don't know exactly what this means, but could it be that the second time it worked because I used `-d`?

Does `-d` change something in the eeprom or in the boot partition?

timg236, 2020-09-02:
It was probably the -d flag. From rpi-eeprom-update
   -d Use the default bootloader config, or if a file is specified using the -f
      flag use the config in that file. This option only applies when a
      bootloader EEPROM update is needed; if the bootloader EEPROM is up-to-date
      then its config will not be changed.

Where "default" is the one that's in the pieeprom.bin supplied in the APT package. Without this, the current EEPROM config is preserved.

ilg-ul, 2020-09-02:
Perhaps the documentation describing how to perform such updates needs some clarifications.