raspberrypi/rpi-eeprom #215: Missing dependency on bsdmainutils due to use of hexdump

MichaIng, 2020-09-02:
This commit introduced the use of the `hexdump` command: https://github.com/raspberrypi/rpi-eeprom/commit/eea80bc4ab45821319265662c23cf4f33180eda4#diff-86eaef688a7bf372514e072950593ed7
This is part of the bsdmainutils package: https://packages.debian.org/buster/amd64/bsdmainutils/filelist
The package hence needs to be added to the dependency list, or `hexdump` needs to be replaced with a command from an "essential" system package.
I can commit the change but could not find the source of the DEB package meta files 😉.

timg236, 2020-09-02:
@MichaIng The package source is on the debian/buster branch on this repo
https://github.com/raspberrypi/rpi-eeprom/tree/debian/buster/debian

Or you could add get it to use the Raspberry Pi specific /proc/cpuinfo file if it exists first

MichaIng, 2020-09-02:
`/proc/cpuinfo` was used before the linked commit. Reading the commit text, it has been changed to be compatible with mainline kernel which does not report the RPi model correctly in `/proc/cpuinfo`:
> Upstream kernels do not list the board revision in /proc/cpuinfo. Get it
from the device tree instead.

MichaIng, 2020-09-02:
PR opened: https://github.com/raspberrypi/rpi-eeprom/pull/216

If there is a better way to get the RPi model without an additional dependency, that would be great of course, but until then the PR will do. I added the removal of obsolete transitional dummy package `rpi-eeprom-image` as dependency, which has been integrated into `rpi-eeprom` with v7.2 already. This hence only causes a dependency loop 😄.

lurch, 2020-09-02:
> If there is a better way to get the RPi model without an additional dependency

I've just been googling for how to read binary files as hexadecimal (in bash), and many search results (e.g. [this one](https://stackoverflow.com/questions/6292645) ) suggest using one of `hexdump`, `od` or `xxd`, so probably makes sense to stick with hexdump?

timg236, 2020-09-03:
The tool must be available as part of busybox

lurch, 2020-09-03:
`hexdump` and `od` are both included in the list at https://busybox.net/BusyBox.html

timg236, 2020-09-03:
I think it's reasonable to add a dependency on hexdump

MichaIng, 2020-09-03:
`xxd` is a tiny utility, so much less "overhead" then `bsdmainutils`: https://packages.debian.org/buster/xxd
On the other hand, `bsdmainutils` are installed on most systems anyway, it is on Raspberry Pi OS Lite, so from that point of view I'd prefer `bsdmainutils`.

But `od` can be used without adding any further dependency since it's part of the "essential" `coreutils` package: https://packages.debian.org/coreutils
Essential packages are generally expected, so never added to package dependencies and cannot be removed without a big warning + interactive input. If the syntax is not much more complicated then `hexdump`, I'd go with `od`.

timg236, 2020-09-03:
I think the original change to move away from /proc/cpuinfo only benefitted a small number of users, I'm not going to spend time re-testing this since it's super low priority so currently I'm tempted to revert the whole lot.

MichaIng, 2020-09-03:
To not cause a regression to users which now started to rely on function with mainline kernel, I think adding the dependency for now is trivial enough? Also the number of users affected by the additional `bsdmainutils` is minimal (okay, many of our DietPi users 😢 😉). And when someone is bored, or I am in mood, testing `od` can be done at a later date. All very low priority, I agree.

timg236, 2020-09-03:
The support for upstream broke something for all older Raspberry Pis so the correct thing to do in retrospect was just to revert until the correct fix is available. 
If the package definitions can be sorted out today it can go in, otherwise, upstream users will have to wait a bit longer

timg236, 2020-09-03:
switched to od

MichaIng, 2020-09-03:
Nice, I just found exactly the same and was only annoyed by the fact that one cannot suppress the spaces in the first place. `tr` is from coreutils as well so that fits well together. I'll update my PR to only remove `rpi-eeprom-images` from dependencies.

timg236, 2020-09-03:
Thanks for investigating this. I did have to read the man-pages od is quite ugly, but it's worth it to keep this script as the plumbing for any UIs / OS services

MichaIng, 2020-09-03:
> I did have to read the man-pages od is quite ugly

Dito 😄. It doesn't seem to be made for scripting, also due to the default offset print. But since we ended up with the same `od` options independently, that should be fine now.

lurch, 2020-09-03:
I guess that means this issue can be closed? :slightly_smiling_face:

closed 2020-09-03 by MichaIng