raspberrypi/rpi-eeprom #220: rpi-eeprom-update don't flash

vslinuxdotnet, 2020-09-04:
Hello,

I was unable to flash the eprom again, the  rpi-eeprom-update  says that flash it ok, but then after a reboot all remains the same.
This is RPI 4 b model with 2G ram.

```
root@raspberrypi:/home/pi# rpi-eeprom-update -a -d
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /boot
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: update available
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Fri Jul 31 13:43:39 UTC 2020 (1596203019)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: update available
CURRENT: 000137ad
 LATEST: 000138a1
BOOTFS /boot
EEPROM updates pending. Please reboot to apply the update.
```
When reboot completes says the same:

```
root@raspberrypi:/home/pi# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
CURRENT: Thu Apr 16 17:11:26 UTC 2020 (1587057086)
 LATEST: Fri Jul 31 13:43:39 UTC 2020 (1596203019)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: update available
CURRENT: 000137ad
```
I detect it because I need to change the boot order, but the changes don't apply.
There is a read only mode on eeprom ?
Any other ideas ?

Thank you

lurch, 2020-09-04:
Are you booting your Pi from micro-SD, USB or network? Which OS-version are you using? Are you using standalone Raspberry Pi OS or NOOBS-installed Raspberry Pi OS?

After the first step (and before the reboot), what does `ls -l /boot/` say?

vslinuxdotnet, 2020-09-04:
> Are you booting your Pi from micro-SD, USB or network? Which OS-version are you using? Are you using standalone Raspberry Pi OS or NOOBS-installed Raspberry Pi OS?
> 
> After the first step (and before the reboot), what does `ls -l /boot/` say?

I use last Raspbian with all updates. But I found the problem/solution.
I remove the USB pen drive In USB3 port, then flash it again, then reboot and the eeprom was updated!
That is my strange report.

```
root@raspberrypi:/home/pi# rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Fri Jul 31 13:43:39 UTC 2020 (1596203019)
 LATEST: Fri Jul 31 13:43:39 UTC 2020 (1596203019)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

timg236, 2020-09-04:
Sounds like a stale pieeprom update file on one of the devices.
If you have an SD card which consistently fails then please post that

vslinuxdotnet, 2020-09-04:
> Sounds like a stale pieeprom update file on one of the devices.
> If you have an SD card which consistently fails then please post that

No, SD card is 32G class 10 from SanDisk and is new.
No problems on logs.

lurch, 2020-09-04:
> I remove the USB pen drive In USB3 port, then flash it again, then reboot and the eeprom was updated!

I wonder if the bootloader was trying to boot / update from the USB drive in preference to the SD card? :shrug:

timg236, 2020-09-04:
It looks for the files on the current boot mode. Only one file system is loaded at a time

closed 2020-09-06 by timg236

bartmol, 2020-10-26:
I have the exact same problem, however the fix is not working for me.

pi@raspberrypi:~ $ sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
CURRENT: Mon 15 Jul 12:59:55 UTC 2019 (1563195595)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: update available
CURRENT: 00013701
 LATEST: 000138a1

After I update the Eeprom and reboot, I get the same message again. I tried unplugging the USB hard drive, but this didn't work.

robmorin, 2020-12-01:
I have same issue, I am using RPI 4GB with Ubuntu 20.04

root@3072-1:~# /usr/local/bin/rpi-eeprom/rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
CURRENT: Tue Sep 10 10:41:50 UTC 2019 (1568112110)
 LATEST: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /usr/local/bin/rpi-eeprom/firmware/critical
VL805: update available
CURRENT: 000137ad
 LATEST: 000138a1
After reboot, still has old firmware..

ls -la /boot:
-rw-------  1 root root  4164733 Oct  5 05:16 System.map-5.4.0-1021-raspi
-rw-------  1 root root  4164641 Oct 15 09:16 System.map-5.4.0-1022-raspi
-rw-r--r--  1 root root   218809 Oct  5 05:16 config-5.4.0-1021-raspi
-rw-r--r--  1 root root   218826 Oct 15 09:16 config-5.4.0-1022-raspi
lrwxrwxrwx  1 root root       43 Dec  1 11:01 dtb -> dtbs/5.4.0-1022-raspi/./bcm2711-rpi-4-b.dtb
lrwxrwxrwx  1 root root       43 Nov 11 13:59 dtb-5.4.0-1021-raspi -> dtbs/5.4.0-1021-raspi/./bcm2711-rpi-4-b.dtb
lrwxrwxrwx  1 root root       43 Dec  1 11:01 dtb-5.4.0-1022-raspi -> dtbs/5.4.0-1022-raspi/./bcm2711-rpi-4-b.dtb
drwxr-xr-x  7 root root     4096 Oct 21 07:01 dtbs
drwxr-xr-x  4 root root     4096 Dec 31  1969 firmware
drwxr-xr-x  2 root root     4096 Aug  6 18:49 grub
lrwxrwxrwx  1 root root       27 Oct 21 07:00 initrd.img -> initrd.img-5.4.0-1022-raspi
-rw-r--r--  1 root root 34104163 Nov 11 13:59 initrd.img-5.4.0-1021-raspi
-rw-r--r--  1 root root 34137163 Dec  1 11:01 initrd.img-5.4.0-1022-raspi
lrwxrwxrwx  1 root root       27 Oct 21 07:00 initrd.img.old -> initrd.img-5.4.0-1021-raspi
-rw-r--r--  1 root root       80 Dec  1 15:00 pieeprom.sig
-rw-r--r--  1 root root   524288 Dec  1 15:00 pieeprom.upd
-rw-r--r--  1 root root    98196 Dec  1 15:00 recovery.bin
-rw-r--r--  1 root root    99224 Dec  1 15:00 vl805.bin
-rw-r--r--  1 root root       65 Dec  1 15:00 vl805.sig
lrwxrwxrwx  1 root root       24 Oct 21 07:00 vmlinuz -> vmlinuz-5.4.0-1022-raspi
-rw-------  1 root root  8306127 Oct  5 05:16 vmlinuz-5.4.0-1021-raspi
-rw-------  1 root root  8341855 Oct 15 09:16 vmlinuz-5.4.0-1022-raspi
lrwxrwxrwx  1 root root       24 Oct 21 07:00 vmlinuz.old -> vmlinuz-5.4.0-1021-raspi

lsblk:
NAME        MAJ:MIN RM  SIZE RO TYPE  MOUNTPOINT
loop0         7:0    0 48.8M  1 loop  /snap/core18/1936
loop1         7:1    0 48.8M  1 loop  /snap/core18/1888
loop2         7:2    0   63M  1 loop  /snap/lxd/16945
loop3         7:3    0 85.3M  1 loop  /snap/rpi-imager/130
loop4         7:4    0 59.9M  1 loop  /snap/lxd/18152
loop5         7:5    0 26.9M  1 loop  /snap/snapd/9730
loop6         7:6    0   27M  1 loop  /snap/snapd/10243
loop7         7:7    0   16K  1 loop  /snap/software-boutique/56
sda           8:0    0  3.7T  0 disk  
└─sda1        8:1    0  3.7T  0 part  
  └─md0       9:0    0  3.7T  0 raid1 /mnt/md0
sdb           8:16   0  3.7T  0 disk  
└─sdb1        8:17   0  3.7T  0 part  
  └─md0       9:0    0  3.7T  0 raid1 /mnt/md0
mmcblk0     179:0    0 29.7G  0 disk  
├─mmcblk0p1 179:1    0  256M  0 part  /boot/firmware
└─mmcblk0p2 179:2    0 29.5G  0 part  /