raspberrypi/rpi-eeprom #228: BOOTFS mountpoint detection

stek29, 2020-09-21:
**Describe the bug**
BOOTFS isn't detected correctly if it's mounted to folder other than /boot

**To Reproduce**
Try to run rpi-eeprom-update on Ubuntu Server
The script installs update files to /boot, but mmcblk0p1 is mounted at /boot/firmware, so nothing happens on reboot

**Expected behaviour**
The script detects that /dev/mmcblk0p1 is mounted at /boot/firmware and sets BOOTFS variable accordingly

timg236, 2020-09-21:
rpi-eeprom-update is only supported on Raspberry Pi OS so this isn't something that I'll spend time on. However, if someone wants to submit a PR for Ubuntu support I'll merge it provided that it doesn't cause regressions.

waveform80, 2020-09-21:
I'm currently in the process of attempting to get rpi-eeprom into Ubuntu Server (with a local patch for the varied boot mount-point). There's a version in proposed for Groovy currently but it's out of date, and will probably be rejected in favour of an updated one I'm prepping to deal with MIR comments. Relevant links below:

https://launchpad.net/~waveform/+archive/ubuntu/eeprom
https://launchpad.net/ubuntu/+source/rpi-eeprom
https://bugs.launchpad.net/ubuntu/+source/rpi-eeprom/+bug/1895137

timg236, 2020-09-21:
@waveform80 thanks for the update!

stek29, 2020-09-23:
@timg236 would a PR that detects BOOTFS mountpoint in some generic way falling back to current value be acceptable?

timg236, 2020-09-25:
Yes, that would be acceptable. Create PR referencing this bug, it needs to work on Raspberry Pi OS, Ubuntu, SUSE, Kodi, Debian (at the last count!). I normally manage to break something for one of the distros :)

btw: I'm about to add 'rpi-eeprom-update -b' which will return the resolved BOOTFS so that the new interactive --edit option in rpi-eeprom-config can determine if there is a pending pieeprom.upd 
https://github.com/raspberrypi/rpi-eeprom/pull/229.
So long as findBootFS() can work out the expected path on the given OS this should all just work.

satmandu, 2020-09-30:
This is what I'm doing in an update script I run, which works on both raspi-os & ubuntu:

```
bootfsmount=$(mount | grep boot | awk '{print $3}')
[[ $(echo $bootfsmount | wc -w) == "1" ]] && BOOTFS=${bootfsmount}
BOOTFS=${BOOTFS:-/boot}
```

(ubuntu tends to have the ```/dev/mmcblkp01``` mounted at ```/boot/firmware```, and this picks it up.)

seamusdemora, 2020-10-01:
@timg236 said:

>rpi-eeprom-update is only supported on Raspberry Pi OS so this isn't something that I'll spend time on. However, if someone wants to submit a PR for Ubuntu support I'll merge it provided that it doesn't cause regressions. 

Just so I'm clear on this - does this mean that other (non RPi OS) distros are not supported - or supported only coincidentally?

timg236, 2020-10-01:
@seamusdemora As with the firmware, we don't have the resource to test multiple distros so it's only tested on Raspberry Pi OS. However, the implementation was designed so that the core features (*) only depend upon common base packages (e.g. busybox utilities and python2 or python3) so that it should "just work" on most distros or be trivially portable.

I'm happy to accept PRs to fix things for other distros (e.g. using od instead of hexdump) so long as it works on Raspberry Pi OS and is unlikely to cause regressions for other distros. I trust others to shout if it does break in which case the PR will need to be reverted/reworked because I don't want compatibility to get worse.

(*) The original and core purpose of the tool is to provide an automatic EEPROM update mechanism for critical bug or hardware compatibility fixes for users who don't know or care about bootloaders. It also provides the basic plumbing for anyone who wants to add a UIs for advanced users e.g. the machine parseable output and script hooks.

seamusdemora, 2020-10-01:
@timg236 : Thanks - that helps clarify. 

One other question if I might: Is the RPi's USB driver included in the firmware - or is that an open-source bit of code?

timg236, 2020-11-06:
@seamusdemora It's closed source. U-boot was updated to support USB boot on the Raspberry Pi 4B revision 1.4. I've not looked at it but I expect it would be a useful reference if you want something which is open-source but smaller than the Linux Kernel driver.

Ruffio, 2021-03-23:
stek29 would you provide a PR and/or can this issue be closed?

timg236, 2021-10-11:
Closed due to inactivity

closed 2021-10-11 by timg236