raspberrypi/rpi-eeprom #230: rpi4 AArch64 Arch Linux rpi-eeprom-update failure

FancyFane, 2020-09-26:
**Describe the bug**
When running the rpi-eeprom-update command it results in a failure on a fresh install of Arch Linux ARM 64 bit running on Raspberry Pi 4. 

**To Reproduce**
Install Arch Linux AArch64 version, selected support/troubleshooting packages

`# pacman -Syu --needed nfs-utils mkinitcpio-nfs-utils vim lsof tcpdump gnu-netcat sysstat bash-completion ethtool iputils man-pages man-db strace sudo rsync inetutils docker git mlocate ebtables socat cni-plugins python3 binutils fakeroot gcc pkgconf make autoconf automake lzo go go-tools python-docutils patch bison bc`

Install the yay utility here; then install the two packages needed for rpi-eeprom-update from the AUR

`$ yay -S rpi-eeprom raspberrypi-userland-aarch64-git`

Once this completed run the rpi-eeprom-update command to get the error below:

```
[root@alarm ~]# rpi-eeprom-update 
od: /sys/firmware/devicetree/base/system/linux,revision: No such file or directory
```

**Expected behaviour**
When working this should provide output to show if bootloader is up to date:

```
[root@alarm ~]# rpi-eeprom-update 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/stable
VL805: up-to-date
CURRENT: 00000000
 LATEST: 00000000
```

**Bootloader version and configuration**
```
[root@alarm ~]# vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
```

**SD card boot (please complete the following information):**
```
[root@alarm ~]# uname -r
5.8.9-2-ARCH

[root@alarm ~]# cat /etc/os-release 
NAME="Arch Linux ARM"
PRETTY_NAME="Arch Linux ARM"
ID=archarm
ID_LIKE=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinuxarm.org/"
DOCUMENTATION_URL="https://archlinuxarm.org/wiki"
SUPPORT_URL="https://archlinuxarm.org/forum"
BUG_REPORT_URL="https://github.com/archlinuxarm/PKGBUILDs/issues"
LOGO=archlinux

[root@alarm ~]# cat /proc/device-tree/model      
Raspberry Pi 4 Model B
```

**Additional context**
Issue #4 provided context before it was closed out. Someone mentioned using vcgencmd to get this information I've changed to this method and was able to get the script to work. 

```
[root@alarm ~]# /opt/vc/bin/vcgencmd otp_dump | grep -Po '(?<=30:).{8}'
00d03114
```

This pulls line 30 (Revision) from output and provides the following 8 characters which is the same output that od was providing. I've tested the above command on both 32bit and 64bit versions successfully. More information on the otp_dump command can be found here: [vcgencmd documentation](https://elinux.org/RPI_vcgencmd_usage)

This error is occurring because the code is currently pulling the Reversion number from a file which isn't populated on this version of linux. 

```
[root@alarm ~]# ls -lah /sys/firmware/devicetree/base/system/linux,revision
ls: cannot access '/sys/firmware/devicetree/base/system/linux,revision': No such file or directory
```

timg236, 2020-09-27:
I think you need to investigate why the device tree entry is missing on Aarch64.

FancyFane, 2020-09-27:
@timg236 I've read over the previous issues. The issue is that this flavor of linux does not provide the revision information from /proc/cpuinfo either. That would of made things easier. 

```
[root@alarm ~]# cat /proc/cpuinfo 
processor	: 0
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
BogoMIPS	: 108.00
Features	: fp asimd evtstrm crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3
```

Could we consider maybe doing a test to see if the firmware file is in the device-tree? If it is, we can pull the info using the od method, and if it fails we could fall back on getting the information from otp_dump. 

https://github.com/FancyFane/rpi-eeprom/blob/05cf34f7b4157ee047512bb700d1ab399d0d5aec/rpi-eeprom-update#L271-L275

Admittedly, I'm more from the linux admin world than a dev role, so if I'm doing pull request or coding things wrong any feedback would be greatly appreciated.

timg236, 2020-09-28:
Is it possible that you are missing /boot/bcm2711-rpi-pi4.dtb ?

I think checking device-tree, then /proc/cpuinfo then the raw OTP should be safe.  I can't remember if busybox grep supports the -Po flags which is why the original version used cut

timg236, 2020-09-28:
I resurrected the old code and made it try DT, cpuinfo and OTP in order. If you can verify that it works for you then that would be great. The missing device-tree entries are a cause for future concern, I think more scripts will assume that firmware generated device tree nodes will be visible

timg236, 2020-09-29:
@FancyFane  Please could you try the latest version of rpi-eeprom-update script on the master branch?

FancyFane, 2020-10-01:
@timg236 thank you for all the updates, I apologize for not getting back to you sooner, yes the new version works perfectly. This issue is safe to close out with this code in place. 

```[root@alarm ~]# /opt/github/rpi-eeprom/rpi-eeprom-update 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /opt/github/rpi-eeprom/firmware/critical
VL805: up-to-date
CURRENT: 00000000
 LATEST: 00000000
```

I've also followed up on the file you were referring to in the boot partition and it appears to be intact, I'll provide the md5sum just in case that's relevant and needs to be compared to a different file (I realize that may be out of scope for this code).

```[root@alarm ~]# ls -lah /boot/bcm2711-rpi-4-b.dtb 
-rwxr-xr-x 1 root root 41K Aug  8 21:00 /boot/bcm2711-rpi-4-b.dtb

[root@alarm ~]# md5sum /boot/bcm2711-rpi-4-b.dtb
04f2dd06c65cd7ad2932041cbe220a13  /boot/bcm2711-rpi-4-b.dtb
```

You make a very valid point about grep perl regex maybe being too complicated for busybox. I wanted to challenge myself to come up with a line that grabs the same information using just awk just for fun. 

`vcgencmd otp_dump | awk '/^30:/{print $2}; {FS=":"};'`

Thank you again for all your quick responses, explanations and time.

lurch, 2020-10-01:
@timg236 The
```
VL805: up-to-date
CURRENT: 00000000
 LATEST: 00000000
```
in the output above seems strange? :shrug:

timg236, 2020-10-02:
The setup sounds a bit non-standard and if the kernel is missing the mailbox call to load the VLI firmware then I'd expect to see that output.

timg236, 2020-10-02:
Closing bug because the script change described in the bug report was resolved. If there's a different problem then that should be a new bug report

closed 2020-10-02 by timg236

lurch, 2020-10-02:
Ahhh, I'd missed the "VL805 firmware in bootloader EEPROM" part of the output - apologies for causing any confusion!

fryfrog, 2020-10-30:
Could you guys do a new release that contains this fix?