raspberrypi/rpi-eeprom #233: Error using the edit feature of rpi-eeoprom-config

locutusofsand, 2020-10-05:
For general boot questions please check the read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) sticky post on the forums.  

N.B The bootloader does not persist in memory and if the rainbow splash screen has been displayed the issue is likely to be in the firmware or Linux. If so, it's better to target the bug in the [Firmware](https://github.com/raspberrypi/firmware/) or [Linux](https://github.com/raspberrypi/linux/) repositories first e.g. NFS, USB or dmesg logs would be Linux issues.

**Describe the bug**
A clear and concise description of what the bug is.
**The edit feature of the rpi-eeprom-config function fails on the current critical release firmware.**

**To Reproduce**
Steps to reproduce the behavior:
**sudo -E rpi-eeprom-config --edit**

**Expected behaviour**
A clear and concise description of what you expected to happen.
**A text editor to spawn so that configuration options could be changed or added.**

**Screenshots**
If applicable, add a photograph of the bootloader HDMI diagnostics screen.
```
pi@user:~ $ sudo -E rpi-eeprom-config --edit
Traceback (most recent call last):
  File "/usr/bin/rpi-eeprom-config", line 347, in <module>
    main()
  File "/usr/bin/rpi-eeprom-config", line 325, in main
    edit_config(args.eeprom)
  File "/usr/bin/rpi-eeprom-config", line 130, in edit_config
    pending = os.path.join(bootfs, 'pieeprom.upd')
  File "/usr/lib/python3.7/posixpath.py", line 94, in join
    genericpath._check_arg_types('join', a, *p)
  File "/usr/lib/python3.7/genericpath.py", line 151, in _check_arg_types
    raise TypeError("Can't mix strings and bytes in path components") from None
TypeError: Can't mix strings and bytes in path components
pi@user:~ $ sudo rpi-eeprom-update
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

```

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration (vcgencmd bootloader_config) and version (vcgencmd bootloader_version)

**SD card boot (please complete the following information):**
 - OS e.g. **Raspberry Pi OS**
 - SD card type **32GB Sandisk Extreme**
 - Partition information (fdisk -l) if you are able to obtain this from another computer. **I dont think this really applies but I can provide if helpful.**

**Additional context**
Add any other context about the problem here.
**I have tested this on a test setup and a production system both fully updated and got the same results.**

timg236, 2020-10-06:
This should be fixed now

locutusofsand, 2020-10-06:
I can confirm that this is fixed in 10.1-1 . Thanks for the prompt response!

closed 2020-10-06 by locutusofsand