raspberrypi/rpi-eeprom #235: Pi 4 does not boot reliably

Nastras, 2020-10-14:
Hello, I am booting my Pi 4 from a USB 3.0 stick. I often get this error after a restart and the Pi does not start. Only when i disconnect the pi from power, remove the usb-stick and wait for approx. 20 seconds, it will boot again and start the OS.

[image]

Do you have an idea, why?

Thanks!

Nastras, 2020-10-14:
One more note. The error had not always been there, it has only recently appeared, maybe since the last update, but I can't say exactly.

I still have two Pi 4 with the same USB stick model in operation where the error does not occur.

timg236, 2020-10-14:
Please answer the questions in the bug template especially the USB MSD section

Nastras, 2020-10-14:
**Describe the bug**
Hello, I am booting my Pi 4 from a USB 3.0 stick. I often get this error after a restart and the Pi does not start. Only when i disconnect the pi from power, remove the usb-stick and wait for approx. 20 seconds, it will boot again and start the OS.

**To Reproduce**
Reboot the pi

**Expected behaviour**
The Pi should start the OS and not get stuck at boot.

**Screenshots**
See above post 1.

**Bootloader version and configuration**
nastras@SHS:~ $ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

nastras@SHS:~ $ vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103

**USB boot (please complete the following information):**
The start of sd card is possible, i have used the pi 4 with the same power supply without problems until the official stable usb boot was released. I also use two pi 4 with the same power supply and the same usb sticks as boot media where this error does not occur.

nastras@SHS:~ $ lsusb -vvv
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
bLength                18
bDescriptorType         1
bcdUSB               2.00
bDeviceClass            9 Hub
bDeviceSubClass         0 
bDeviceProtocol         1 Single TT
bMaxPacketSize0        64
idVendor           0x1d6b Linux Foundation
idProduct          0x0002 2.0 root hub
bcdDevice            5.04
iManufacturer           3 
iProduct                2 
iSerial                 1 
bNumConfigurations      1
Configuration Descriptor:
bLength                 9
bDescriptorType         2
wTotalLength       0x0019
bNumInterfaces          1
bConfigurationValue     1
iConfiguration          0 
bmAttributes         0xe0
Self Powered
Remote Wakeup
MaxPower                0mA
Interface Descriptor:
bLength                 9
bDescriptorType         4
bInterfaceNumber        0
bAlternateSetting       0
bNumEndpoints           1
bInterfaceClass         9 Hub
bInterfaceSubClass      0 
bInterfaceProtocol      0 Full speed (or root) hub
iInterface              0 
Endpoint Descriptor:
bLength                 7
bDescriptorType         5
bEndpointAddress     0x81  EP 1 IN
bmAttributes            3
Transfer Type            Interrupt
Synch Type               None
Usage Type               Data
wMaxPacketSize     0x0004  1x 4 bytes
bInterval              12

**Additional context**
I hope the information is sufficient and helps you to find the bug. I thank you in advance for your support!

Best Regards
Nastras

timg236, 2020-10-14:
Thanks, please can you also update the bug and title with details of the manufacturer and product. Lots of people are using the latest bootloader with a variety of different USB flash / USB SATA devices so it help to be specific.

You might want to investigate increasing USB_MSD_PWR_OFF_TIME in the EEPROM config. When the board is reset there is an unavoidable interruption to the USB port power. Some devices may want a much longer delay. Unfortunately, the screenshot seems to show it finding the device but it hasn't either responded or timed out yet

To change the config
sudo apt update
sudo apt upgrade
sudo -E rpi-eeprom-config --edit

Nastras, 2020-10-14:
I have adapted the title :-)
Then I opened the EEPROM config. But here I can't find a value for USB_MSD_PWR_OFF_TIME? Or should I supplement this if so with what value?

nastras@SHS:~ $ sudo -E rpi-eeprom-config --edit
Updating bootloader EEPROM
 image: /lib/firmware/raspberrypi/bootloader/stable/pieeprom-2020-09-03.bin
config_src: vcgencmd bootloader_config
config: /tmp/tmpqtodn2i2/boot.conf
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

BCM2711 detected
Dedicated VL805 EEPROM detected
*** INSTALLING /tmp/tmpqtodn2i2/pieeprom.upd  ***
BOOTFS /boot
EEPROM update pending. Please reboot to apply the update.

timg236, 2020-10-14:
You need to add it yourself. The default behaviour will be 1000 (1 second)
USB_MSD_PWR_OFF_TIME=1000

So you can try zero to make it as small as possible or up to 5000

lurch, 2020-10-14:
See https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md for more details about the config options.

Nastras, 2020-10-14:
Thank you both!

I have now increased the value to 3000 for testing. Unfortunately the Pi won't start anymore. I tried all usb ports and disconnected it from the power for a longer time. It now always stays in the boot screen.

I'm just about to import a backup. I have used another identical usb-stick and will try if there is a defect on the usb-stick. Is that possible?

[image]

Nastras, 2020-10-15:
After I changed the USB stick the error seems to disappear. Therefore I assume that it is a hardware defect.
If it happens again I will open the thread again.

Thanks for your support!

closed 2020-10-15 by Nastras

timg236, 2020-10-15:
@Nastras was the replacement USB stick the same make/model?

Nastras, 2020-10-15:
Yes exactly the same.