raspberrypi/rpi-eeprom #236: Reproducible Failed MSD Boot

claford-v-lawrence, 2020-10-19:
For general boot questions please check the read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?t=58151) sticky post on the forums.  

N.B The bootloader does not persist in memory and if the rainbow splash screen has been displayed the issue is likely to be in the firmware or Linux. If so, it's better to target the bug in the [Firmware](https://github.com/raspberrypi/firmware/) or [Linux](https://github.com/raspberrypi/linux/) repositories first e.g. NFS, USB or dmesg logs would be Linux issues.

**Describe the bug**
My raspberry Pi 4 would boot from USB if I just shut it down from SD card. It would fail to boot, however, if it was previously booted from USB

**To Reproduce**
Boot rpi4 from SD card, shut it down, boot with USB with Samsung 970 EVO and an Orico USB 3.2 nvme enclosure. It would boot successfully. However, if I shut it down from that point forth, no reboot is successful.

**Expected behaviour**
Rpi4 should boot successfully regardless of where it booted from.

**Screenshots**
If applicable, add a photograph of the bootloader HDMI diagnostics screen.

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration (vcgencmd bootloader_config) and version (vcgencmd bootloader_version)

**SD card boot (please complete the following information):**
 - OS : Raspbian
 - SD card type:  A1
 - Partition information (fdisk -l) if you are able to obtain this from another computer.

**Network boot (please complete the following information):**
didn't use

**USB boot (please complete the following information):**
can boot from USB if previously booted from SD card, otherwise cannot boot from USB

**Additional context**
I also had some success when I connected and disconnected RPI to power very quickly.

timg236, 2020-10-19:
It sounds like the enclosure does not handle USB disconnect properly.  During a reset, the USB port will be disconnected, you can make the disconnect longer by increasing USB_MSD_PWR_OFF_TIME see -
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md 

Setting it to zero might work although what you actually see is a very brief disconnect which some USB devices won't tolerate

timg236, 2020-10-19:
Please can you post the board-revision of your Pi
cat /proc/cpuinfo | grep Revision

claford-v-lawrence, 2020-10-21:
Apologies for replying so late. I didn't check my email in the last 2 days.

claford-v-lawrence, 2020-10-21:
What's even more funny was that after the experiment last time, the rpi4 was unable to boot. It booted from USB just now without me having to insert an SD card first.

claford-v-lawrence, 2020-10-21:
Board revision is c03112

claford-v-lawrence, 2020-10-21:
I followed your advice and changed the USB_MSD_PWR_OFF_TIME to 5000, and I have been rebooting it for 10 minutes, and I didn't see a single failed reboot! Thanks for your outstanding work!

timg236, 2020-10-21:
Thanks for retesting and sharing the information. I'm doing some interop testing in the background and it's useful to gather data which can help us improve default parameters or make it easier to tweak settings for any devices which fall outside of what can be supported in the default config.

claford-v-lawrence, 2020-10-21:
No problem:) Please let me know if there's something more I could do to help!

lurch, 2020-10-21:
@claford-v-lawrence If you have time to test, it might be worth trying to reduce `USB_MSD_PWR_OFF_TIME` to smaller values, to identify where the always-fails / sometimes-fails / always-works boundaries are?

claford-v-lawrence, 2020-10-21:
Well, this is a bit awkward. I just rebooted it again and it failed to boot this time. I'm reopening this issue for further investigation. If we can find the cause for that, I'll definitely test where are the boundaries.

claford-v-lawrence, 2020-10-21:
Now it's still stuck at `Read start4.elf bytes 2278432`

claford-v-lawrence, 2020-10-21:
I would also like to add that when it did boot, it showed SD card found, despite the fact that there is no SD card inserted. The only thing plugged in is my SSD enclosure.

timg236, 2020-10-21:
The SD card detect is muxed with WL_ON so it might return the wrong value, it's only used as a UI indicator in the bootloader

claford-v-lawrence, 2020-10-22:
Thanks. Are there extra ways for me to provide you with more information that would help debugging?

lurch, 2020-10-22:
See `XHCI_DEBUG` and `NETCONSOLE` options in https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

claford-v-lawrence, 2020-10-23:
Well, this is funny. I just spent the past half hour trying to reproduce the issue and failed. I'll update as soon as I have a failed boot.

lurch, 2020-10-23:
It sounds like something very similar is going on in #241

timg236, 2020-10-23:
Please can you try the experimental bootloader release here
https://github.com/raspberrypi/rpi-eeprom/issues/241#issuecomment-715416176

claford-v-lawrence, 2020-10-25:
Sorry, just finished my weekend. Will do first thing tomorrow.

claford-v-lawrence, 2020-10-26:
I just did multiple tests. Cold reboot and soft reboot both works. Will reboot after a few hours to observe further results:P I could see that the boot time just reduced drastically. Thanks for your hard work!

claford-v-lawrence, 2020-10-26:
Subsequent reboots show everything works fine. Good job! Thank you very much:P

closed 2020-10-26 by claford-v-lawrence