raspberrypi/rpi-eeprom #237: rpi-eeprom-update fails on non-RPi-OS systems due to missing dpkg md5sum file

HiassofT, 2020-10-20:
**Describe the bug**
Since commit 29fe479af9c3fad2ac4c1b1d65aedf88b6f97626 rpi-eeprom-update fails to install updates on LibreELEC because `/var/lib/dpkg/info/rpi-eeprom.md5sums` is missing.

**To Reproduce**
Run eg `rpi-eeprom-update -a` on LibreELEC or a system without the dpkg md5sum file.

commit 29fe479af9c3fad2ac4c1b1d65aedf88b6f97626 gives this output:
```
LibreELEC:~ # rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTFS /flash
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: up-to-date
CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /usr/lib/kernel-overlays/base/lib/firmware/raspberrypi/bootloader/critical
VL805: update available
CURRENT: 000137ad
 LATEST: 000138a1
cat: can't open '/var/lib/dpkg/info/rpi-eeprom.md5sums': No such file or directory
```
and commit 718820b this slightly different one
```
LibreELEC:~ # rpi-eeprom-update -a
BCM2711 detected
Dedicated VL805 EEPROM detected
*** INSTALLING EEPROM UPDATES ***
BOOTLOADER: up-to-date
CURRENT: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu Sep  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /usr/lib/kernel-overlays/base/lib/firmware/raspberrypi/bootloader/critical
VL805: update available
CURRENT: 000137ad
 LATEST: 000138a1
grep: /var/lib/dpkg/info//rpi-eeprom.md5sums: No such file or directory
cat: can't open '/var/lib/dpkg/info//rpi-eeprom-images.md5sums': No such file or directory
md5sum: /tmp/tmp.H1SybC: no checksum lines found
rpi-eeprom checksums failed - try reinstalling this package
Unable to validate EEPROM image package checksums
```

**Expected behaviour**
It would be nice if rpi-eeprom-update could be fixed for non-RPi-OS distributions, eg by restoring the previous behaviour (skip md5sum check if  `/var/lib/dpkg/info//rpi-eeprom.md5sums` doesn't exist) or by retrieving a setting from `/etc/default/rpi-eeprom-update` to skip the ckeg that we could simply set in LibreELEC.

OFC I'm open to other suggestions, too.

**Bootloader version and configuration**
This is the `/etc/default/rpi-eeprom-update` we use in LibreELEC
```
# Use direct path to firmware as update script doesn't dereference sym links.
FIRMWARE_ROOT="/usr/lib/kernel-overlays/base/lib/firmware/raspberrypi/bootloader"
FIRMWARE_BACKUP_DIR="/storage/.config/rpifw-backup"
BOOTFS=${BOOTFS:-/flash}
```

timg236, 2020-10-20:
Does /var/lib/dpkg/info exist? It could set IGNORE_DPKG_CHECKSUMS if it doesn't exist

HiassofT, 2020-10-20:
No, we don't have `/var/lib/dpkg` at all

HiassofT, 2020-10-20:
thanks for the quick fix, current master is fine now!

closed 2020-10-20 by HiassofT

lurch, 2020-10-20:
(Fixed by #238 )