raspberrypi/rpi-eeprom #242: Boot issue with X835: boot when connected on USB2 but not on USB3

RamaSpaceShip, 2020-10-23:
**Describe the bug**
I have a P4 4GB with RaspiOS Desktop on it. USB storage is via a X835 board and a Sandisk SSD.
I can boot with /boot on the SD card and / on the SSD, but full USB boot on USB3 port doesn't work. :(
I tried using a USB2 port and it worked???

I setup netconsole to investigate and it boots on USB3 when netconsole is active.

**To reproduce**
Connect a X835 with an SSD on a USB3 port of the Pi4 and try to boot on it.

**Expected behaviour**
The system boots normally

**Actual behaviour**
It does not boot and wait for a bootable device

If netconsole is set, it boots but the network is broken

**System**
Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage4

Linux raspberrypi 5.4.51-v7l+ raspberrypi/firmware#1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Revision : c03112
Serial : 100000006baa8fa6
Model : Raspberry Pi 4 Model B Rev 1.2
Throttled flag : throttled=0x0
Camera : supported=0 detected=0
Videocore information

Sep 2 2020 21:13:13
Copyright (c) 2012 Broadcom
version 4439d2aaa6c376a2d1ef4402f142e1cf4de37c43 (clean) (release) (start)

alloc failures: 0
compactions: 0
legacy block fails: 0
Filesystem information

Sys. de fichiers blocs de 1K Utilisé Disponible Uti% Monté sur
/dev/root 30832068 6855768 22387076 24% /
devtmpfs 1827296 0 1827296 0% /dev
tmpfs 1959392 5120 1954272 1% /dev/shm
tmpfs 1959392 952 1958440 1% /run
tmpfs 5120 4 5116 1% /run/lock
tmpfs 1959392 0 1959392 0% /sys/fs/cgroup
/dev/sda1 294736 56032 238704 20% /boot
/dev/sda4 924971004 145904 877769408 1% /home
tmpfs 391876 8 391868 1% /run/user/1001
tmpfs 391876 0 391876 0% /run/user/1000

Filename Type Size Used Priority
/dev/sda3 partition 4194236 0 -2
/var/swap file 102396 0 -3
Package version information

raspberrypi-ui-mods:
Installé : 1.20200611
raspberrypi-sys-mods:
Installé : 20200812
openbox:
Installé : 3.6.1-8+rpt4
lxpanel:
Installé : 0.10.0-2+rpt12
[... 251 lines cut ...]
USB3 rport 5 status 00000000 -> 000002a0
USB2 rport 1 status 00000000 -> 400002e1
USB2 rport 1 status 400002f1 -> 40200e03
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3 rport 3 status 000002b1 -> 00281203
DEV [02:00] 3.32 000000:03 class 9 VID 05e3 PID 0620
HUB init [02:00] 3.32 000000:03
DEV [03:01] 2.00 000001:01 class 0 VID 046d PID c016
DEV [04:02] 3.00 000001:03 class 0 VID 174c PID 0829
MSD device [04:02] 3.00 000001:03 conf 0 iface 0 ep 81#1024 02#1024
PM_RSTS: 0x00001000
MBR: 0x0000ffff,  589815 type: 0x0c
MBR: 0x0009fff6,62913600 type: 0x83
MBR: 0x03c9fc36, 8388480 type: 0x82
MBR: 0x0449fbb6,1881572426 type: 0x83
PM_RSTS: 0x00001000
DEV [05:01] 2.00 000003:01 class 0 VID 046d PID c318
DEV [06:01] 2.16 000002:01 class 9 VID 05e3 PID 0610
HUB init [06:01] 2.16 000002:01
DEV [07:02] 3.00 000002:03 class 0 VID 174c PID 0829
MSD device [07:02] 3.00 000002:03 conf 0 iface 0 ep 81#1024 02#1024

**Additional context**
none

timg236, 2020-10-23:
Please can you try the updated bootloader here
https://github.com/raspberrypi/rpi-eeprom/issues/241#issuecomment-715416176

I've been testing this with an X835 and it seems to work ok with both cold boots and reboots

RamaSpaceShip, 2020-10-24:
Unfortunately the Pi final owner has taken it and I can no longer do any test with it.

Sorry

timg236, 2020-11-06:
No problem, closing bug as fixed in pieeprom-2020-10-28 since I have an X835 to test with.

closed 2020-11-06 by timg236