raspberrypi/rpi-eeprom #243: Partition type 0x83 halts boot process

dasJ, 2020-10-24:
**Describe the bug**

When encountering a boot partition with type 0x83, the boot fails even when not booting from SD card.

**To Reproduce**
Steps to reproduce the behavior:

- Configure the bootloader with `BOOT_ORDER=0x12`
- Boot without SD card, verify it booting from network
- Boot without network but with SD where the boot partition has type 0xc, verify it booting from SD card
- Change the type of the partition to 0x83, see it hanging after power on - no output on UART with `BOOT_UART=1`

**Expected behaviour**

The SD card should be skipped

**Screenshots**

No output - no screenshot

**Bootloader version and configuration**
Firmware 2020-09-03

timg236, 2020-10-25:
Please can you try the attached fix? If it doesn't work please post the full output fdisk /dev/mmcblk0. I suspect it was failing because it didn't find a FAT partition before the Linux one.
[rpi-eeprom-recovery-partid-0x83.zip](https://github.com/raspberrypi/rpi-eeprom/files/5434666/rpi-eeprom-recovery-partid-0x83.zip)

cleverca22, 2020-11-01:
i suspect it may be a ROM bug

```
Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x245a585c

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       2048 15523839 15521792  7.4G 83 Linux
[root@system76:~]# mkfs.ext4 /dev/mmcblk0p1
```
a freshly formated all ext4 SD card, results in zero output on the uart, despite having BOOT_UART=1 set
if i remove the SD card and re-apply power, i instantly get the expected eeprom output:
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0xd328cdce 0x00c03112
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
```

timg236, 2020-11-01:
It was resolved in the latest beta

closed 2020-11-01 by timg236

tomty89, 2020-11-28:
@timg236 It is not fixed at all:
```
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Wed Oct 28 05:32:40 PM UTC 2020 (1603906360)
 LATEST: Wed Oct 28 05:32:40 PM UTC 2020 (1603906360)
 FW DIR: /home/tom/rpi-eeprom/firmware/beta
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

It just keeps trying "partition 0".

I don't know how the fix was attempted / implemented, but it should not just skip  the 0x83 one(s), instead it should use a whitelist approach (0xc, 0xb...). Either way not even 0x83 is not skipped.

For the record the diagnostic page does not show when it's stuck at the SD card. It shows only when it's stuck at a USB drive (that is connected to the VL805), otherwise the screen just stays black.

timg236, 2020-11-28:
Please create new bug reports instead of adding comments to closed issues unless this is definitely the problem reported by the OP.

The ROM issue with exFAT is a known issue in B0 - don't make the first partition exFAT.

The 2711 bootloader does not scan for the first FAT boot partition. It will use partition 0 or the number specified via the reboot command. It might be enhanced to do that but I think some start.elf changes would be required to cover some corner cases.

tomty89, 2020-11-28:
@timg236 So let's make this clear. The OP is, if the first partition of the SD card is of type 0x83 (and what about others? or even `0` a.k.a. non-existing?), however it is formatted / whether it is formatted or not, then the SD card should be skipped (and a USB drive will be booted), right? I think I couldn't get that to work either (I can try a couple more times).

Or does the filesystem of the first partition matters as well? Since you mention "exFAT", which doesn't exactly make sense unless you are referring to partition type (code) 0x7 (which is shared with NTFS). Or is this B0 another boot stage that you are talking about? Which cares only about filesystem but not partition type (code) (or both?)? Can you point me to somewhere that the details of this issue are recorded?

pelwell, 2020-11-28:
If @timg236 thinks you have a different problem then you probably do. Start a new issue, stating the behaviour you are seeing and how it differs from what you expect, and let us worry about the why.

tomty89, 2020-11-28:
@pelwell *My* issue started with the idea / need to have the first partition as exfat-formatted (*and* of type 0x7, for obvious reason). If it is indeed a *known* issue then I don't exactly have anything to file about, unless I should file one *as long as I still want to know the details about it*.

But during the process / control tests I couldn't seem to reproduce the effect of the fix of *this issue* either, and I also find the (message about) the fix ambiguous. While the OP might only want to get 0x83 working/skipped, that doesn't mean it should be how the issue is fixed. So I want to know more details about the fix of *this issue* as well. (If I should file yet another one for that too, will do.)

timg236, 2020-11-30:
"P.S. It would be great if it will try all the primary partitions (and skip accordingly)."

It doesn't do that, it has never done that. It's reasonable and probably desirable behaviour but will probably require changes to firmware and Pi3 or older bootcode.bin and NOOBS to interoperate correctly.

As noted in the original post for historical reasons 0x83 was in the list of understood partitions which caused a failure specific to partition 0x83.