raspberrypi/rpi-eeprom #251: blink error codes have an extra short blink at the start now

cleverca22, 2020-11-07:
**Describe the bug**
all error blinks from the eeprom firmware now have an extra short blink at the start, making it harder to get the real code

**To Reproduce**
boot an rpi4 with `VERSION:694e118a DATE: Oct 28 2020 TIME: 17:32:43` firmware, but `start4.elf` missing

**Expected behaviour**
error 4 should give 4 blinks, not 5

**Screenshots**
a scope probing one leg of the ACT led while its giving a code 0x04 for `start4.elf` not found, not that it has 5 blinks, and the 1st is very short
[image]

**Bootloader version and configuration**
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:694e118a DATE: Oct 28 2020 TIME: 17:32:43 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1603906360 0xd328cdce 0x00c03112 0x00099c5c
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 16Gb x2 total-size: 32 Gbit 3200
...
Boot mode: USB-DEV (03) order 0
...
start.elf not found (6)
Firmware not found
ERROR: 00000004
Boot mode: SD (00) order 0
Insert SD-CARD
```
i intentionally booted without a start4.elf to trigger the blink error

**Additional context**
Add any other context about the problem here.

timg236, 2020-11-07:
It looks as though there is some SPI activity

timg236, 2020-11-07:
Sorry, wrong Github button!

It looks as though there was some SPI activity and the driver reset the GPIOs back to the default INPUT state. 
For fatal errors, this will be less of a problem because the pattern is repeated and there's no chance of SPI activity.

I think it should be possible to improve this for the one-shot error shown above

cleverca22, 2020-11-24:
[image]
```
RPi: BOOTLOADER release VERSION:6fdd3d2b DATE: Nov 24 2020 TIME: 15:08:08 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1606230484 0xd328cdce 0x00c03112 0x000a4b19
...
GENET STOP: 0
NETBOOT CANCEL
NETBOOT init failed
Boot mode: SD (00) order 0
Insert SD-CARD
```
https://github.com/raspberrypi/rpi-eeprom/commit/3702e706aff73a0b0ad545223f9aa4b7d98aeb5c
the release notes say its fixed, but a quick check on my pi4 says it isnt

the bcm xhci boot also looks like a nice addition, i have been wondering about that, since it allows much better code reuse

timg236, 2020-12-11:
Thanks for the retest. The LED patterns have been slightly reworked in the latest beta 80fd3016d0563580edd1ec6f862c208259a84845 to have a gap before the error pattern and change the default state to off after the first error has occurred. This should make it easier to separate error patterns from any activity caused by SPI EEPROM access.

cleverca22, 2020-12-19:
major improvement, enough to say its fixed

the 5th blink is still there, but its so short that you have to be staring at it in a dark room and know what to expect

probing with my scope, i can see that the 5th blink is actually 2 bursts of 8 pulses, likely SPI traffic as it queries the SPI flash?
but the whole blink is only 2.5uS long, so i wouldnt worry about it

closed 2020-12-19 by cleverca22

timg236, 2020-12-19:
Thanks again for the retest.  It's possible that SPI traffic is slightly different on Pi 400. If it's not too distracting from the error pattern then it might be left as is, at least until all the recent major changes are can be released as a new default update.

cleverca22, 2020-12-20:
yeah, its plausible the pi400 is having a longer pulse, but its harder to pop a scope onto that pin, and its brief enough that id just accept it as-is