raspberrypi/rpi-eeprom #253: No Bootable Partitions

STUKguy, 2020-11-11:
Hi, 

So I thought I would make a back up of my SD card with Win32, once done I plugged my SD card back into my PI4 and this happened 
[image]
I didn't hit delete/overright on Win32 so unsure why this has happened I'm burning the image I did to an SD card to see what happens. 
But just incase does this mean I need to redo my Raspbian image?

timg236, 2020-11-11:
Please see the Raspberry Pi General discussion forums for support questions
https://www.raspberrypi.org/forums/viewforum.php?f=63

That looks like a blank ex-FAT SD card so presumably, the disk copying process went wrong

closed 2020-11-11 by timg236

STUKguy, 2020-11-11:
@timg236 ah sorry did realize I posted to the wrong form. Thank you for your help