raspberrypi/rpi-eeprom #255: USB-DEV mode reports incorrect configuration descriptor size

cleverca22, 2020-11-15:
**Describe the bug**
`[916706.286385] usb 6-2: config index 0 descriptor too short (expected 55, got 32)`
when the usb-dev code is first reporting the configuration descriptor, it claims a total length of 55 bytes (2 interfaces, 4 endpoints)
but when you then try read those 55 bytes, it changes its mind, and claims the configuration descriptor is 32 bytes
you can see both the number of interfaces and the total-length changing in wireshark+usbmon

checking the disassembly, the cause is also clear, the descriptors get `memcpy`'d into a reply packet, then the original copy of the descriptor is patched
so the patch only takes effect the 2nd time the usb host reads the descriptor

**To Reproduce**
start a CM4 or rpi4 in usb-dev mode, and just look at the usb traffic when it enumerates

**Expected behaviour**
it should report the same configuration descriptor each time its read

**Bootloader version and configuration**
```
RPi: BOOTLOADER release VERSION:694e118a DATE: Oct 28 2020 TIME: 17:32:43 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1603906360 0xd328cdce 0x00c03112 0x00099c5c
Boot mode: USB-DEV (03) order 0
```
`bootconf.txt`:
```
[all]
BOOT_UART=1
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=192.168.2.15
TFTP_PREFIX=0
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5

gpio=5=ip,pu

[gpio5=0]
BOOT_ORDER=0x241

[gpio5=1]
BOOT_ORDER=0x3

[none]
FREEZE_VERSION=1
```

timg236, 2020-12-11:
The USB descriptors have been tidied up in the latest beta 80fd3016d0563580edd1ec6f862c208259a84845

N.B. There's a couple of minor issues in the BCM2711 ROM descriptors but these all seem mostly harmless.

cleverca22, 2020-12-19:
```
[730459.420256] usb 6-2: USB disconnect, device number 18
[730464.081011] usb 6-2: new high-speed USB device number 19 using ehci-pci
[730464.212110] usb 6-2: New USB device found, idVendor=0a5c, idProduct=2711, bcdDevice= 0.00
[730464.212113] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[730464.212114] usb 6-2: Product: BCM2711 Boot
[730464.212115] usb 6-2: Manufacturer: Broadcom
[730464.212116] usb 6-2: SerialNumber: daaa69e8
```
i can confirm that the descriptor sizes look fixed on pieeprom-2020-12-11.bin
the SerialNumber has also been fixed, it now reports the correct hex-string, rather then Broadcom, which solves a problem i was having in my rpiboot implementation!!

cleverca22, 2020-12-19:
i also notice some other minor changes
originally, the serialnumber was "Broadcom" (same string# as iManufacturer), and it had 2 bulk-out endpoints

but now the serialnumber is correct, and it has a bulk-out on 1, and a bulk-in on 2

cleverca22, 2020-12-19:
on a closer inspection, i think the serial# change by itself is bad...

https://github.com/raspberrypi/usbboot is now broken:
https://github.com/raspberrypi/usbboot/blob/master/main.c#L739-L741

a BOT rom returns:
```
  iManufacturer           1 Broadcom
  iProduct                2 BCM2711 Boot
  iSerial                 3 d328cdce
```
pieeprom-2020-11-24 returns:
```
  iManufacturer           1 Broadcom
  iProduct                2 BCM2711 Boot
  iSerial                 1 Broadcom
```
and now the new pieeprom-2020-12-11 returns:
```
  iManufacturer           1 Broadcom
  iProduct                2 BCM2711 Boot
  iSerial                 3 d328cdce
```

there is no way to tell the difference between a rom wanting bootcode4.bin, and a bootcode.bin wanting the fileserver
a backwards compatible way to fix it is to just move the serial# to string descriptor 4, and leave 3 unused
a more correct fix is to maybe use bInterfaceProtocol to advertise which mode your in

cleverca22, 2020-12-19:
dangit, you did exactly that, and i was just flashing the wrong .bin!

```
  iManufacturer           1 Broadcom
  iProduct                2 BCM2711 Boot
  iSerial                 4 d328cdce
```
no idea how it got stuck on the november bin, but your already using the backwards compatible fix

cleverca22, 2020-12-19:
yeah, usbboot works perfectly, i just missed a pieeprom file, flashed the wrong one, and then things went funky, all problems on my end

closed 2020-12-19 by cleverca22

cleverca22, 2020-12-19:
the only complication is for people not using libusb

webusb is being too helpful, and only gives you the string behind the serial#, and never the iSerial, so webusb cant tell the difference on the surface
you must open the device, and send a control-in to fetch the device descriptor, and parse it yourself

bit of a pain, but the code is written now, so no big deal

timg236, 2020-12-19:
@cleverca22 thanks for retesting this