raspberrypi/rpi-eeprom #259: SD card is not skipped for booting when its first partition is not of type 0xc

tomty89, 2020-11-30:
Firmware is the 1028 beta. Supposedly it should at least skip when the type is 0x83, instead of only 0xc (or maybe some other type for the FAT family). blkdiscard'ed the whole card before partitioning.

Tested / desired "next attempt" is a USB3 drive.

Besides, unless there's a technical difficulty to implement so, it should do it in a whitelist approach (i.e. skip / ignore all non-FAT codes), instead of "blacklisting" only 0x83.

tomty89, 2020-11-30:
Hmm, it seems like 0x83 and 0x7 are the problematic ones. At least 0x8e is skipped.

tomty89, 2020-11-30:
Btw when the issue occurs, neither the diagnostic page nor the rainbow splash is seen. It seems like the blockage occurs at a very early stage.

tomty89, 2020-11-30:
It might be related to the "known" (known where? no idea) exfat issue. If the 0x8e is formatted with pvcreate, it's still skipped, but when it's formatted to exfat, similar (as in, nothing is displayed) blockage occurs.

timg236, 2020-11-30:
I've re-classified this as an enhancement and changed the title to reflect the functionality which you are requesting.

lurch, 2020-11-30:
> Btw when the issue occurs, neither the diagnostic page nor the rainbow splash is seen. It seems like the blockage occurs at a very early stage.

AFAIK (and this isn't my area of expertise) the boot-process on the Pi4 goes bootrom -> EEPROM -> start.elf -> kernel7l.img
 * The bootrom is embedded inside the BCM2711 SoC and is non-updateable (from Tim's comment on the [other issue](https://github.com/raspberrypi/rpi-eeprom/issues/243) it sounds like there was an exfat-related bug in the BCM2711B0 (as used in the Pi 4B) which was maybe fixed in the BCM2711C0 (as used in the Pi400 and CM4) ? :man_shrugging: )
 * The EEPROM image is [updateable](https://github.com/raspberrypi/rpi-eeprom), and I believe it's this that displays the diagnostics screen. The EEPROM then determines where to load start.elf from...
 * [start.elf](https://github.com/raspberrypi/firmware/tree/master/boot) displays the rainbow splash screen and then loads kernel7l.img
 * kernel7l.img is the (LPAE) Linux kernel which then does the rest of the boot process...

tomty89, 2020-11-30:
@timg236 I have no idea where is your conclusion from. Never mentioned anything about "specified number". While I mentioned *looping through* all the (four) primary entries in the locked #243, this issue has nothing to do with that. Only one partition was created in the testing subject.

It's ironic that you insists that I should file a new issue while you don't even feel like there's a chance that #243 is not actually fixed or there's similar problem at another point/stage.

Try it yourself. Blank (blkdiscard) the card, create one partition, type 0x83 or 0x7, no need to format it, the pi4 doesn't boot at all. End of story.

tomty89, 2020-11-30:
@lurch Yeah that sounds plausible. I guess I've been wasting my time either way.

timg236, 2020-11-30:
#243 Fixed a specific issue in the second stage loader which is why it was verified and closed.

Partition 0x7 exFAT is a different an unfixable error in the ROM on B0.

If no-valid partitions are found then partition 0 will be treated as "superfloppy" i.e. assumes it's FAT which might cause problems if 0x83 really is exFAT.

tomty89, 2020-11-30:
Please. Just stop making stuff up that I didn't say / explicitly stated otherwise. I told you multiple that card / partition is blanked.

No one ever verified in #243 that the issue is fixed (as in, there's an actual behavioral change) except you.

I'm done with this. Btw I think I deserve to right to say "don't change the title of an issue unless you are sure that's what the OP is about". It's even what I've ever wanted.

Closing this as I want nothing to do with a bad idea. File a new issue if you want a reminder for yourself.

@pelwell And seriously, after all this I have no idea you could even said what you said in #243. I almost feel like I'm owed an apology.

closed 2020-11-30 by tomty89

pelwell, 2020-11-30:
You want an apology for this?:

> If @timg236 thinks you have a different problem then you probably do. Start a new issue, stating the behaviour you are seeing and how it differs from what you expect, and let us worry about the why.

a. This is our repo, and you play by our rules.
b. I see nothing offensive in either the information I was attempting to convey or how it was worded.
c. You have a right to feel offended if you want to, but you don't have a right to not be offended.

tomty89, 2020-11-30:
https://github.com/raspberrypi/rpi-eeprom/issues/243#issuecomment-735240310

Mainly the first sentence.