raspberrypi/rpi-eeprom #260: Rpi 4 keeps flashing and do not boot

petrkr, 2020-12-03:
Probably after flash latest rpios `2020-08-20-raspios-buster-armhf-lite.zip` rpi keeps flashing green led in about 500ms interval, continuous, there is no error code regarding to EEPROM led flash codes. 

Already tried latest EEPROM Recovery, all same..

There is not care if sd is inserted or not, still same. Can that be connected to corrupted EEPROM of some wrong EEPROM image ?

timg236, 2020-12-03:
For support questions please use the Raspberry Pi forums and follow the  "STICKY: Is your Pi not booting" thread
https://www.raspberrypi.org/forums/viewtopic.php?p=437084

petrkr, 2020-12-03:
If I will have question I will do that, but this is bug report about continuous flashing LED, which is undocumented state. So there can be BUG in bootloader, documentation or HW.

In table on that forum (and regarding pages) are only upto 10 flashes, then 2sec gap.. but this state is non-stop flashing in interval around 500ms, where blink can be like 10-20ms long, I do not have log analyzer here, so I only guess by "eye". And since it do that also without SD card, then it could NOT be about that 32GB SD issue, again meniton in forum.

I tried also UART, but seems bootloader does not write anything there and for JTAG I have not documentation/cables/SDK

timg236, 2020-12-03:
Please answer the questions in the bug report template.

petrkr, 2020-12-03:
**Describe the bug**
Seems like not reading SD even EEPROM, only continuous flashing green ACT red

**To Reproduce**
Do not know as I do not want brick another Rpi, but steps done:
1) buy
2) test without anything - can see HDMI output with bootloader
3) Tested USB boot -- not working, because some older bootloader (do not remember exact version, but docs says "not support")
4) insert SD card with `2020-08-20-raspios-buster-armhf-lite.zip`
5) bricked

**Expected behaviour**
not bricked

**Screenshots**
No screenshot as it has blank screen without signal on HDMI

**Bootloader version and configuration**
do not know as it's blank

**SD card boot (please complete the following information):**
Does not care if SD is there or not
SD card type: tested 16G San disk from other Rpi4 where it is working, 32G Sand Disk both Class 10

**Network boot (please complete the following information):**
there is NO bootloader at all

**USB boot (please complete the following information):**
not working 

**Additional context**
Compared to other Rpi4 (where it's working)
without anything connected (also without SD card)
working Rpi:
 -- green led lights up for like 1-2sec (fast flashing like reading)
 -- keeps 4 times short flash,, 2 sec delay (regarding to documentation, this is righ state for "not start.elf found" or somethig)

not working Rpi:
 -- green led short flashing unlimited times right after power is connected (there is NO that "reading" flashing as on working one)

So my guess is HW problem or EEPROM problem or ROM problem... And because it happen after insert SD card with that image mention before (32GB that time) I suppose if there is also not some wrong eeprom.img which could brick that.

Is there any way how to check CPU itself? JTAG? UART?

Or maybe just will use warranty as last option.

petrkr, 2020-12-03:
I forgot mention it's 8GB version.

petrkr, 2020-12-03:
Almost similar like this `https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=253776` but they have  problem thta green only bllink once, at my side it flashing without stop... Nevermind, just ordered new Rpi 8GB so let's try tomorrow same steps if it will brick another one.

timg236, 2020-12-03:
Thanks for the information, much better! Details like the board revision are very important and we don't have time to ask the same questions every bug report.

All 8GB boards should have the HDMI diagnostics screen enabled so it must be failing very early on.

Unfortunately, it sounds like there is either a problem with either the SDCARD interface or possibly SDRAM/power supply.

It's very unlikely to be an EEPROM failure because it's flashed and checked during manufacture and those chips are very resiliant and located on a part of the board which would not get damaged easily.

petrkr, 2020-12-03:
Ok some more discovery:
 - whenever green flashes, red one littlebit glows down
 - chips near USB-C heats too much

tried also power-up by GPIO - same

So really probably HW issue, nevermind, already sending to warranty, will report if they accept or not.

timg236, 2020-12-03:
Ok, thanks, it's sounding like an issue with the PMIC.

timg236, 2020-12-17:
Closing - hardware failure

closed 2020-12-17 by timg236

petrkr, 2020-12-17:
Suggesting to write this to documentation as something like "flashing nonstop, without any HDMI signal --> probably HW failure" so peoples will not looking for SW problem as me did. Anyway still not have feedback from warranty.

mrowles, 2022-01-16:
This also happened to my Raspberry Pi 4 8GB after installing Ubuntu ~20 and running some normal apt-get updates. I shut it down and it has never turned back on. I've tried multiple different operating systems, configs for HDMI, resetting/updating the EEPROM - to no avail. Green. Blinking. Lights.