raspberrypi/rpi-eeprom #261: USB-Boot fails after doing a soft reboot

RefineryX, 2020-12-07:
**Describe the bug**
I am booting from a **[Crucial BX500 SSD](https://www.amazon.co.uk/dp/B07G3L3DRK/ref=cm_sw_em_r_mt_dp_ZZqUFb63J292V?_encoding=UTF8&psc=1)** using a **[StarTech USB 3.0](https://www.amazon.co.uk/gp/product/B00HJZJI84/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)**(USB3S2SAT3CB) cable. When doing a soft reboot via ``sudo reboot`` the system does not boot back up (but it does work fine if doing a hard reboot).

@tmm1 was having similar issues in #241. I installed the ``2020-10-28 BETA`` and seemed to be working fine  but has returned on the ``2020-11-24`` version. 

_Unfortunately, I do not have a HDMI cable to attach a monitor to get the output of the screen. If you think that will help (if no one else has reported such issues), I can purchase one to help with the issue._

**Expected behaviour**
Always boot via Crucial BX500 SSD when doing a soft reboot via ``sudo reboot``.

**Bootloader version and configuration**
```
Nov 24 2020 15:08:04
version 6fdd3d2be0127f72508c4c2ed48df7b260637ed9 (release)
timestamp 1606230484
update-time 1607196873
capabilities 0x0000001f
```

**USB boot:**

I am booting from: **[Crucial BX500 SSD](https://www.amazon.co.uk/dp/B07G3L3DRK/ref=cm_sw_em_r_mt_dp_ZZqUFb63J292V?_encoding=UTF8&psc=1)** using a **[StarTech USB 3.0](https://www.amazon.co.uk/gp/product/B00HJZJI84/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)** (USB3S2SAT3CB) cable.

In addition, I have a [WD Elements](https://shop.westerndigital.com/products/external-drives/wd-elements-desktop-usb-3-0-hdd#WDBWLG0030HBK-NESN) connected that stores media only and not used for booting.

<details>
<summary>'lsusb -vvv' Output</summary>
<br>

```
Bus 002 Device 003: ID 1058:25a3 Western Digital Technologies, Inc. Elements Desktop (WDBWLG)
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x1058 Western Digital Technologies, Inc.
  idProduct          0x25a3 Elements Desktop (WDBWLG)
  bcdDevice           10.19
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x002c
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                8mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 356 lines cut ...]
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```
</details>

lurch, 2020-12-07:
Have you tried experimenting with the `USB_MSD_PWR_OFF_TIME` setting? https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

RefineryX, 2020-12-07:
Thanks for the link @lurch. I've just read through and two settings stand out: 

It seems with ``USB_MSD_EXCLUDE_VID_PID=`` I can exclude my WD Elements drive from being considered for booting. I have included this in my config (including the VID and PID.

I also try a few settings using ``USB_MSD_PWR_OFF_TIME=``. I tried 2000, 3000, 4000 and 5000. I also tried ``USB_MSD_PWR_OFF_TIME=0``. Unfortunately these do not work. 

_Given these settings, I have updated my config in [post 1](https://github.com/raspberrypi/rpi-eeprom/issues/261#issue-758832571).

RefineryX, 2020-12-12:
Just did a fresh install on my Pi. Here's a couple of things I discovered _(still with the same boot loader config)._

**1.** With both devices plugged in (Crucial BX500 and WD Elements HDD), it boots fine every-time At this stage the WD Elements is not mounted.
**2.** When mounting the WD Elements in ``/etc/fstab``, after a reboot this appears to be the point of failure. It no longer warm reboots and fails every time. A hard restart is required.

Contents in ``/etc/fstab``:
```
proc            /proc           proc    defaults          0       0
PARTUUID=a47ba398-01  /boot           vfat    defaults          0       2
PARTUUID=a47ba398-02  /               ext4    defaults,noatime  0       1
UUID=bcec36c4-0184-4183-b8be-912f8f5c9c43 /mnt/elements ext4 defaults,nofail, 0 0
```
``PARTUUID=a47ba398`` is the BX500 and the the other (ending in ``9c43``) is the one I mounted.

lurch, 2020-12-12:
Could you try **1a.** - boot with both devices plugged in but with WD Elements not mounted, mount it manually using the `mount` command (i.e. not via `fstab`) and then see if it warm-reboots successfully?

This sounds like such a strange edge-case that I suspect Tim will need [XHCI_DEBUG logs](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) in order to diagnose this further.

pelwell, 2020-12-12:
I would recommend getting an HDMI cable or hooking up a serial interface in order to find out at what point the boot fails. Without this information we are just guessing.

Also think about the power supply - one difference with the automatic mounting is the higher peak power consumption during boot when, thanks to systemd, everything is happening at once.

RefineryX, 2020-12-12:
Thank you both (@lurch + @pelwell - appreciate your time!

>  1a. - boot with both devices plugged in but with WD Elements not mounted

I have just commented out the line in ``fstab`` and stopped the HDD from mounting. Rebooting again and strangely, the rPi does not boot.

> I would recommend getting an HDMI cable or hooking up a serial interface in order to find out at what point the boot fails. Without this information we are just guessing.

Thanks Phil, you're right. I have went ahead and order a cable which should arrive tomorrow. Hoping that there is a obvious error waiting for me when I login.

I think my power supply is fine and has plenty of power. It is the only one connected into an [Anker PowerPort 60 W](https://www.amazon.co.uk/gp/product/B00PK1IIJY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). Lucky this has been pretty solid over the years.

So far, I am thinking it's fine when my main bootable SSD is plugged in (BX500), when when I my elements hard drive is also plugged in, it does not like it. Once the cable has arrived, I'll plug in and reveal any findings.

RefineryX, 2020-12-14:
So the able arrived but a an opps moment! I ordered the a mini USB cable rather than the micro! Should arrive on Thursday and should be able to show you what is going on at boot.

I also forgot to mention, I have another USB attached which is the conbee ii Zigbee stick which I use for my home automation system. I managed to get the UID and PID for this also and just added it to the config... made no difference however.

Here is the latest config:

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
USB_MSD_EXCLUDE_VID_PID=105825a3,1cf10030
USB_MSD_PWR_OFF_TIME=0
```

lurch, 2020-12-16:
A Zigbee stick probably doesn't appear as a USB Mass Storage Device, and so there's no need to add it to the MSD_EXCLUDE list :slightly_smiling_face:

RefineryX, 2020-12-16:
Alright, cable arrived. I plugged it in and rebooted and here's the error:

[image]

timg236, 2020-12-21:
Please can you try the experimental bootloader on this comment https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-748455913 to see if it improves the situation.

RefineryX, 2020-12-21:
Thanks! I have uploaded this firmware and it gets passed the error screen. Now that I have a monitor, I can now see things I couldn't before. Upload booting, I noticed an error with 'Network-Manager' which looks like it was left over from some other install (Home Assistant) and I deleted it. After a few restarts, I am still facing the same issue of not being able to login to SSH after a soft-reboot (I am able to login directly on the Pi but not on another system on the network).

**Fresh install:** Fast forward a couple of hours, I decided to start a clean install and not configure a single thing yet to rule everything I have done out. First boot, I login via the default credentials then initiate the reboot ``sudo reboot``. It failed on the first attempt. Normally I would remove the power, wait 10s and plug it back in and it would work. This time, I thought I would remove the ethernet cable and plug it back in and interestingly I was able to login via SSH.

**Summary:** I am not sure what is going on here. It seems like it just looses the connection on a soft-reboot and struggles to recover from it. Upon removing the ethernet and plugging it back in, I can again login to SSH. I can always login to SSH directly on the Pi. When in the state when I am unable to login externally... I try to ping my router via the Pi and it does not respond to pings. I remove the ethernet cable, plug it in again and suddenly everything works again.

[image]

RefineryX, 2020-12-22:
Here's a couple of things I found out:
1.  Plugging in the hard drive and using the USB 2.0 port works. 
**Appears that the problem maybe isolated to the USB 3.0 port**
2. Running ``mii-tool -r eth0`` seems to reset and allow a connection
3. Unplugging the ethernet cable resets the connection and allows ssh to connect agian

lurch, 2020-12-24:
If Linux is running, then your problems with networking are a Linux-related issue rather an EEPROM-related issue, so you should probably ask for help on the Raspberry Pi [forums](https://www.raspberrypi.org/forums/).

timg236, 2021-02-23:
The 2021-02-16 stable release now configures now configures the Ethernet PHY rxid mode in the same manner as the 5.10 Raspberry Pi OS kernel. This might improve the situation on Home Assistant.

timg236, 2021-03-10:
Closed due to inactivity

closed 2021-03-10 by timg236