raspberrypi/rpi-eeprom #262: SSD boot, doesn't work for USB3, also often loses connection.

BlackTurtle123, 2020-12-09:
https://github.com/home-assistant/operating-system/issues/1012#issue-748036889

**Describe the bug**
When booting from ssd setup, I can't boot from usd3. I am using the rpi4 4GB. Top link is to link to my hassos ticket were I started to open the issue.
I am using a samsung evo970  NVME M2

**Bootloader version and configuration**
eeprom is currently the version in this hassos release. https://github.com/home-assistant/operating-system/releases/tag/5.6
But before it was the latest from the critical repo in raspbian itself.

**USB boot (please complete the following information):**
SD-card boot does do the job. And USDB2 boot also does the job.
I am using following case: https://plugable.com/products/usbc-nvme/
Tried following external power sources;
- https://www.coolblue.be/nl/product/824357/kensington-k39122eu-usb-3-0-4-poorts-hub.html
- https://www.coolblue.be/nl/product/852110/bluebuilt-7-poorts-usb-a-c-3-0-hub.html

timg236, 2020-12-09:
Please can you read through the bug template and add relevant logs as to at what point fails bootloader version.

You'll also need to reproduce the bug in Raspberry Pi OS latest.

BlackTurtle123, 2020-12-09:
> Please can you read through the bug template and add relevant logs as to at what point fails bootloader version.
> 
> You'll also need to reproduce the bug in Raspberry Pi OS latest.

There are no logs since it simply doesn't boot when using USB3... And yeah, I did this. I also installed the raspberry pi bootloader from the sd-card

timg236, 2020-12-09:
You'll need to provide the bootloader logs before anyone will spend any time on this. I suggest installing Raspberry Pi OS to the SSD using the Raspberry Pi Imager and at least capturing the HDMI diagnostics screen.

BlackTurtle123, 2020-12-09:
Hello.
Reflashed the ssd. With the image from the website.
https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2020-12-04/2020-12-02-raspios-buster-armhf.zip
The image is the first screen, the wetransfer link is the video. No clue how I otherwise need to get the logs to you guys.
Hope this helps.
[image]

https://we.tl/t-qJSDNDJzbs

timg236, 2020-12-10:
Was the video taken from power-on from cold or after a soft-reboot? It looks as though the USB / NVME adapter did not respond after the first reset but then did enumerate afterwards "MSD LUN" line. The video stopped there but I'm guessing that it didn't respond to the MSD SCSI command to get_capacity etc

BlackTurtle123, 2020-12-10:
The result is both the same in hot or cold boot. After the video it goes
just to black screen. Like nothing happens.

timg236, 2020-12-10:
"just to black screen. Like nothing happens."

The absence of errors messages after "LUN 0" makes me think that this is some sort of USB back power issue when the SSD starts. If you have a USB serial cable then it would be interesting to enable verbose tracing and see if that's the command.

What's the state of the LEDs after the screen goes blank?

BlackTurtle123, 2020-12-10:
Mmm, is it possible that usb3 has this potential power issue but usb2
doesn't?
I don't have such a cable by hand at the moment. But I can try yet another
powered usb hub maybee?

timg236, 2020-12-10:
"Mmm, is it possible that usb3 has this potential power issue but usb2 doesn't?"

Yes, drive enclosures often behave quite differently with USB 3.0 compared to USB 2.0. It might be worth booting with an SD card and verifying that the hotplugging the drive from Linux works plus rsyncing some large files to/from the device.

carras, 2020-12-11:
What's the controller chip of your USB enclosure?

carras, 2020-12-11:
I have this issue, which btw It does boot HassOS if you turn it on without USB and then you connect it.

I use an enclosure with the JMS580 from Jmicron, that It used to work.

I've try the EEPROM on the beta channel from dec11 and it doesn't work either.

Now, I'm trying to go back with EEPROM version because It used to work, maybe going back a few version solve the problem until a better solution.

timg236, 2020-12-11:
@carras Please create a separate issue unless you have the same USB SSD + enclosure as the OP.

carras, 2020-12-11:
> @carras Please create a separate issue unless you have the same USB SSD + enclosure as the OP.

Well, from the link he offered there are two version of his case:

"To help identify the new SSD enclosures we have changed the silkscreen, in the figure above the left image is the original JMicron JMS583 chipset, and the right is the updated silkscreen with the Realtek RTL9210 chipset. The newer devices also include serial numbers starting with “2645-” and up."

If he uses JMS583, I use JMS580. There are controller from Jmicron that seem to have the same issue.

If I do open a new issue, I feel I'll be asking for the same thing for the same issue.

BlackTurtle123, 2020-12-11:
Hello!
Some update.
I notices when attaching the ssd enclosure directly to the rpi, i would be
able to make it boot. But would later on crash due powee consumption. This
made me realise it's the external power supply that is the bad actor or
something connected. I sended back the coolblue one, since this simply
wouldn't do a thing. And in usb2 would disconnect. I went around and
started play. I swapped ports, i changed my conbee and mouse usb stick to
usb2 for conbee and my mouse stick also. Was before on the external usb
hub. And i managed to boot through the kingston one. Not sure if it was a
lucker, or not. Also not sure if it's gonna be stable or lose the drive
connection at some point. I also use custom boot paraps which turn off the
power saving. And I used the quirks (also tried that with the other
external source, but didnt worked). And it's running now for almost 18
hours. So it might be fully solved on hardware level mostly.

carras, 2020-12-14:
I've open a new issue for controllers JMS583 and JMS580 of Jmicron: https://github.com/raspberrypi/rpi-eeprom/issues/266

Wolverine80, 2020-12-15:
I updated my Pi4b today, which installed rpi-eeprom:armhf 11.0-1, and also have the problem now that at reboot the USB power to the attached Adata USB SSD is turned off  (power LED on the drive goes on and then off). 
I can get it to boot when I connect the drive right after the Pi has been powered on. 
So not only when powered on, even at restart it  shuts the USB drive off. 
I booted from SD to confirm that the USB SSD is not listed in lsusb. I have to reconnect it.

timg236, 2020-12-15:
> I booted from SD to confirm that the USB SSD is not listed in lsusb. I have to reconnect it.

If the drive does not appear when booted from SD then it's not really suitable as boot device because it's not handling disconnect / reconnect properly. You could try setting USB_MSD_PWR_OFF_TIME=0 in the EEPROM config but there's no guarantee that power will be preserved during a reboot.  On Pi4 8GB USB port power will be disconnected by the hardware whic is why USB_MSD_PWR_OFF_TIME=0 is not really supported.

Wolverine80, 2020-12-15:
I will try. I am just wondering because it worked for month without a problem, until I updated to version 11.0-1 today (was 10.3-1 before)

timg236, 2020-12-15:
The timing of various stage of the bootloader will have changed in releases, it may help some drive enclosures but be worse for others.

Wolverine80, 2020-12-15:
I see. I just tried it with USB_MSD_PWR_OFF_TIME=0, did not help in my case.

timg236, 2020-12-19:
Please can you try the test bootloader on this comment
https://github.com/raspberrypi/rpi-eeprom/issues/266#issuecomment-748455913

Wolverine80, 2020-12-21:
I tried. And it successfully booted again with the Adata USB3 disk. 
I only tried a few soft and hard reboots, but looks promising so far.

pnt, 2021-01-29:
To chime in, I'm also using a Adata USB3 disk and neither does the referenced build in #266 from Dec 19th nor the official Jan 16 bootloader fix things. 
Setting USB_MSD_PWR_OFF_TIME=0 also makes no difference.
@timg236, any other troubleshooting I can do to contribute info to the bug?

leontodd, 2021-02-20:
After following all the same steps I am also having the same issue.  I'm using an Adata drive too.

timg236, 2021-02-20:
The bug report is about Samsung EVO drives which I believe it not an Adata drive. Best to raise a new bug with links to the the specific product (enclosure + SSD NVME)

timg236, 2021-03-03:
Verified that the current stable release 2021-02-16 works with the OP usb nvme with Pi4B 1.1 and official power supply

closed 2021-03-10 by timg236

jhoughjr, 2023-03-04:
Having this issue in 2023.

peterharperuk, 2023-03-06:
@jhoughjr If you have an issue I would suggest you raise a new report with full details. Thanks.