raspberrypi/rpi-eeprom #263: Update - broke firmware

m88beau, 2020-12-10:
Using a Pi4B 8GB, with 32GB Class 10 SDHC card.
Unit was working fine, ran sudo apt update, then sudo apt full-upgrade
When it got to setting up rpi-eeprom-update, it just hanged.
I opened another SSH, Pi was still responding. no activity light, so I rebooted the unit.
It still boots but whenever I try to run any rpi-eeprom commands, i get this,

timg236, 2020-12-10:
If you are able to run Linux and run vcgencmd then the bootloader and EEPROM must be running correctly. 

I suspect something has gone wrong updating the firmware or userland and the forums would be the best place to ask for support on that.

closed 2020-12-10 by timg236

lurch, 2020-12-10:
If you can, it might be worth trying a fresh install of the latest Raspberry Pi OS on a fresh (spare) SD card, and seeing if that gives you the same errors?

As @pelwell mentions [here](https://github.com/raspberrypi/linux/issues/3919#issuecomment-739488196), a corrupt EEPROM is likely to lead to a Pi that won't boot, so I wonder if you just have some corrupt files on your current SD card? :shrug:

m88beau, 2020-12-10:
I can't run any vcgencmd commands, they all fail with vchi_msg_dequeue -> -1(22)
Which -1 errors is listed as Problem with VCHI.

pelwell, 2020-12-10:
What do the following commands report?
```
$ dmesg -l err
$ sudo vcdbg log msg 2>1
```
The `2>1` sends stderr to stdout, which might be useful if you want to write the output to a file:
```
$ sudo vcdbg log msg 2>1 > log.txt
```

pelwell, 2020-12-10:
Also, and this may seem a bit random, but if you are using both HDMI ports, try disconnecting one and see if that makes a difference.

m88beau, 2020-12-10:
dmesg -l err
[    5.365179] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.646965] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.658538] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar 23 2020 02:19:54 version 7.45.206 (r725000 CY) FWID 01-88ee44ea

sudo vcdbg log msg 2>1 > log.txt
nothing is written to the log file.

Also:
sudo rpi-eeprom-config
ERROR: vcgencmd bootloader_config timeout

This is what worries me the most to be honest...

pelwell, 2020-12-10:
The vcgencmd failure just suggests that one of the comms channels between the VPU and the ARMs is unhappy. You can verify that with a harmless command such as `vcgencmd version`. Assuming that fails, try `strace vcgencmd version` and report the output.

m88beau, 2020-12-11:
Full log of sudo strace vcgencmd version
[log.txt](https://github.com/raspberrypi/rpi-eeprom/files/5679565/log.txt)

Notably:
`clone(child_stack=0xb6d73f98, flags=CLONE_VM|CLONE_FS|CLONE_FILES|CLONE_SIGHAND|CLONE_THREAD|CLONE_SYSVSEM|CLONE_SETTLS|CLONE_PARENT_SETTID|CLONE_CHILD_CLEARTID, parent_tidptr=0xb6d744c8, tls=0xb6d74920, child_tidptr=0xb6d744c8) = 20936
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x2, 0x1c), 0xbec01564) = -1 EIO (Input/output error)
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0) = -1 EINVAL (Invalid argument)
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0) = -1 EINVAL (Invalid argument)
ioctl(3, _IOC(_IOC_WRITE, 0xc4, 0x4, 0xc), 0xbec015c4) = -1 EINVAL (Invalid argument)
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xd, 0), 0) = -1 EINVAL (Invalid argument)
fstat64(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x1), ...}) = 0
write(1, "vc_gencmd_send returned -1\n", 27) = 27
ioctl(3, _IOC(_IOC_NONE, 0xc4, 0xc, 0), 0) = -1 EINVAL (Invalid argument)
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0xc4, 0x8, 0x10), 0xbec015c0) = -1 EINVAL (Invalid argument)
write(2, "vchi_msg_dequeue -> -1(22)\n", 27vchi_msg_dequeue -> -1(22)`

Sorry for long post

JsBergbau, 2021-08-29:
I also got error `vchi_msg_dequeue -> -1(22)` with command `/opt/vc/bin/vcgencmd measure_temp`
There the problem was a faulty camera cable. After replacing and rebooting the Raspberry PI command worked again as expected.