raspberrypi/rpi-eeprom #272: 8gb pi4 usb boot

ShorTie8, 2021-01-07:
Can not get usb boot to work on a pi4 8gb.
Revision        : d03114
Trying with 2020-12-02-raspios-buster-armhf.zip
No other usb devices are attached.
The same usb drive is a Samsung usb 3.0 flash drive bar, works fine in my other 2 4gb pi4's.
Updated to stable pieeprom-2020-12-11.bin this morning and still no go.
What I get is a small screen with red lines and
Failed to open device: "sdcard" (cmd 371a0010 status 1fff0001).
Insert a sdcard and it boots.

 vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x1
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0

vcgencmd bootloader_version
Dec 11 2020 11:15:17
version c3f26b6070054bca030366de2550d79ddae1207a (release)
timestamp 1607685317
update-time 1610026496
capabilities 0x0000001f

timg236, 2021-01-07:
This sounds like a hardware issue

ShorTie8, 2021-01-07:
Sure does sound like a hardware issue.
Like a pi4 8gb hardware issue.
The usb drive works perfectly in 4gb pi4's.
How could it be a write problem that Raspberry Pi Imager could fix when it works in the other 2 ??

timg236, 2021-01-07:
Yes could be a problem on that pi board. A photo of the diagnostics screen would be useful. The red lines could indicate a hdmi cable issue try swapping boards and cables around

NeddySeagoon, 2021-01-07:
Copy the SPI firmware drom a working Pi and flash the 8G Pi with it.

timg236, 2021-01-07:
Moved to "awaiting information" - until logs are provided

ShorTie8, 2021-01-08:
Sorry it took so long, had to get a new printer for taxes.
Bypassing my hdmi switch took care of the red lines.
here is a photo
[image]

timg236, 2021-01-08:
It looks as though you don't have USB BOOT enabled (BOOT_ORDER=1). I think the best thing to do is to use the Raspberry Pi Imager to flash the default factory settings EEPROM (misc utils) to a spare SD card.

Or from within Raspberry Pi OS use the following command to edit the EEPROM config and replace it with 

[all]
BOOT_UART=0
WAKE_ON_GPIO=1

# apt update / apt upgrade first
sudo -E rpi-eeprom-config --edit

ShorTie8, 2021-01-08:
Did the apt update/upgrade
But get
 sudo -E rpi-eeprom-config --edit
ERROR: vcgencmd bootloader_config timeout

ShorTie8, 2021-01-08:
Does this
[all]
BOOT_UART=0
WAKE_ON_GPIO=1

Go into config.txt ??

ShorTie8, 2021-01-08:
Well Thankz for your unclear directions or sumfin.
I've flashed back to pieeprom-2020-09-03.bin and even copied recovery.bin to boot.
I've reflashed the sdcard multiple times.
Now the pi is trashed.
When I try a vcgencmd bootloader_version it goes off into never never land.
All the icons disappear and if I ssh in a 2nd time and try to reboot or shutdown now -h, nothing happens.
Waiting on what to do next I guess ??

ShorTie8, 2021-01-08:
Ps: Don't really like that new computer voice thing.
Sounds like some 1980's computer voice that I can't even understand .. :/~

timg236, 2021-01-08:
OP says USB does not work when posted config shows USB not enabled. Github bug as not support tickets and given the tone of your previous comments I'm not spending any more time on this.

closed 2021-01-08 by timg236