raspberrypi/rpi-eeprom #275: Rpi-eeprom-update from 1970??

himuura, 2021-01-12:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model
Pi 4 4Gb
* Board revision (cat /proc/cpuinfo | grep Revision)
Revision        : c03111
* Operating system version
Raspberry OS 32-bit
* Details of any hardware attached e.g. links to USB 
Boot from SSD enclosure (with a Kingston SSD) via USB.
* Photo of the HDMI diagnostics screen, UART trace.
Don't think it is actually needed

**Describe the bug**
Simple, just updated my pi 4 and got:

BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: sex dez 11 11:15:17 UTC 2020 (1607685317)
 LATEST: qui jan  1 00:00:00 UTC 1970 (0)
 FW DIR: /lib/firmware/raspberrypi/bootloader/default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

Latest from 1970 is kind of weird to me :P 

**To Reproduce**
rpi-eeprom-update should be enough

**Expected behaviour**
Not an epoch date would be cool :)

**Bootloader version and configuration**
Dec 11 2020 11:15:17
version c3f26b6070054bca030366de2550d79ddae1207a (release)
timestamp 1607685317
update-time 1608028468
capabilities 0x0000001f

timg236, 2021-01-12:
It's a script bug due to the way in which 'find' handles symlinks. Should be fixed by
https://github.com/raspberrypi/rpi-eeprom/commit/690872a3ca4274f1f53d090a154750c3409fc23b

himuura, 2021-01-12:
> 
> 
> It's a script bug due to the way in which 'find' handles symlinks. Should be fixed by
> [690872a](https://github.com/raspberrypi/rpi-eeprom/commit/690872a3ca4274f1f53d090a154750c3409fc23b)

Thought the same :) Thanks for your reply mate, cheers!

timg236, 2021-01-12:
Thanks for reporting that, the APT update should now be available.

himuura, 2021-01-12:
Exquisite!!! Working as intended 😄 
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: sex dez 11 11:15:17 UTC 2020 (1607685317)
 LATEST: qui set  3 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

Many, many thanks!Consider it closed with style! Cheers!

timg236, 2021-01-13:
Thanks. Closed

closed 2021-01-13 by timg236