raspberrypi/rpi-eeprom #277: Not a bug but a request.

tlsalex, 2021-01-13:
**Mandatory information**
* Raspberry Pi 4 model B
* Board revision (cat /proc/cpuinfo | grep Revision)
* Operating system version .
* Details of any hardware attached e.g. links to USB 
* Photo of the HDMI diagnostics screen, UART trace.

**Describe the bug**
I tried to find the official steps to boot my Raspberry Pi 4 from USB3.0. From google and youtube, I found much info about how to boot PI4 from USB3, but some of them is a little old and all of them is not official. I may know how to install or update application on Centos or Debain(you may say Pi OS is based on Debain). but I am not master PI OS , and I want to know the official steps to upgrade the firmware of Raspberry Pi 4 to latest version.

Most projects has a brief or detail steps to guide users to use its applications.  Could someone please add the steps about how to update the firmware to README.md

I think there is no relationship between OS version and firmware version. I just need to update firmware version, then I can use whatever version of OS, maybe I am wrong, please forgive me.

timg236, 2021-01-13:
Please post support requests on the forums. 
https://www.raspberrypi.org/forums/

closed 2021-01-13 by timg236

lurch, 2021-01-13:
README has a link to https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md which appears to be what you're looking for.