raspberrypi/rpi-eeprom #279: pieeprom-2021-01-11: reboot fails

KrisztianKende, 2021-01-16:
Raspberry Pi 4 Model B Rev 1.1 - 4 GB
SD card ([SanDisk Extreme Pro 32 GB](https://shop.westerndigital.com/products/memory-cards/sandisk-extreme-pro-uhs-i-microsd#SDSQXCG-032G-GN6MA)), without any USB devices.

Bootloader config:
```
[all]
BOOT_UART=1
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
ENABLE_SELF_UPDATE=0
USB_MSD_PWR_OFF_TIME=0
```
[pieeprom-2021-01-11](https://pastebin.com/raw/duCLHRZh) vs [pieeprom-2021-01-05](https://pastebin.com/raw/RB710KjN)

What is missing (with pieeprom-2021-01-11) after reboot command:
```
PM_RSTS: 0x00001020
RPi: BOOTLOADER release VERSION:e4cda8a7 DATE: Jan  5 2021 TIME: 16:09:58 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1609862995 0xda31a10e 0x00c03111 0x000a9734
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 1.8V
GLOBAL_RESET: 01 0
```

timg236, 2021-01-16:
[rpi-eeprom-recovery-sd1v8.zip](https://github.com/raspberrypi/rpi-eeprom/files/5824385/rpi-eeprom-recovery-sd1v8.zip)

I think I can see the issue, missed this whilst testing the gpio-ex reset change because my card worked with either voltage :(

KrisztianKende, 2021-01-16:
Thanks, [it looks like](https://pastebin.com/raw/yBXHyEyL) it's working fine now. :-)

closed 2021-01-16 by KrisztianKende

timg236, 2021-01-16:
Thanks for the retest - the 1V8 to 3V3 reset via PMIC reset is visible in the linked log which is the correct behaviour.

timg236, 2021-01-16:
https://github.com/raspberrypi/rpi-eeprom/pull/280