raspberrypi/rpi-eeprom #282: Bootloader fails to find SD-Card and does not boot after soft reboot

galano25, 2021-01-19:
**Mandatory information**
* Raspberry Pi model: Raspberry Pi 4 Model B Rev 1.4
* Board revision (cat /proc/cpuinfo | grep Revision): b03114
* Operating system version: Linux version 5.4.72-v8 (oe-user@oe-host) (gcc version 9.3.0 (GCC)) #1 SMP PREEMPT Mon Oct 19 11:12:20 UTC 2020
* Details of any hardware attached: No USB devices attached, only SD Card and HDMI
* Trace of the issue:
* `root@raspberrypi4-64:~# reboot
[  OK  ] Removed slice system-bthelper.slice.
[  OK  ] Removed slice system-getty.slice.
[  OK  ] Stopped target Bluetooth.
[  OK  ] Stopped target Multi-User System.
[  OK  ] Stopped target Login Prompts.
[  OK  ] Stopped target Host and Network Name Lookups.
         Stopping Login Service...
         Stopping Load/Save Random Seed...
[  OK  ] Stopped Kernel Logging Service.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped Telephony service.
[  OK  ] Stopped Serial Getty on ttyS0.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped Bluetooth service.
[  OK  ] Stopped Mender OTA update service.
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Removed slice system-serial\x2dgetty.slice.
[  OK  ] Stopped target Network is Online.
[  OK  ] Stopped target Network.
         Stopping Configure Bluetooth Modems connected by UART...
[  OK  ] Stopped Wait for Net[  304.555980] Bluetooth: hci0: sending frame failed (-49)
work to be Configured.
         Stopping Network Name Resolution...
         Stopping WPA supplicant da��…interface-specific version)...
[  OK  ] Stopped Network Name Resolution.
         Stopping Network Service...
[  OK  ] Stopped Network Service.
[  OK  ] Stopped WPA supplicant dae��… (interface-specific version).
[  OK  ] Stopped target Network (Pre).
[  OK  ] Stopped IPv6 Packet Filtering Framework.
[  OK  ] Stopped IPv4 Packet Filtering Framework.
[  OK  ] Stopped Configure Bluetooth Modems connected by UART.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped CUPS Scheduler.
[  OK  ] Stopped Monitor VC4 driver loaded to start psplash.
[  OK  ] Stopped Dispatch Password ��…ts to Console Directory Watch.
[  OK  ] Stopped Forward Password R��…uests to Wall Directory Watch.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed Avahi mDNS/DNS-SD Stack Activation Socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Closed CUPS LPD Server Socket.
[  OK  ] Closed CUPS Scheduler.
[  OK  ] Closed GUI Socket to start service.
[  OK  ] Closed InLine Diag Socket to start service.
[  OK  ] Closed sshd.socket.
[  OK  ] Stopped target System Initialization.
[  OK  ] Closed Syslog Socket.
         Stopping Hardware RNG Entropy Gatherer Daemon...
[  OK  ] Stopped Apply Kernel Variables.
[  OK  ] Stopped Load Kernel Modules.
         Stopping Network Time Synchronization...
[  OK  ] Stopped Hardware RNG Entropy Gatherer Daemon.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /data...
         Unmounting Temporary Directory (/tmp)...
         Unmounting /uboot...
[... 318 lines cut ...]
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0xf41
SD_BOOT_MAX_RETRIES=0
NET_BOOT_MAX_RETRIES=0
DISABLE_HDMI=1
[none]
FREEZE_VERSION=0
`
**SD card boot (please complete the following information):**
 - SD card type: Microcenter 32GB Class 10. Link to card info:
 - https://www.amazon.com/Micro-Center-Class-Memory-Adapter/dp/B07K81Z6DF
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer:
 `Disk /dev/mmcblk0: 29.55 GiB, 31719424000 bytes, 61952000 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xc501daf3

Device         Boot   Start     End Sectors  Size Id Type
/dev/mmcblk0p1 *      49152  131071   81920   40M  c W95 FAT32 (LBA)
/dev/mmcblk0p2       131072 2015231 1884160  920M 83 Linux
/dev/mmcblk0p3      2015232 3899391 1884160  920M 83 Linux
/dev/mmcblk0p4      3899392 4161535  262144  128M 83 Linux`

timg236, 2021-01-19:
Please could you see whether this happens with Raspberry Pi OS 32bit latest image from the Raspberry Pi Imager. I’m wondering if something is missing bin the reboot case.

galano25, 2021-01-19:
Hello,
Yes it reproduces on Raspbian OS. Here are the logs:

> PM_RSTS: 0x00000020
> RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0x8afcbc2e 0x00b03114
> uSD voltage 3.3V
> Initialising SDRAM 'Samsung' 16Gb x1 total-size: 16 Gbit 3200
> VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
> VL805 0xfff00000 0xfff20000
> XHCI-STOP
> xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
> xHC ports 5 slots 32 intrs 4
> Boot mode: SD (01) order f4
> SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
> SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
> CID: 009c534f55534430301068a5ae110141
> CSD: 400e00325b590000ec537f800a400000
> SD: bus-width: 4 spec: 2 SCR: 0x02058083 0x00000000
> SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
> MBR: 0x00002000,  524288 type: 0x0c
> MBR: 0x00082000,61419520 type: 0x83
> MBR: 0x00000000,       0 type: 0x00
> MBR: 0x00000000,       0 type: 0x00
> lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
> rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
> PM_RSTS: 0x00000020
> Partition: 0
> lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
> rsc 32 fat-sectors 4033 c-count 516190 c-size 1 r-dir 2 r-sec 0
> Read config.txt bytes     2133 hnd 0x00016987 hash 'c7c0082c7ec3ad3e'
> recover4.elf not found (6)
> recovery.elf not found (6)
> Read start4.elf bytes  2215776 hnd 0x0000d407 hash '906e34e8700956d7'
> Read fixup4.dat bytes     5429 hnd 0x000003f8 hash '1d0725a9904ebc93'
> 0x00b03114 0x00000000 0x0000003f
> MEM GPU: 76 ARM: 948 TOTAL: 1024
> Starting start4.elf @ 0xfec00200 partition 0
> 
> Raspbian GNU/Linux 10 raspberrypi ttyS0
> 
> raspberrypi login:pi
> Password:
> Last login:Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l
> 
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> pi
> 
> Wi-Fi is currently blocked by rfkill.
> Use raspi-config to set the country before use.
> 
> pi@raspberrypi:sudo reboot
> [   28.430231] reboot: Restarting system
> 
> PM_RSTS: 0x00000020
> RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0x8afcbc2e 0x00b03114
> uSD voltage 3.3V
> Initialising SDRAM 'Samsung' 16Gb x1 total-size: 16 Gbit 3200
> VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
> VL805 0xfff00000 0xfff20000
> XHCI-STOP
> xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
> xHC ports 5 slots 32 intrs 4
> Boot mode: SD (01) order f4
> SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
> SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
[... 98 lines cut ...]
    print Reboot loop test...
    send "sudo reboot"
loop1:
    expect {
        "login:" send "pi"
        "Password:" send "raspberry"
        "pi@raspberrypi:" break
        timeout 2147483647 goto failed1
    }
    sleep 10
    send "sudo reboot"
    goto loop1
    exit
failed1:
    print System did not boot
    exit
```

You can run from the terminal:
```
sudo minicom -S ~/raspberry-pi-yocto/build/reboot_test_rpi.txt -C rpi_logs.txt
```
where reboot_test_rpi.txt contains the above script.

Thanks for your help and the amazing job on the Raspberry Pi!

timg236, 2021-01-19:
Thanks does it fail if "sdhci.debug_quirks2=4" is added to cmdline.txt? That version of the bootloader is quite old now and we leaving Pis in a reboot loop is something that's part of standard testing so I can't see what's different here except maybe the sd-card.

galano25, 2021-01-19:
I tried "sdhci.debug_quirks2=4" and it still happens. Here is how the new cmd line looks like:

> [    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
> [    0.000000] Linux version 5.4.72-v8 (oe-user@oe-host) (gcc version 9.3.0 (GCC)) #1 SMP PREEMPT Mon Oct 19 11:12:20 UTC 2020
> [    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
> [    0.000000] efi: Getting EFI parameters from FDT:
> [    0.000000] efi: UEFI not found.
> [    0.000000] Reserved memory: created CMA memory pool at 0x000000001ec00000, size 256 MiB
> [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
> [    0.000000] percpu: Embedded 32 pages/cpu s90328 r8192 d32552 u131072
> [    0.000000] Detected PIPT I-cache on CPU0
> [    0.000000] CPU features: detected: EL2 vector hardening
> [    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
> [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
> [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hea
> dphones=1 video=HDMI-A-1:480x1920M@30 smsc95xx.macaddr=DC:A6:32:F1:AB:25 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 c
> onsole=ttyS0,115200  rootfstype=ext4 rootwait   logo.nologo  root=/dev/mmcblk0p2 sdhci.debug_quirks2=4

Note that sometimes after 6 or more retries the bootloader eventually is able to boot the card. So you see this error:
> FAT read failed @ 16290
> Read start.elf failed
> start.elf not found (2)
> Firmware not found
> ERROR: 00000004

6 or more times and then it makes.

timg236, 2021-01-19:
Reboot works by doing a full chip-reset so the controller will be fully reset. It sounded like a UHS voltage related issue e.g. the card needing to be powered off for longer to switch back to 3V3 but the sdhci quirks should have avoided that. There's an updated stable release of the bootloader - select 'latest' in rasp-config - 2021-01-16 which resets the GPIO expander at power on. It's unlikely to fix it but should be harmless and easy to revert (via the Raspberry Pi Imager)

Meanwhile, I'll try to order some of these cards to add to the test pool.

galano25, 2021-01-19:
Thanks @timg236 . I'll update to the latest firmware and try again.

timg236, 2021-01-19:
Raspberry Pi and some resellers have been using Sandisk Ultra HC UHS U1 16GB or 32GB cards for the pre-packaged NOOBS image.  Panasonic UHS uSD cards have also been used recently but I couldn't find an Amazon link for them, perhaps volume order only.

https://www.amazon.co.uk/SanDisk-microSDHC-Adapter-Performance-SDSQUA4-032G-GN6MA/dp/B08GY9NYRM/

Samsung EVO seem quite popular on the forums and have been reliable for me.

galano25, 2021-01-20:
Hi @timg236 ,
Some good news here! With the latest firmware it works with no problem. Here is the firmware I installed:
> root@raspberrypi4-64:~# vcgencmd bootloader_version
> Jan 16 2021 14:10:13
> version d1be7b5ba3adb192b6192ea2561584aa5c7be6bd (release)
> timestamp 1610806213

I see now in the bootloader logs that it prints doing a PCI reset, I wonder if that is what fixed the issue. I see also your release notes that you mentioned **Fix 1V8 SD voltage reset for Pi 4B R1.1**. Maybe the 1.8V line for the SD Card was not stable before?

Is this firmware (Jan 16 2021) stable enough to use? I always get confused about whether latest/stable is good enough to use (instead of critical/default).

Thanks for your help here!

timg236, 2021-01-20:
PCIe is independent of the SDHCI controllers, different buses.

The 1V8 SD fix in Jan 16 fixes a specific regression in the 14th Jan release. This wouldn't happen on a b03114 board as shown in your log because when the board is rebooted the firmware has already toggled SD card power to get it out of 1V8 and into 3V3 mode before the chip was reset.

galano25, 2021-01-20:
OK, not sure then what could have fixed the issue. I'll keep testing and let you know if I see it again. But I used to reproduce it often and now it has been running for an hr or so and nothing.

timg236, 2021-02-02:
Marking as resolved in 2021-01-16 release

closed 2021-02-02 by timg236