raspberrypi/rpi-eeprom #286: bootloader does not blink LED

ajd4096, 2021-01-22:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model
Pi4 8GB

* Board revision (cat /proc/cpuinfo | grep Revision)
Revision        : d03114

* Operating system version .
None

* Details of any hardware attached e.g. links to USB 
None

* Photo of the HDMI diagnostics screen, UART trace.
None

**Describe the bug**
With ONLY power connected and no sdcard, the bootloader should blink 4 times.
This works with pieeprom-2020-04-16.bin but not with any version since.

**To Reproduce**
Steps to reproduce the behavior:
Attach ONLY power, no other cables or sdcard.

**Expected behaviour**
A clear and concise description of what you expected to happen.

The LED should blink 4 times as described in the documentation (https://www.raspberrypi.org/documentation/configuration/led_blink_warnings.md) and the forum troubleshooting steps (https://www.raspberrypi.org/forums/viewtopic.php?p=437084)

timg236, 2021-01-22:
It’s working as intended because it will boot once an SD card is inserted. Blocking to display an error pattern would just delay advancing to the next bootable media.

closed 2021-01-22 by timg236

pelwell, 2021-01-22:
If you want it to flash the LED without an SD card then you need to set the configuration to not retry, i.e. remove the `f` in the `BOOT_ORDER` property,

timg236, 2021-01-22:
Thanks @pelwell for the better explanation  :)

BOOT_ORDER=0x1 will flash the LED if no firmware is found and then wait for card-detect to indicate that a card is inserted. That's been somewhat by the 0xf restart behaviour where the bootloader continuously scans USB/SD for bootable media to appear since it's unlikely that anyone will want to force a power cycle in order to restart boot.