raspberrypi/rpi-eeprom #292: Raspberry pi 4B 4GB won't boot

casperboi, 2021-02-07:
I have a raspberry pi 4B 4GB version board and I was trying to get it up and running. It worked for like 2 times and while I was upgrading the system via SSH (as I couldn't afford to buy a display) from my windows computer, the connection closed accidentally and the pi stopped booting up.

The only indication it gives when I try to boot it up with a 16GB SanDisk micro SDHC, is the ACT light blinks rapidly for exactly 42 times and then 4 blinks with a long pause (LED is ON).

Without the SDHC card there is no rapid blinking part, but the 4 blinks with a pause. (So I think there is some indication that the SDHC card is detected by the pi.)

No other devices are connected to the Pi. I used Raspberry pi imager to flash the SDHC. I have tried with another SDHC but the problem is still there. I have tried all the buster versions of the RaspiOS. 

I tried to upgrade the EEPROM with the latest version using the SD card and it gives me the "File signature/hash mismatch" error. Then I tried with the 2019-05-10 firmware version and it flashes correctly.

 I'm using 5V 3A adaptor. I have tried everything in the The Boot Problems Sticky and still no luck.

timg236, 2021-02-07:
"File signature/hash mismatch" error. Then I tried with the 2019-05-10 firmware version and it flashes correctly."

It might be worth trying pieeprom-2021-01-16.bin if you have a USB serial cable for better diagnostics but since this is reproducible on multiple SDHC cards then it sounds like a hardware problem and the next step is to contact the reseller.

casperboi, 2021-02-08:
First of all thank you taking time to reply. I downloaded the pieeprom-2021-01-16.bin and the recovery.bin file from this link [https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware/beta], copied them to the SD card and tried to flash the EEPROM. The ACT LED blinks once and then stays off. I'm sorry that I can't provide you with enough details to diagnose the problem as I don't have a display or a USB serial cable. 
Contacting the reseller won't help much as I don't have a warranty for the Pi. Guess I'll have to throw it away then. 😔

timg236, 2021-02-08:
It might also be worth swapping the power supply to rule that out

timg236, 2021-02-15:
Closed - hardware failure

closed 2021-02-15 by timg236