raspberrypi/rpi-eeprom #294: Can't firmware boot from ESP fat partition.

Ae3NerdGod, 2021-02-14:
I'm working on making a "uefi mode," using the acpi uefi firmware, for PINN multiboot bootloader/os installer. 

PINN does not support GPT, uses a specific partition scheme with an extended partition. 

I tried making the first partition, the partition on which PINN resides, an ESP partition, to no avail. It froze at the eeprom boot screen. 

it might simplify things for the future of UEFI installations if the pi4 could read a type 0xEF FAT partition (on MBR) and boot a start.elf from it. It would also allow for the uefi firmware to reside in ESP for non PINN-installs

Tl;dr:
Froze during eeprom boot. Afaik, EFI System Partitions (ESP) are still fat32 partitions. Would like to be able to boot the pi4 off one if possible.

lurch, 2021-02-14:
Duplicate of #126 , which is apparently fixed? (Maybe the version of PINN you're using simply doesn't have a new enough start4.elf ?)

Ae3NerdGod, 2021-02-15:
hmm. yea according to that it was fixed... pinn loads fine on fat32 so im not sure the start.elf being old is the problem, but ill try updating it?

although i believe theyre talking about GPT, so ill test GPT ESP vs MBR ESP and report back.
could be a regression? 

ill confirm pinn isnt the problem by using raspberry lite or something too

Ae3NerdGod, 2021-02-15:
i retried the setup that made me make this post-- i think i was wrong about the actual problem:

i made a pinn  _ENTRY_  that makes an ESP and that entry failed to boot. I just assumed that it wouldnt boot ANY ESP

i think what may be happening is once PINN informs the gpu to reboot and boot from a different partition number (as i understand thats how it works, correct me if im wrong) its failing to find the extended ESP and is trying to boot from "partition number: 0" i just noticed. It should be trying to boot partition 6. 

not sure if its PINN failing to get the correct number of the ESP or the EEPROM has a different 'sub-routine' for booting a partition other than the first one, and its tripping itself up on a logical ESP rather than a primary (or just an ESP via the sub-routine at all)

to re-iterate NOT GPT; MBR. just to make perfectly clear

i realized i havent tested

Ae3NerdGod, 2021-02-15:
so idk if its that it cant boot from a ESP that isnt the first one on the disk, or if its the fact that its inside an extended partition, or if pinn just isnt counting ESP as a partition and telling the pi to boot from partition 0, but thats what the eeprom is telling me its trying to do. 

if someone knows the command so i can manually tell the pi which partition to boot from next reboot, i can confirm whether the issue is with PINN or the EEPROM. i was unable to find the command it just leads to a lot of people needing to change their commandline.txt `root=` variable, which is not what i need.

procount, 2021-02-15:
In PINN, `rebootp N` at the cmdline will boot to partition N.

timg236, 2021-02-15:
NOOBS uses extended partitions and rpi-eeprom works with that. The "reboot N" partition mechanism is exactly the same

Ae3NerdGod, 2021-02-15:
@timg236 the issue seems to be with specifically with logical ESP (fat with 0xEF type) not extended partitions itself, nor ESP partitions. sorry if i wasnt clear, i wasnt even fully aware of the actual problem. my bad. 

i never removed that quirk setting from the command line so it should have. however i have found a workaround and gathering the information required will take longer than tomorrow anyway. so i will close and possibly re-open in the future

closed 2021-02-15 by Ae3NerdGod

timg236, 2021-02-15:
ok thanks