raspberrypi/rpi-eeprom #295: Cannot upgrade eeprom due to hash/signature error

lrenaud, 2021-02-14:
__Background:__
Brand new hardware, nothing tested other than trying to boot an OS and install software. No electronics attached beyond the required display, PSU, and USB keyboard.

**Mandatory information**
* __Model:__ Raspberry Pi 4 Model B Rev 1.4
* __Revision:__ d03114
* __OS Version:__ N/A, tested 2021-01-11-raspios-buster-armhf-lite, 2020-05-27-raspios-buster-lite-armhf, and ubuntu-20.10-preinstalled-desktop-arm64+raspi
* __Hardware Attached__: Mini-HDMI adapter, 3A USB-C CanaKit power supply, 64GB SD Card. All other hardware removed.
* __HDMI Display__: Red Bars
* __HDMI Information when booting without SD Card__: see below
```
Raspberry Pi 4 - 8GB
bootloader: 940f978d Mar 19 2020

 board d03114 8882ea552 dc:a6:32:c2:bb:ce
  boot: mode 6 order 0x00000001 rsts 0x00001000
SD CID:
  part: 0 mbr [0x00:00000000 0x00:00000000 0x00:00000000 0x00:00000000]
    fw:
   net: down ip: 0.0.0.0 sn: 0.0.0.0 gw:0.0.0.0
  tftp: 0.0.0.0
```

**Describe the bug**
The system is unable to reboot following any attempted (automatic) EEPROM upgrade.

__Steps to reproduce the behavior:__
When booting any operating system tested so far, the system starts up normally on the first boot. As part of the boot process an updated `pieeprom.upd` and `pieeprom.sig` are placed into the `/boot` partition of the SD card. Manually verification `sha256sum /boot/pieeprom.upd` matches the ASCII contents of `pieeprom.sig`.

When rebooting the HDMI display shows a pattern of dark/light red bars. The LED blinks 2-long, 4 short.

Inserting the SD card into a PC, deleting the two `pieeprom.*` files renders the system able to boot again. The EEPROM is never upgraded.

**Expected behaviour**
The system should reboot showing the "rinibow" pattern, and the EEPROM should be upgraded to the automatically selected version.

**Bootloader version and configuration**
```
$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0X1
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0

```
```
$ vcgencmd bootloader_version
Mar 19 2020 14:27:25
version 940f978d13e45be9baef77f3f4a13b76a832f7b4 (release)
timestamp 1584628045
```

**SD card boot (please complete the following information):**
 - 64GB Samsung Evo Select
```$ sudo fdisk -l /dev/sde # from desktop via SD-Card reader
Disk /dev/sde: 59.64 GiB, 64021856256 bytes, 125042688 sectors
Disk model: USB3.0 CRW   -SD
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x1e1f6308

Device     Boot  Start       End   Sectors  Size Id Type
/dev/sde1         8192    532479    524288  256M  c W95 FAT32 (LBA)
/dev/sde2       532480 125042687 124510208 59.4G 83 Linux
```

**Network boot (please complete the following information):**
N/A

**USB boot (please complete the following information):**
N/A (not tested, not supported by this firmware version anyway?)

The [Bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

timg236, 2021-02-15:
That indicates a hardware failure with either the Raspberry Pi or SD card

lrenaud, 2021-02-15:
Damn. I verified with a second SD card from a different product line to the same effect. Firmly a hardware failure or serious software bug of some sort. I'll start looking into an RMA.

timg236, 2021-03-10:
Assumed hardware fault  closing bug

timg236, 2021-03-10:
Thanks for confirming

closed 2021-03-10 by timg236