raspberrypi/rpi-eeprom #303: rpi-eeprom-update is broken with busybox since switch from strings to tr

HiassofT, 2021-02-25:
Running latest git of rpi-eeprom-update on LibreELEC master is broken, we are using busybox and it's tr command doesn't support the '[:print:]` class - see https://git.busybox.net/busybox/tree/coreutils/tr.c

This results in the following output:
```
LibreELEC:~ # rpi-eeprom-update 
BCM2711 detected
Dedicated VL805 EEPROM detected
sh: bad number
sh: bad number
Checking for updates in /usr/lib/kernel-overlays/base/lib/firmware/raspberrypi/bootloader/default
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: up-to-date
CURRENT: Tue Feb 16 13:23:36 UTC 2021 (1613481816)
date: invalid date '@'
 LATEST:  ()
RELEASE: default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
LibreELEC:~ # tr --help
BusyBox v1.32.1 (2021-02-21 16:55:33 CET) multi-call binary.

Usage: tr [-cds] STRING1 [STRING2]

Translate, squeeze, or delete characters from stdin, writing to stdout

	-c	Take complement of STRING1
	-d	Delete input characters coded STRING1
	-s	Squeeze multiple output characters of STRING2 into one character
```

More specifically:
```
LibreELEC:~ # strings /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2021-02-16.bin | grep BUILD_TIMESTAMP | sed 's/.*=//g'
1613481816
LibreELEC:~ # tr -c [:print:] "\n" < /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2021-02-16.bin | grep BUILD_TIMESTAMP
LibreELEC:~ #
```

Please revert the switch to tr, or prefer using "strings" over "tr" if it's available

vianpl, 2021-02-25:
@timg236 This has caused enough trouble already, so feel free to revert the changes. Otherwise, if you prefer, I can try implementing @HiassofT's suggestion.

lurch, 2021-02-25:
Truly goes to show that you can't please all of the people all of the time! :rofl:

timg236, 2021-02-25:
I wonder if this is possible in pure sed ...

lurch, 2021-02-25:
https://unix.stackexchange.com/a/346298 suggests that trying to use `sed` with binary data might have "problems"... (and even if you get it working on RaspiOS, it might run into problems on busybox, just like this `tr` problem we just had?) :shrug:

HiassofT, 2021-02-25:
If you switch to sed please also check if it works with busybox - it's again a very minimal implementation https://git.busybox.net/busybox/tree/editors/sed.c

In LibreELEC we just install strings from the binutils package on the target, that's a ~640k binary and not too bad - haven't checked if the strings applet from busybox might work, too, but 640k isn't too bad so I'm not really worried much about it.

HiassofT, 2021-02-25:
FYI: busybox's `strings` applet seems to work fine, too, so I think it should be fine to go back to the previous version with strings
```LibreELEC:~ # which strings
/usr/bin/strings
LibreELEC:~ # ls -l /usr/bin/strings
lrwxrwxrwx    1 root     root             7 Feb 25 18:57 /usr/bin/strings -> busybox
LibreELEC:~ # strings /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2021-02-16.bin | grep BUILD_TIMESTAMP
BUILD_TIMESTAMP=1613481816
LibreELEC:~ # strings /sys/firmware/devicetree/base/aliases/blconfig
/reserved-memory/nvram@0
LibreELEC:~ # strings /sys/firmware/devicetree/base/aliases/blconfig | od -t x1
0000000 2f 72 65 73 65 72 76 65 64 2d 6d 65 6d 6f 72 79
0000020 2f 6e 76 72 61 6d 40 30 0a
0000031
LibreELEC:~ # strings /lib/firmware/raspberrypi/bootloader/beta/pieeprom-2021-02-16.bin | grep BUILD_TIMESTAMP | od -t x1
0000000 42 55 49 4c 44 5f 54 49 4d 45 53 54 41 4d 50 3d
0000020 31 36 31 33 34 38 31 38 31 36 0a
0000033
```

timg236, 2021-02-26:
I gave up and pushed the reverts.

Solving this problem for the intersection of busybox and coreutils without just a switch that detects busybox or not is a problem for another day!

timg236, 2021-02-26:
Given that this package depends on python for rpi-eeprom-config I think it could be implemented as a minimal Python script or perhaps part of rpi-eeprom-config

e.g. rpi-eeprom-config --eeprom-version pieeprom-2021-02-16

However, I'll wait until the dust has settled first

HiassofT, 2021-02-26:
thanks, rpi-eeprom-update works fine again

closed 2021-02-26 by HiassofT