raspberrypi/rpi-eeprom #305: FEATURE REQUEST : Change boot order without SD Card

pquiring, 2021-02-28:
Just a feature request.  Would be nice to change boot order without requiring the rpi-eeprom-config tool or any SD Card.  
Maybe a very simple settings screen similar to those found in PC BIOS systems.  
Something to think about for Pi5 ?
Enabling Boot Order 0x2 (network booting) is a bit complicated.

Thanks,

timg236, 2021-03-01:
You can do this via raspi-config. There are no plans to build pc bios screen

closed 2021-03-01 by timg236

pquiring, 2021-03-06:
Sure, but rasp-config requires booting a full OS from the SD Card.
Don't need a pc bios screen. maybe just some hotkeys during bootup to toggle boot order.
CTRL+N = toggle net boot
CTRL+S = toggle SD boot
CTRL+U = toggle USB boot
I plan to run many PIs with networking booting but configuring each one with raspi-config would be cumbersome.
Just a suggestion ;)

Thanks

lurch, 2021-03-06:
If you've got a fixed eeprom config you want to flash to multiple Pis, you can use the `recovery.bin` mode to do that. Much faster (it only takes a few seconds) than booting a full OS :slightly_smiling_face: 
See the docs here https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md