raspberrypi/rpi-eeprom #313: Recognize hidden FAT partition (0x1c) as a valid boot

tianyuanhao, 2021-03-15:
**Mandatory information**
* *Raspberry Pi 4 Model B Rev 1.2*

**Describe the bug**
Cannot boot from hidden FAT partition (0x1c).

**To Reproduce**
Just power on the machine.

**Expected behaviour**
Can boot from hidden FAT partition (0x1c).

**Bootloader version and configuration**
`vcgencmd` is not available.

**SD card boot (please complete the following information):**
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.
```
Device      Boot  Start    End Sectors  Size Id Type
sdcard.img1        2048 264191  262144  128M 1c Hidden W95 FAT32 (LBA)
```

timg236, 2021-03-15:
Please can you explain why this is useful? AFAIK Raspberry PI has never supported this

tianyuanhao, 2021-03-15:
This is only used to prevent the partition from being automatically mounted on Windows or Linux. It is no different from a normal FAT partition (0xc), I think it is very simple to support it. If it is really difficult, I will use the EFI partition (0xef) instead.

timg236, 2021-03-15:
The ROM won't recognize this so recovery.bin won't work and I suspect that start.elf won't recognize this partition id either.

tianyuanhao, 2021-03-15:
@timg236 Did you mean `0xc` or `0xef`? Currently I am using `0xef` and it works. In fact, this issue is a feature request.

tianyuanhao, 2021-03-15:
If you think this feature is really not worth implementing, I will close it.

timg236, 2021-03-15:
I meant 0x1c.

Please leave the issue open and add any relevant detail about hiding the boot partition from Windows. It's useful to have this information there in-case there is a broader requirement here. I can't commit to implementing this feature or not but if it is then this issue will get updated.
Unfortunately, features that also require start.elf to be updated take a long time to become available becuase start.elf then has to be updated in rpi-update, apt, Raspberry Pi OS and other distros. The rpi-update to APT stage is necessarily slow because it depends all the other FW changes.

timg236, 2021-10-11:
Closed due to inactivity

closed 2021-10-11 by timg236