raspberrypi/rpi-eeprom #316: RPI4 8GB problem with boot JMicron JMS578

pepsonEL, 2021-03-18:
Hi. I have a problem that SSD mSATA disk in case  with chip JMicron JMS578 no boot on RPI4 8GB. EERpom in RPI4 is from 17.03.2021. I also change Firmware in my case.... I try all which i found in network. But nothing help.... ALso try solution in this thread but also nothing..... Solution: with the command "sudo -E rpi-eeprom-config --edit" you edit the EEPROM BOOTLOADER then you the next configuration:
USB_MSD_PWR_OFF_TIME=0

https://github.com/raspberrypi/rpi-eeprom/issues/266

When try boot led on case blink and then stop blinking.... and nothing happen.... please help me.... is any chance to solve this problem..... ?

timg236, 2021-03-18:
Use the stable release instead of the beta

pepsonEL, 2021-03-18:
I try also on STABLE. And also not working.

pepsonEL, 2021-03-18:
lsusb
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

timg236, 2021-03-18:
Please try Raspberry Pi forums for issues with JMicron USB devices. It's likely that you need to set the quirks to disable UAS.

Installing the latest Raspberry Pi OS via the Raspberry Pi Imager will give you a known starting point. As a debug step if you can try moving the start.elf files to a subdirectory. This will prevent booting but show whether the bootloader is able to access the device. From your description the blank screen and LED activity indicates that the bootloader has loaded start.elf but it's stuck un Linux

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

pepsonEL, 2021-03-18:
i added it to cmdline usb-storage.quirks=152d:0578:u but nothing..... I try run Home Assistant operating system but also try run RaspberryOS lite and the same problem.

timg236, 2021-03-18:
Please can you add USB serial logs showing the point at which boot stops - see bootloader_config OR a screenshot indicating the HDMI diagnostics failure  - see previous comment

pepsonEL, 2021-03-18:
Can you tell me how read USB serial logs....

timg236, 2021-03-18:
BOOT_UART https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

timg236, 2021-03-18:
Btw Why do people always just mention the chipset and not link to the actual product, power supplies etc are important!

pepsonEL, 2021-03-18:
`
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:d6d82cf9 DATE: Feb 16 2021 TIME: 13:23:40 BOOTMO                                                                       DE: 0x00000006 part: 0 BUILD_TIMESTAMP=1613481816 0x49221560 0x00d03114 0x000a5d                                                                       af
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
PCI reset
PCI reset
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order f4
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Retry SD 1
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
EMMC
SD retry 1 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD retry 2 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SDV1
SD retry 3 oc 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (                                                                       1250) status: 0x1fff0000 delay: 1080
SD CMD: 0x371a0010 (55) 0x0 0x1fff0001
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Boot mode: USB-MSD (04) order f
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB3[2] 000002b1 connected
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3[2] 00281203 connected enabled
USB3 root HUB port 2 init
DEV [02:00] 3.00 000000:02 class 0 VID 152d PID 0578
MSD device [02:00] 3.00 000000:02 conf 0 iface 0 ep 81#1024 02#1024
MSD [02:00] 3.00 000000:02 register MSD
MSD [02:00] 3.00 000000:02 LUN 0
MSD INQUIRY [02:00] 3.00 000000:02
MSD [02:00] 3.00 000000:02 lun 0 block-count 64 block-size 512
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
Unable to read partition as FAT
lba: 0 oem: '' volume: '            '
rsc 0 fat-sectors 0 c-count 0 c-size 0 r-dir 0 r-sec 0
PM_RSTS: 0x00001000
Trying partition: 0
Unable to read partition as FAT
lba: 0 oem: '' volume: '            '
rsc 0 fat-sectors 0 c-count 0 c-size 0 r-dir 0 r-sec 0

`

timg236, 2021-03-19:
What does it show on the HDMI display?

pepsonEL, 2021-03-19:
[image]

timg236, 2021-03-19:
It looks like the disk imaging process did not complete because the first sector is blank

lurch, 2021-03-19:
For some reason the bootloader thinks your [MBR](https://en.wikipedia.org/wiki/Master_boot_record) is all-zeroes, which obviously isn't right!
```
MBR: 0x00000000, 0 type: 0x00
MBR: 0x00000000, 0 type: 0x00
MBR: 0x00000000, 0 type: 0x00
MBR: 0x00000000, 0 type: 0x00
```
Are you able to see the drive from within Raspberry Pi OS when you boot from a SD-card? If so, what does `sudo fdisk -l` say?

pepsonEL, 2021-03-19:
After run from SD Card then also not see this disk.
`pi@raspberrypi:~ $ sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 14.9 GiB, 16003891200 bytes, 31257600 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x7d17f5ac

Device         Boot  Start      End  Sectors  Size Id Type
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 31257599 30725120 14.7G 83 Linux

Disk /dev/sda: 32 KiB, 32768 bytes, 64 sectors
Disk model: Generic
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
pi@raspberrypi:~ $
`

pepsonEL, 2021-03-19:
Other disk mSATA in this case work ok. But only this ADATA SX300 256GB not working....

pepsonEL, 2021-03-19:
But also try on adapter with chip ASMT 2115 an also on this not working.

timg236, 2021-03-19:
It could be power related, try connecting via the USB 2.0 port. to see if you can get it to work at all. If you have a powered USB 3.0 port that might help.

pepsonEL, 2021-03-19:
Try on USB 2.0 and the same.... MBR show is 0.... But other ssd show correct. Probably SSD ADATA SX300 is no support.

timg236, 2021-03-19:
It might be worth posting something on the forums about the SSD ADATA SX300. The same seems familiar but maybe they had a different adapter or had a hub

676077263, 2021-04-19:
我也遇到相同的问题，JMS578主控的芯片一直存在这种问题，以前以为是eeprom还未更新到usb3.0，后又以为是未更新到nvme，目前更新到nvme了，问题依然存在，甚至更严重了。
未刷usb3.0时的eeprom：
usb3.0=>不能引导启动（用TF卡引导间接也不能启动）+作为普通磁盘时也不稳定；
usb2.0=>能引导启动=>作为普通磁盘也稳定；
刷usb3.0的eeprom，没有效果，状态同上。
刷nvme的eeprom后：
usb3.0和usb2.0均不能引导启动，作为普通磁盘时，3.0不稳定，2.0正常。

JMS578在其它操作系统，其它计算机硬件表现良好，请求树莓派4B兼容JMS578
--------------------------------------------------------------------------------------------------------------------
I also encounter the same problem. The chip of jms578 master always has this problem. Previously, I thought that EEPROM has not been updated to USB3.0. Later, I thought that it has not been updated to nvme. Now that it has been updated to nvme, the problem still exists, even more serious.

EEPROM without USB 3.0:

USB3.0 = > can't boot (can't boot indirectly with TF Card) + it's not stable as a normal disk;

USB 2.0 = > bootable = > stable as a normal disk;

Brush USB 3.0 EEPROM, no effect, status as above.

After brushing the EEPROM of nvme:

Both USB 3.0 and USB 2.0 can't boot. As a normal disk, 3.0 is unstable and 2.0 is normal.

Jms578 performs well in other operating systems and other computer hardware. Request raspberry pie 4B to be compatible with jms578

timg236, 2021-04-19:
Removed unrelated comment

timg236, 2021-10-11:
Closed due to inactivity

closed 2021-10-11 by timg236