raspberrypi/rpi-eeprom #328: RPI4 8GB do not boot (SDRAM failure or File signature/hash mismatch)

stachovski, 2021-04-20:
Hi everyone! 

Before I start, I'd like to say that I've looked the foruns and The Boot Problems Sticky (https://www.raspberrypi.org/forums/viewtopic.php?p=437084) and didn't find any solutions (maybe it's my mistake and I apologize in advace).

I have a RPi 4B 8GB and been experiencing some headaches. After a power outage the board simply does not boot anymore and gives me a vertically stripped screen:

[image]

Every time I try to update de RPIEEPROM I get an error message:

If I try to use the Imager version I get an File signature/hash mismatch - Pi 4 message (2 long and 4 short flashes of the green light)

If I try to use the latest release (https://github.com/raspberrypi/rpi-eeprom/releases/tag/v2021.03.18-138a1) it gives me a red screen and SDRAM failure message (8 short green light flashes)

[image]

I'm using differente SD cards. Multilaser (32Gb), Sandisk (4 and 16Gb). All of them are class ten and brand new.

I hope you can help me. Thanks

timg236, 2021-04-20:
Unfortunately, it sounds like the power outage has damaged either the PMIC or SDRAM. It would be worth unplugging all accessories and trying a different SD-card or the and the 2020-09-03 bootloader version (just in-case) but the red-screen indicates that the ROM was able to read the bootloader into L2 cache but the SDRAM was returning errors.

stachovski, 2021-04-20:
I had an closed SD just in-case you said that. The situation stays the same.
I tried many configurations. 

- Used the recommended ROM OS from imager and put the eeprom recovery files from the 2020-09-03 bootloader version with it. (here I get the stripped screen)
- Used the recommended ROM OS from imager and put the eeprom recovery files from the 2020-09-03 bootloader version with it and renamed pieeprom.bin to pieeprom.upd. (here I get the stripped screen)
- Did the same things with the 2021-03-18 bootloader version. (here I get the red screen)
- Used the pieeprom from the 2020-09-03 version with the recovery.bin file from the 2021-03-18 version. (here I get the red screen)

The only thing I find curios is the different behavior presented as we change the recovery.bin file.

lurch, 2021-04-20:
> The power supply applied directly to the GPIO pins goes through the PMIC?

Yes, see the partial schematics at https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md
AFAIK all the 5V signals are connected together. (and the Pi4 needs many more different voltage levels than just 5V, which is the job of the PMIC)

stachovski, 2021-04-20:
> Yes, see the partial schematics at https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/README.md
> AFAIK all the 5V signals are connected together. (and the Pi4 needs many more [...]

Ok. Thanks.
Still the same situation.

timg236, 2021-04-20:
The reason for the change in behaviour is that the latest recovery.bin now does a very simple SDRAM test at the start of day and if that fails it uses the HVS display a solid red background colour. Previously, it would have got a hash mismatch because SDRAM is returning garbage. The green stripes are displaying what HVS is receiving from SDRAM instead of a solid red frame-buffer.

stachovski, 2021-04-20:
> The reason for the change in behaviour is that the latest recovery.bin now does a very simple SDRAM test at the start of day and if that fails it uses the HVS display a solid red background colour. [...]

Well, in other terms: I'm screwed.

Thanks for your responses.

closed 2021-04-20 by stachovski