raspberrypi/rpi-eeprom #330: Low power mode actually increases power consumption after `sudo halt` !?

seamusdemora, 2021-04-23:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model: RPi 4B
* Board revision (cat /proc/cpuinfo | grep Revision): `Revision	: b03111`
* Operating system version . `Raspbian GNU/Linux 10 (buster)`
* Details of any hardware attached e.g. links to USB : none while testing this
* Photo of the HDMI diagnostics screen, UART trace. N/A - running "Lite"/Headless

**Describe the bug**
A clear and concise description of what the bug is.
* Configure for "low power mode"; `WAKE_ON_GPIO=0`, `POWER_OFF_ON_HALT=1` 
* While operating, RPi draws 500mA. After `sudo halt`, RPi draws 630mA
* Previous testing in July 2020 w/ `2020-06-15` firmware reduced current from 500mA to 40mA

**To Reproduce**
Steps to reproduce the behavior: 

1. `vcgencmd bootloader_config` reports
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

2. `$ vcgencmd bootloader_version` reports:
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

3, Measure current into RPi
4. `sudo halt`
5. Measure current into RPi

**Expected behaviour**
A clear and concise description of what you expected to happen.

Current consumption after `sudo halt` should have decreased from 500mA to 40mA - as it did w/ `2020-06-15` firmware

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration vcgencmd bootloader_config and version (vcgencmd bootloader_version)

See above

**SD card boot (please complete the following information):**
 - SD card type:  32GB SanDisk
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.

**Network boot (please complete the following information):**
Network boot bug normally require one or more of the following log types. [PiServer](https://github.com/raspberrypi/piserver) is the officially supported network boot server.

 - DHCP server configuration files e.g. dnsmasq.conf
 - Wireshark binary packet capture
 - UART logs

N/A - boot from SD card

**USB boot (please complete the following information):**
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'

**Additional context**
Add any other context about the problem here. 

The [Bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

seamusdemora, 2021-04-23:
I have tried to restore the EEPROM version from Jun, 2020 - the version that yielded a reduction FROM: 500mA TO: 40mA. 

It now seems to yield the same results as the Sep, 2020 EEPROM version: FROM: 510mA  TO: 620mA 

A detailed log of this experiment follows: 

```
BEGIN RESTORATION OF 2020-6-15 EEPROM BINARY

SUSPEND AUTOMATIC UPDATES in systemd

pi@raspberrypi4b:~ $ sudo systemctl mask rpi-eeprom-update
Created symlink /etc/systemd/system/rpi-eeprom-update.service → /dev/null.

VERIFY THE 2020-6-15 EEPROM binary is configured for LOW POWER

pi@raspberrypi4b:~ $ cd eeprom_changes
pi@raspberrypi4b:~/eeprom_changes $ ls -l
total 1032
-rw-r--r-- 1 pi pi    172 Jul 12  2020 bootconf.txt
-rw-r--r-- 1 pi pi    172 Jul 14  2020 boot_lowpwr.txt
-rw-r--r-- 1 pi pi 524288 Jul 12  2020 pieeprom.bin
-rw-r--r-- 1 pi pi 524288 Jul 14  2020 pieeprom-lopwr.bin
pi@raspberrypi4b:~/eeprom_changes $ rpi-eeprom-config pieeprom-lopwr.bin
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41

VERIFIED: pieeprom-lopwr.bin HAS WAKE_ON_GPIO and POWER_OFF_ON_HALT properly set for low power after halt

INSTALL THE pieeprom-lopwr.bin EEPROM image

pi@raspberrypi4b:~/eeprom_changes $ sudo rpi-eeprom-update -d -f pieeprom-lopwr.bin
*** INSTALLING pieeprom-lopwr.bin  ***
BOOTFS /boot
EEPROM update pending. Please reboot to apply the update.
pi@raspberrypi4b:~/eeprom_changes $ vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000

REBOOT THE SYSTEM TO INSTALL pieeprom-lopwr.bin

pi@raspberrypi4b:~/eeprom_changes $ sudo reboot
Connection to raspberrypi4b.local closed by remote host.
Connection to raspberrypi4b.local closed.

seamus@Dung-Pro ~ % ssh pi@raspberrypi4b.local
Linux raspberrypi4b 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l

You have new mail.
Last login: Fri Apr 23 16:15:39 2021 from 192.168.1.178

VERIFY THAT pieeprom-lopwr.bin IS INSTALLED

pi@raspberrypi4b:~ $ vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
update-time 0
capabilities 0x00000000

HALT RPi AND CHECK CURRENT DRAW

pi@raspberrypi4b:~ $ sudo halt
Connection to raspberrypi4b.local closed by remote host.
Connection to raspberrypi4b.local closed.

RESULTS:
BEFORE HALT: 510mA
AFTER HALT: 620mA

seamus@Dung-Pro ~ % ssh pi@raspberrypi4b.local
Linux raspberrypi4b 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l

You have new mail.
Last login: Fri Apr 23 17:20:53 2021 from 192.168.1.178

VERIFY THAT pieeprom-lopwr.bin IS STILL INSTALLED

pi@raspberrypi4b:~ $ vcgencmd bootloader_version
Jun 15 2020 14:36:19
version c302dea096cc79f102cec12aeeb51abf392bd781 (release)
timestamp 1592228179
update-time 0
capabilities 0x00000000
pi@raspberrypi4b:~ $
``` 

I am tempted to conclude that that this behavior (increased power consumption after `halt`) is not caused by any changes to the EEPROM binary, but there's no way for me to know that - so I'll conclude nothing at all. It does cause me to wonder if the explanation lies somewhere in `/boot/config.txt`, and following is a list of all ***un-commented*** lines that may provide some clues: 

>\# Enable audio (loads snd_bcm2835)
\# seamus edit for BT sound experiments - disable/cmt next line:
dtparam=audio=on

>\# experiment with signal on GPIO 26 at `shutdown`
dtoverlay=gpio-poweroff

>[pi4]
\# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

>[all]
\#dtoverlay=vc4-fkms-v3d

NOTE: The items under `[pi4]` seem odd as I am running the "Lite" distro ("headless").

seamusdemora, 2021-04-24:
### A discovery: 

Commenting out the `dtoverlay=gpio-poweroff` in `/boot/config.txt` reduces the power consumption after `halt` to the 40mA figure I got in previous experiments. 

I've seen nothing in the documentation suggesting why this dtoverlay should upend the low power configuration of `WAKE_ON_GPIO=0` and `POWER_OFF_ON_HALT=1` in the EEPROM configuration. I actually need and use the `dtoverlay=gpio-poweroff` as a signal to remove all power (via external circuitry) from the RPi, and put it in a "Zero Power" mode. Here's a [screen shot of the signal produced on `GPIO 26` after executing a `halt`](https://imgur.com/a/AE06PCG).

I guess this changes the bug report, but I don't know if or where the responsibility is shifted. Please advise.

pelwell, 2021-04-24:
The whole point of the gpio-poweroff driver and overlay is to get the ARMs to a point where the power can be safely removed then signal an external agent to do so. If you aren't actually removing the power at that point then you are breaking the contract.

seamusdemora, 2021-04-24:
?? I do not understand your point. Who said I wasn't removing power... how did you get that out of what I said? and "Contract"?! wow.

timg236, 2021-04-24:
I've just tried this on a Pi4B 0xc03111 with the latest 2021-03-08 and it worked as expected i.e.

* sudo halt
* Linux shutdown
* Activity LED displays HALT pattern
* Measure 3V3 - off

Please can you verify that you see the last two steps, if not, then I suspect the Linux halt sequence did not complete. Bootloader UART traces will also indicate whether the PMIC outputs are being turned off.

seamusdemora, 2021-04-24:
@timg236 : Thanks for the feedback. Did you have the `dtoverlay=gpio-poweroff` set when you tried this (ref "Discovery" comment above) ? 

I'll check the green LED & take a volt measurement on 3V3 bus & report findings.

lurch, 2021-04-24:
@seamusdemora I think by "breaking the contract" Phil was implying that the _expectation_ is that if you're using gpio-poweroff then you'll be removing the external power to the Pi after GPIO26 goes high, in which case the "high power consumption" shouldn't matter because you've removed the power-supply anyway? :shrug:

pelwell, 2021-04-24:
The contract is with the kernel - failure to remove the power in the allowed time (3 seconds, I think) will result in a kernel panic.

This usage pattern is erroneous, and the issue should be closed.

closed 2021-04-24 by timg236

seamusdemora, 2021-04-25:
@lurch : Yes - I imagined that might be his implication, but that's not what I said. I need both of these to work. In some cases (longer down times & lower battery voltage) I need to go to "Zero Power", and I must have the `gpio-poweroff` overlay to accomplish this. In other cases, I can get by with the 30-40mA draw in "Low Power" which requires WAKE_ON_GPIO and POWER_OFF_ON_HALT variables in EEPROM. 

As it is now, it seems that "Zero Power" and "Low Power" are *mutually exclusive*; i.e. you cannot have both configured in the system at the same time. I'm surprised that this *mutual exclusivity* has not been flagged - or perhaps it has, and I've overlooked it??  

I'm going to run the experiment Tim requested now. I'm not sure he noticed the comment re `gpio-poweroff`, but we'll see soon enough :)

seamusdemora, 2021-04-25:
@pelwell : If this usage pattern is erroneous, would it not make sense to you that it be documented somewhere?

seamusdemora, 2021-04-25:
I have loaded/installed the 2021-03-08 firmware, applied the `dtoverlay=gpio-poweroff`, and run the test suggested by @timg236. I got the same result as before: _RPi power consumption **increased** by 25%_. 

The maintainers' posts state there exists a ["contract"](https://www.merriam-webster.com/dictionary/contract) with the kernel that specifies the `gpio-poweroff` overlay should not be set when the "low power" options are set in the EEPROM bootcode. 

I understand this to mean: As now structured, the "low power" option in the RPi 4B system's bootcode is not supported if the `dtoverly=gpio-poweroff` option is used. The net effect seems to be that one may configure "low power" option in bootcode OR the `dtoverly=gpio-poweroff` option - but not both.

pelwell, 2021-04-25:
The standard Linux gpio-poweroff driver prevents a poweroff reset - this is by design. You are free to enable the hard poweroff in the boot configuration, but it will never be used because the board will only reset to reboot, at which point you don't want it to switch off.

seamusdemora, 2021-04-25:
Well that certainly clears everything up.

pelwell, 2021-04-26:
Here's a preview of the updated gpio-poweroff documentation:
```
Info:   Drives a GPIO high or low on poweroff (including halt). Using this
        overlay interferes with the normal power-down sequence, preventing the
        kernel from resetting the SoC (a necessary step in a normal power-off
        or reboot). This also disables the ability to trigger a boot by driving
        GPIO3 low.

        Users of this overlay are required to provide an external mechanism to
        switch off the power supply when signalled - failure to do so results
        in a kernel BUG, increased power consumption and undefined behaviour.
```

seamusdemora, 2021-04-26:
That's pretty good. It would certainly be helpful to briefly mention that if enabled, it interferes with low power configuration in the EEPROM bootcode. The "low power configuration" I'm referring to is of course the "`WAKE_ON_GPIO=0` & `POWER_OFF_ON_HALT=1`" configuration. This could certainly save someone from wasted time & effort filing a "NOT A BUG" report. Those of us looking to use the RPi in off-grid applications are the most likely ones to be experimenting with these settings.

lurch, 2021-04-26:
@pelwell Is it possible to "remove" or "disable" the effect of the `gpio-poweroff` dtoverlay at run-time? Or is that only possible by editing `/boot/config.txt` and rebooting?
If it _is_ possible at run-time, I wonder if that might fulfil Seamus's desire to support both the "Zero Power" and "Low Power" modes that he refers to? (although I have to admit that wanting to support both at the same time seems like somewhat of an edge-case, and it seems that most applications would only use one or the other.)

pelwell, 2021-04-27:
> Is it possible to "remove" or "disable" the effect of the gpio-poweroff dtoverlay at run-time?

No, that's not possible - installing the gpio_poweroff driver disables the usual poweroff handler, but removing it doesn't restore it, leaving the device with no way to power off normally.

seamusdemora, 2021-04-27:
I've been curious about something for a while, and this seems a good time to ask about it: 

A comment was made in the forums some time ago that the "Low Power mode" (WAKE_ON_GPIO=0 & POWER_OFF_ON_HALT=1) would reduce current draw to 3-4 mA. I was disappointed when it turned out to be 40mA, and was told by an "outsider" that this was due to the necessity to keep the 5V bus up because some HATs needed it for some reason. 

If all of that's true, then perhaps another bootcode config (5VBUS_OFF ?) variable would solve my problem. I can live with a few mA - I can't live with 40mA.

pelwell, 2021-04-27:
> removing it doesn't restore it, leaving the device with no way to power off normally.

This doesn't have to be the case, though. Although gpio-poweroff is a common Linux driver, we are already using a patched version to allow it to be installed on a system that already has a poweroff handler.

The updated version proposed in https://github.com/raspberrypi/linux/pull/4309 remembers the original poweroff handler for two reasons:
1. so that it can be restored when the module is unloaded, and
2. as a fallback in the event that the power is not withdrawn before the timeout expires.

lurch, 2021-04-27:
Nice, it sounds like 2. will solve all of Seamus's concerns? (when it eventually filters through to a packaged kernel of course)

lurch, 2021-04-27:
> If all of that's true, then perhaps another bootcode config (5VBUS_OFF ?) variable would solve my problem.

AFAICT from looking at https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf the 5V rail from the USB-C power-input is directly connected to the GPIO header, and therefore it's not possible to "turn off" the 5V bus? :shrug:

seamusdemora, 2021-04-29:
@lurch :
> AFAICT from looking at https://www.raspberrypi.org/documentation/hardware/raspberrypi/schematics/rpi_SCH_4b_4p0_reduced.pdf the 5V rail from the USB-C power-input is directly connected to the GPIO header, and therefore it's not possible to "turn off" the 5V bus?

Perhaps; but two things: 1. that's not a complete schematic, and 2. I'm not sure connection of the 5V input from the USB to the GPIO header is significant to the objective of reducing power. Let me try to explain what was behind my comment: I'm under the impression that the 5V input is fed to a PMIC (maybe that's U2?) where the PMIC logic controls which subsystems get "powered up" & which don't. If all of these subsystems were turned OFF, and the 5V input were connected **only** to the header, we wouldn't see the 40 mA current consumption we currently see. Therefore, the PMIC must be supplying power to some internal RPi circuitry/subsystems. 

In other words - if nothing is connected to the 5V pins in the header, what is drawing 40mA of current in the "Low Power" mode? Perhaps what I should have said was, "TURN_OFF_ALL_SYSTEMS_CONNECTED_TO_5VBUS".  :)

seamusdemora, 2021-04-29:
@pelwell : 

I'm sure I am not following you **completely**, but I am cautiously optimistic :) 

Does this mean that your proposed changes to to the `gpio-poweroff` overlay will : a) continue to provide a signal for an external switch to disconnect power from the RPi, AND b) fall back to "Low Power/40mA" mode if the external switch does not remove power?

seamusdemora, 2021-04-29:
@pelwell : 
> a system that already has a poweroff handler.
?? Are you referring to the features in the EEPROM bootcode that reduce power consumption on RPi 4 to ~ 40mA? Or something else?

pelwell, 2021-04-29:
On Raspberry Pis, both power-off and reboot are handled by the kernel triggering a full-chip reset, at which point the VPU runs from the ROM, then the EEPROM. The jump to the reset is arranged by registering a pm_power_off handler, which is the same mechanism used by gpio-poweroff - hence the problem. The enhanced version of the gpio-poweroff driver will fall back to calling the previous pm_power_off handler (i.e. trigger a reset) after the timeout (3s by default). The change is already in the kernel repository, and available using `sudo rpi-update` (with the usual caveats about bleeding-edge software).

lurch, 2021-04-29:
@pelwell Does this change mean that the "increased power consumption and undefined behaviour." text needs removing from the gpio-poweroff documentation?

pelwell, 2021-04-29:
The warning could be removed at some point.

seamusdemora, 2021-04-30:
@pelwell : I'll try this first opportunity... 2 things: 

1. Just so I'm clear on what to expect: 
   * Set "Low Power mode" (WAKE_ON_GPIO=0 & POWER_OFF_ON_HALT=1), AND enable the `gpio_poweroff` overlay.  
   * execute `sudo helt`
   * after a 3 sec delay, Rpi goes to "Low Power mode" (~ 40 mA)

   Is this correct? 

2. Will any of this work on the RPi 3B+ ?

pelwell, 2021-04-30:
1. Yes.
2. On 3B+ (and any other Pi except a 400) you'll get the same halt behaviour after the timeout as you do without the overlay.

lurch, 2021-04-30:
@seamusdemora In case it helps clarify things (and hopefully I don't confuse you even more!), the 2 settings are actually independent:

1. There's the EEPROM "WAKE_ON_GPIO=0 & POWER_OFF_ON_HALT=1" settings, which (on the Pi4B / Pi400 / CM4 only, as they're the only ones that have the bootloader in EEPROM) tells the bootloader what to do when the Pi is "shutdown" - this is what shuts off the PMIC outputs to put the Pi into "Low power mode".
2. There's the pm_power_off setting that tells the Linux kernel what to do when you run "sudo halt" (and so runs the same on all Pis).

Without the `gpio_poweroff` overlay, when you run "sudo halt" the default pm_power_off handler (in the kernel) sets the "please HALT" register, and resets the whole chip. This goes back to running the bootloader again, which then sees that the "please HALT" register is set, and so turns off the PMIC outputs if you've also set "WAKE_ON_GPIO=0 & POWER_OFF_ON_HALT=1" in the EEPROM (see https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow_2711.md where it talks about `PM_RSTS`).

With the "old" `gpio_poweroff` overlay, when you run "sudo halt" it would set the GPIO pin high, wait 3 seconds, and then "result in a kernel BUG", causing the increased power consumption that you saw, and do nothing else. (this is the bit where the kernel was expecting the external power to be removed when the GPIO pin went high)

With the "new" `gpio_poweroff` overlay (that Phil has modified), when you run "sudo halt" it will set the GPIO pin high, wait 3 seconds, and then go back to running the default pm_power_off handler, which sets the "please HALT" register, and resets the whole chip... which runs the bootloader... etc.

(and just for the sake of completeness, when you run "sudo reboot" the kernel resets the whole chip _without_ setting the  "please HALT" register, which causes the bootloader to load your OS, just as it does when you first connect power)

I don't have any "insider info" and I've not looked at any bootloader or Linux kernel source-code (so there might be bits that I've got wrong), this is just my best understanding from reading https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/bootflow_2711.md in combination with what's been written in this issue.

pelwell, 2021-04-30:
The overlay is called `gpio-poweroff` (note the hyphen/dash).