raspberrypi/rpi-eeprom #332: Dock Station for 2.5"/3.5" SATA hard drives to USB 3.0

Cebeerre, 2021-04-27:
**Mandatory information**
* Raspberry Pi model: Raspberry Pi 4 (4 Gb) Rev 1.2
* Board revision (cat /proc/cpuinfo | grep Revision): c03112
* Operating system version: Raspbian GNU/Linux 10 (buster)
* Details of any hardware attached: https://www.tooq.com/product.php?id=1636&lang=2

**Describe the bug**

Raspberry Pi with Raspbian running on top of the SD. I've attached a Dual Dock SATA station ```Prolific Technology, Inc. PL2773 SATAII bridge controller``` with two 1Tb spinning disks. If the dock is plugged in with the OS up and running, both drives are present and clearly seen with a ```lsblk```. After a reboot or power-cycle, only the drive present in Slot "A" gets detected. 

**To Reproduce**

The behaviour described above happens with all this firmwares from the stable branch detailed below:

```
-rw-r--r-- 1 root root 524288 Jan 14 12:47 pieeprom-2021-01-11.bin
-rw-r--r-- 1 root root 524288 Jan 16 16:26 pieeprom-2021-01-16.bin
-rw-r--r-- 1 root root 524288 Feb 22 17:53 pieeprom-2021-02-16.bin
-rw-r--r-- 1 root root 524288 Mar 18 17:01 pieeprom-2021-03-18.bin
```

Going backwards starting from ```pieeprom-2020-12-11.bin``` works as expected and both drives get correctly detected. So, something was introduced when "2021-01-11 - Timeout stalled USB MSD devices - BETA" was promoted to the stable branch on the 2021-01-14.

Right now as a workaround I'm on ```pieeprom-2020-12-11.bin``` and I have disabled the automatic updates.

**Expected behaviour**
To see both drives in the latest release of the stable branch.

**Bootloader version and configuration**

```
   CURRENT: Fri 11 Dec 11:15:17 UTC 2020 (1607685317)
```

```
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
TFTP_IP=
TFTP_PREFIX=0
BOOT_ORDER=0x1
SD_BOOT_MAX_RETRIES=3
NET_BOOT_MAX_RETRIES=5
[none]
FREEZE_VERSION=0
```

lurch, 2021-04-27:
If you've not already done so, try experimenting with the `USB_MSD_PWR_OFF_TIME` option to see if that helps? https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

timg236, 2021-06-11:
Please could you try the latest beta-eeprom and add USB_MSD_STARTUP_DELAY=5000.

This adds a one short delay during booting after the USB is initialised but before device enumeration. This should give HDDs time to spin up. The problem before is that the enclosure will respond immediately but when the bootloader queries the capacity it has to wait for the drivers to initialise. This can take a long time on some enclosures making it look as though the SCSI command has stalled.

https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/pieeprom-2021-06-11.bin

This should be available via APT early next week or you can download the file directly 
```
wget https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/beta/pieeprom-2021-06-11.bin
# Add USB_MSD_STARTUP_DELAY option
sudo rpi-eeprom-config --edit pieeprom-2021-06-11.bin 
```

dingausmwald, 2021-06-14:
In my case this fried my pi. No update possible anymore. Red and green light constantly on

pelwell, 2021-06-14:
@dingausmwald Who are you, and why are you commenting on this issue?

dingausmwald, 2021-06-14:
Just some enthusiast (the worst) trying to get his ssd to boot at 3am in the morning. Got it to work again, guess the update i did just went wrong and the ssd with raspi lite couldnt boot/update = constant green led. Sry for that

timg236, 2021-07-29:
Closed due to inactivity.

closed 2021-07-29 by timg236