raspberrypi/rpi-eeprom #335: HDMI disagnostic screen appears after 5 seconds - should be hidden until 8 seconds have elapsed

andrum99, 2021-05-06:
Raspberry Pi 4 Model B Rev 1.1
Ubuntu 21.04 Desktop (not relevant)

USB devices attached:
174c:1153 Startech.com 2.5" USB 3 to SATA enclosure, part No. S2510BMU33
413c:2003 Dell keyboard SK-8115
093a:2510 official Raspberry Pi mouse

[image]

HDMI diagnostic screen appears only 5 seconds after power applied, even though no error is shown on the screen. I'm booting from the USB MSD.

Steps to reproduce the behavior:

1. Configure EEPROM bootloader as follows:
```
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf14
USB_MSD_DISCOVER_TIMEOUT=5000
USB_MSD_PWR_OFF_TIME=0
```

2. Image a USB MSD with an OS the Pi can boot.

3. Boot from the USB MSD.

**Expected behaviour**
HDMI boot screen is hidden and the first thing the screen displays is the rainbow splash.

**Bootloader version**
```
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
```

*Additional information*
If I leave out `USB_MSD_PWR_OFF_TIME=0`, the bootloader takes longer than 8 seconds to load the OS, and the HDMI diagnostics screen is shown about 9-10 seconds after power is applied, which is the expected behaviour.

timg236, 2021-05-06:
You need to update to the latest version

andrum99, 2021-05-06:
Apologies - this was caused by user error on my part. I have changed the bootloader EEPROM configuration using `rpi-eeprom-config -e` from within Ubuntu 21.04, but it had silently downgraded the bootloader EEPROM to an older release - the latest it had available, which had a shorter setting from `HDMI_DELAY`. The only hint in the output of `rpi-eeprom-config -e` that this was going to happen was where it told be which image it was using, which I had not bothered to check for.

Perhaps `rpi-eeprom-config` could be enhanced to issue a warning in the case where it is going to have the bootloader downgrade itself?

lurch, 2021-05-06:
I guess the only way it could do that would be by comparing timestamps? And that would probably give "false positives" if you're e.g. switching from the beta releases to the stable releases? :shrug:

timg236, 2021-05-06:
"rpi-eeprom-config --edit" displays information about the source image. 

I think it would be reasonable for rpi-

timg236, 2021-06-15:
The latest APT package now includes a version of rpi-eeprom-update now displays the downgrade warning. Feature implemented

closed 2021-06-15 by timg236

andrum99, 2021-06-15:
Just spotted HDMI_DISABLE is not documented so I filed https://github.com/raspberrypi/documentation/pull/1914