raspberrypi/rpi-eeprom #336: CM4 not booting from NVMe SSD anymore

g30ba1, 2021-05-16:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model: Compute Module 4 Rev 1 (4GB RAM, 8GB eMMC, WiFi)
* Board revision (cat /proc/cpuinfo | grep Revision):  c03140
* Operating system version: Raspbian GNU/Linux 10 (buster) armv7l
* Details of any hardware attached e.g. links to USB: CM4 is installed on Raspberry Pi IO Board, originally running/booting from NVMe SEKC SM250 SSD
* Photo of the HDMI diagnostics screen, UART trace.

**Describe the bug**
After updating the CM4 (Running sudo apt-get upgrade) "raspberrypi-kernel" & "raspberrypi-bootloader" the OS is not booting from NMVe anymore. It's booting from eMMC now.

**To Reproduce**
Steps to reproduce the behavior:
1. Follow "NVMe SSD Boot (BETA)" https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/nvme.md
to configure the boot order to 6 (For NVMe)
2. Use the CM4 
3. Run "sudo apt-get update and upgrade" and install new "raspberrypi-kernel" and "raspberrypi-bootloader"

After this, the SO booted will be the one installed on the eMMC, NOT from the NVMe SSD

**Expected behaviour**
I was using the NVMe SSD with NO problems for months, but after updating "raspberrypi-kernel" & "raspberrypi-bootloader", the instructions described on the guide  "NVMe SSD Boot (BETA)" I can´t boot from NVMe SSD.

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration vcgencmd bootloader_config and version (vcgencmd bootloader_version)

[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try  SD -> NVME -> USB PCIe MSD -> USB 2.0 BCM XHCI -> Network ->  Loop
BOOT_ORDER=0xf25416

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

-------------------------------------------------------------------------------------------------------------------------------------------

Mar 17 2021 11:25:36
version 05552182f98adadd27ba1aa9e3ca671c9dd5123f (release)
timestamp 1615980336
update-time 1621195467
capabilities 0x0000007f

**SD card boot (please complete the following information):**
 - SD card type
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.

**Network boot (please complete the following information):**
Network boot bug normally require one or more of the following log types. [PiServer](https://github.com/raspberrypi/piserver) is the officially supported network boot server.

 - DHCP server configuration files e.g. dnsmasq.conf
 - Wireshark binary packet capture
 - UART logs

**USB boot (please complete the following information):**
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'

**Additional context**
The CM4 was booting and running perfect before updating "raspberrypi-kernel" and "raspberrypi-bootloader". I'm using the most recent software for rpi-update, so, if I try to run:

sudo ROOT_PATH=mnt/ext4 BOOT_PATH=mnt/fat32 rpi-update

The message "Your firmware is already up to date" appears and I can´t do anything to recover booting from NVMe.

[image]

[image]

[image]

The [Bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

timg236, 2021-05-17:
Please can you run "strings start4x.elf | grep VC_BUILD" to verify that rpi-update updated the firmware correctly.

lurch, 2021-05-17:
`rpi-update` doesn't "play nicely" with `apt upgrade`, which can cause problems when each of them tries updating the bootloader and/or kernel.
I think if you delete the `.firmware_revision` file (although I can't remember at which exact path it exists) then `rpi-update` will no longer think it's "up to date"? (and that should then get you the kernel/firmware versions you need for NVMe boot?)

timg236, 2021-05-19:
Awaiting confirmation of the firmware version on NVME driver.

g30ba1, 2021-05-23:
> Please can you run "strings start4x.elf | grep VC_BUILD" to verify that rpi-update updated the firmware correctly.

No luck 👇

strings: 'start4x.elf': No such file

peterharperuk, 2021-05-24:
Sorry I missed this. It seems that the official release of the firmware still doesn't include the changes to support NVME.

> if I try to run:
> sudo ROOT_PATH=mnt/ext4 BOOT_PATH=mnt/fat32 rpi-update
> The message "Your firmware is already up to date" appears

Are you sure the nvme drives have been mounted? rpi-update worked for me to fix this problem.

sudo mount /dev/nvme0n1p1 mnt/fat32
sudo mount /dev/nvme0n1p2 mnt/ext4

timg236, 2021-05-24:
d7f29d96450abfc77cd6cf011af1faf1e03e5e56 is the APT stable release does not have NVME support.

NVMe is in the rpi-update master branch and after removing the `.firmware_revision`  rpi-update version from here 
https://github.com/raspberrypi/firmware/tree/master/boot

When APT stable is updated again NVMe will be supported in the default firmware.

timg236, 2021-06-11:
NVMe seems to be working in the latest rpi-update-firmware. There will be an APT release with this in fairly soon so closing this issue.

closed 2021-06-11 by timg236

g30ba1, 2021-06-18:
> d7f29d96450abfc77cd6cf011af1faf1e03e5e56 is the APT stable release does not have NVME support.
>
> NVMe is in the rpi-update master branch and after removing the `.firmware_revision` rpi-update version [...]

Thank you both for taking the time.

If I´m getting this right, do I need to wait until the APT stable release INCLUDES NVMe support again?

I'm thinking on running the rpi-update tool again, but if the updated firmware doesn´t not include NVMe support, trying to recover/fix the BOOT order will be useless, right?

Thank you in advance!

timg236, 2021-06-18:
There haven't been any APT releases yet which support NVMe. It's only available in the rpi-update "dev/beta" release.

The next APT release will be based upon the rpi-update "dev/beta" release and will therefore support NVMe. IIRC that should be available in a few weeks.

lurch, 2021-06-18:
@g30ba1 `rpi-update` has now been updated with information on how to force an update, if it (incorrectly) thinks that the firmware is already up to date :slightly_smiling_face:   (The `rpi-update` script itself automatically self-updates, so you should see this new message the next time you run it)

g30ba1, 2021-06-19:
> @g30ba1 `rpi-update` has now been updated with information on how to force an update, if it (incorrectly) thinks that the firmware is already up to date :slightly_smiling_face:   (The `rpi-update` script itself automatically self-updates, so you should see this new message the next time you run it)

Just to be clear:

All I have to do is:
1. Mount nvme0n1p1 & nvme0n1p2
2. Execute rpi-udate

There's no need to configure BOOT ORDER again, right?

lurch, 2021-06-19:
Yes, the BOOT_ORDER setting is stored inside the eeprom itself, so as long as you haven't re-flashed your eeprom then your old BOOT_ORDER setting should still be in there.