raspberrypi/rpi-eeprom #339: USB xHC init failed

ramoas86, 2021-05-22:
**Mandatory information**

* Raspberry Pi model
Raspberry Pi 4 - 4GB

* Board revision (cat /proc/cpuinfo | grep Revision)
c03111

* Operating system version.
Raspberry Pi OS (32 bits) updated.

* Details of any hardware attached e.g. links to USB
Wired Keyboard and Mouse.

* Photo of the HDMI diagnostics screen, UART trace.
[image]

**Describe the bug**
Helo! I'm having the following problem. When connecting the Raspberry Pi, keyboard and mouse, sometimes, do not work. I've had this problem for months. Accessing the bootloader, I have the messages that are in the attached photo. I am using a common wired mouse and keyboards, I have already tested it with others and the problem persists. When I bought the Raspberry pi, it didn't have that problem. It started to emerge, after updating the Raspberry Pi OS.

In the photo, the bootloader version is from April 2021. I also tested it with the September 2020 version (default) and the same problem happens.

**To Reproduce**
The error can happen when turning on the device. I realized that, generally, when I turn it off at night and turn it on the next day. If I have the SD card, I can't use the keyboard and mouse when starting the operating system. It is necessary turn off and sometimes it will work again. When you plug it in again, without the SD card, the bootloader appears to be normal, but sometimes, after five restarts, the problem happens. Now as I write this text, I am using it normally.

**Expected behaviour**
Turn on and keyboard and mouse working.

**Bootloader version and configuration**
Apr 29 2021 17:11:25
version c2f8c388c4ee37ad709ace403467d163e8dd91ce (release)
timestamp 1619712685
update-time 1621687552
capabilities 0x0000001f

timg236, 2021-05-24:
It's probably worth checking the power supply. The USB stack in Linux is independent of the bootloader and if both are unable to detect the USB controller then it's likely to be hardware related.
If you have different mouse or keyboard you could try swapping them to see if that makes any difference.

ramoas86, 2021-05-24:
Thank you for the explanation. I'm using the official Raspberry Pi power supply. I tested it with another keyboard and mouse, the result was the same. I'm coming to the same conclusion as you, hardware problem. The device appears to function normally after a few minutes of power.

timg236, 2021-05-24:
> Thank you for the explanation. I'm using the official Raspberry Pi power supply. I tested it with another keyboard and mouse, the result was the same. I'm coming to the same conclusion as you, hardware problem. The device appears to function normally after a few minutes of power.

Another thing to try is adding `over_voltage=2` to config.txt. This slightly boosts the voltage supply to CPU and PCIe, it will run slightly warmer but is safe.

ramoas86, 2021-05-24:
I followed your suggestion, but the problem persists.

timg236, 2021-07-29:
Closing. Hardware issue.

closed 2021-07-29 by timg236

navhawk80, 2022-04-18:
Hello. Just intresting what kind of hardware issue it was. Did you solve it or just replace rpi?

dawiepoolman, 2022-04-27:
ditto, same issue with me

mustafayuce33, 2022-09-19:
I have the same issue. what kind of hardware problem it is? Why did you close this ticket?