raspberrypi/rpi-eeprom #341: CM4 NVME Boot issue(s).

Johnnie390, 2021-06-05:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**

    Raspberry Pi model: Compute Module 4 Lite Rev 1 (8GB RAM, 8GB Non-eMMC, WiFi)
    Board revision: d03140
    Operating system version: 64-bit raspberrypi **5.10.39-v8+**
 
Firmware:

**CM4_ENABLE_RPI_EEPROM_UPDATE=1 rpi-eeprom-update**
BOOTLOADER: up to date
CURRENT: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
LATEST: Thu 29 Apr 16:11:25 UTC 2021 (1619712685)
RELEASE: stable (/lib/firmware/raspberrypi/bootloader/stable)
    Details of any hardware attached e.g. links to USB: CM4 is installed on Oratek Tofu IO Board.
    Photo: Boot screen

**Describe the bug**
Error on booting error text - **unknown boot mode **0x6** trying SD**

**Expected behaviour**
CM4 boot from NVME SSD

Please also see https://www.raspberrypi.org/forums/viewtopic.php?f=98&t=313228&p=1873879&hilit=cm4#p1873879

Any thoughts/ideas/pointers most welcome.

Regards,

Ry

Johnnie390, 2021-06-05:
Further information - 

root@raspberrypi:~# rpi-eeprom-config
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Try SD- > USB PCIe MSD -> USB 2.0 BCM XHCI -> Network -> Loop
BOOT_ORDER=**0xf25416**

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1

timg236, 2021-06-07:
Error on booting error text - unknown boot mode 0x6 trying SD

You need to use the beta release - see NVME documentation https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/nvme.md

Johnnie390, 2021-06-07:
Morning Tim, have used the EEPROM Beta release yesterday. I can now boot using a NVME SSD. Thank you. Please close.

timg236, 2021-06-07:
Thanks @Johnnie390 for confirming

closed 2021-06-07 by timg236