raspberrypi/rpi-eeprom #344: pwr led in low power mode

seamusdemora, 2021-06-11:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
* Raspberry Pi model: RPi 4B
* Board revision (cat /proc/cpuinfo | grep Revision): `Revision : b03111`
* Operating system version .: ` Raspbian GNU/Linux 10 (buster)`
* Details of any hardware attached e.g. links to USB:  none while testing this 
* Photo of the HDMI diagnostics screen, UART trace.:  N/A - running "Lite"/Headless

**Describe the bug**
* RPi 4B configured for "low power mode"; WAKE_ON_GPIO=0, POWER_OFF_ON_HALT=1 
* `vcgencmd bootloader_version` = `Mar 18 2021 08:54:11` 
* RPi draws ~ 40mA following a `sudo halt` 
* All is well, except that the red `pwr-led` remains lit at full brightness in low power mode. 

**To Reproduce**
Steps to reproduce the behavior: 
* Configure for low power mode iaw documentation, then enter low power mode via `sudo halt`.

**Expected behaviour**
A clear and concise description of what you expected to happen: 
* The red pwr-led should change behavior (turned off or reduced brightness or ...) to provide a *visual indication* that the RPi is in low power mode. As it is, it cannot be determined whether the RPi is in operating mode or low power mode except by attempting a network connection (not definitive), or studying the activity-led at close range to watch for activity. In addition, since the objective of the low power mode is presumably to achieve the `lowest possible power state for halt`, providing an option to extinguish/reduce/blink the pwr-led will further that objective.

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration vcgencmd bootloader_config and version (vcgencmd bootloader_version)
```
$ vcgencmd bootloader_config
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41 

$ vcgencmd bootloader_version
Mar 18 2021 08:54:11
version 1b43d5b6fe5b71c300563afc0548122752a0618b (release)
timestamp 1616057651
update-time 1616057651
capabilities 0x0000001f
```

**SD card boot (please complete the following information):**
 - SD card type: **32GB SanDisk**
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.

**Network boot (please complete the following information):**
Network boot bug normally require one or more of the following log types. [PiServer](https://github.com/raspberrypi/piserver) is the officially supported network boot server.

 - DHCP server configuration files e.g. dnsmasq.conf
 - Wireshark binary packet capture
 - UART logs 

**N/A - boot from SD card** 

**USB boot (please complete the following information):**
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'

**Additional context**
Add any other context about the problem here. 

**Things I've tried that didn't have any effect on the pwr-led brightness or state:**
1.  writing to `sysfs` prior to `sudo halt` to enter low power mode: `echo none | sudo tee /sys/class/leds/led1/trigger` 
2. using the `dtparam=pwr_led_activelow=on` in `/boot/config.txt`

The [Bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

timg236, 2021-06-12:
"All is well, except that the red pwr-led remains lit at full brightness in low power mode."

That is the required behavior to indicate that the power is ok. There is no need to add additional configuration complexity to turn this on for developer debug. You could just measure 3V3 to check.

seamusdemora, 2021-06-12:
@timg236 :  "There is no need..."  Respectfully disagree - that is an opinion, not a statement of fact. And as I explained, it has nothing to do with "developer debug".

lurch, 2021-06-13:
In "halt mode" everything is stopped - the only way to display "reduced brightness" or "blinking" for the power-led would be to have the VPU and/or CPU running, and that would increase power consumption...

seamusdemora, 2021-06-14:
@lurch : That makes sense - the only other way to do it would be in hardware - which didn't seem likely. I'd actually be quite happy with `OFF` - I mentioned brightness and blinking as I didn't want to leave out any possibilities. I suppose driving a bright red LED in low power mode might be useful for some, but it makes little sense to me. But I do get that it's a challenge & I can imagine you've plenty of those :)

timg236, 2021-06-14:
The power LED is required to be on to indicate that 5V is on and we aren't going to add extra code to configure the behavior during HALT. You can connect an LED to 3V3 to check whether it's in the power-off state because 3V3 is an output of the PMIC which will be off when POWER_OFF_ON_HALT is activated.

closed 2021-06-14 by timg236

seamusdemora, 2021-06-15:
@timg236 : But surely you can see that having a red LED that glows **both** during low power mode (a non-operational state) AND when the RPi is in a fully operational state is not the ideal situation. Yes - I could add yet another LED to be extinguished when the 3V3 bus is powered off, but that doesn't extinguish the red power led. 

Would you consider **outlining** what needs to be done to make this happen? You provide the outline, someone else writes the code?

timg236, 2021-06-15:
Really. no

lurch, 2021-06-15:
@seamusdemora If you're that concerned about the red LED still glowing in halt mode, you could just cover it with electrical tape?

The code to control the LEDs is all buried inside the closed-source firmware, so isn't modifiable by anyone outside of the RPi engineering team, all of whom have far more important things to be working on.