raspberrypi/rpi-eeprom #347: Failing reboot when using 'pci=pcie_bus_safe' parameter

KrisztianKende, 2021-06-22:
The _pci=pcie_bus_safe_ is [a kernel command-line parameter](https://www.kernel.org/doc/html/v4.14/admin-guide/kernel-parameters.html), and [that's why it's interesting to us](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=314185#p1879397). You can add this to the cmdline.txt file, and then the system will boot normally. But a subsequent reboot will fail if there is no SD card and the operating system is on a USB device. [See posts on the linked topic.](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=314185#p1879716)

RPi4 Rev 1.4 8GB
Bootloader EEPROM debug output:

```
[  143.548472] reboot: Restarting system

PM_RSTS: 0x00001020
RPi: BOOTLOADER release VERSION:37f642ea DATE: Jun 11 2021 TIME: 14:39:15 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1623418752 0x289c4502 0x00d03114 0x000b8b25
PM_RSTS: 0x00001020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 11
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: USB-MSD (04) order f1
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 8
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
USBSTS 9
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
USB2[1] 400202e1 connected
USB2 root HUB port 1 init
DEV [01:00] 2.16 000000:01 class 9 VID 2109 PID 3431
HUB init [01:00] 2.16 000000:01
USB3[3] 00021203 connected enabled
USB3 root HUB port 3 init
DEV [02:00] 3.16 000000:03 class 0 VID 174c PID 55aa
MSD device [02:00] 3.16 000000:03 conf 0 iface 0 ep 81#1024 02#1024
MSD [02:00] 3.16 000000:03 register MSD
MSD [02:00] 3.16 000000:03 LUN 0
MSD INQUIRY [02:00] 3.16 000000:03
MSD [02:00] 3.16 000000:03 lun 0 block-count 0 block-size 512
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
MBR: 0x991e6141,2874212359 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
Unable to read partition as FAT
lba: 0 oem: 'A       ' volume: '            '
rsc 0 fat-sectors 0 c-count 0 c-size 0 r-dir 0 r-sec 0 r-entries 0
PM_RSTS: 0x00001020
Trying partition: 0
Unable to read partition as FAT
lba: 0 oem: 'A       ' volume: '            '
rsc 0 fat-sectors 0 c-count 0 c-size 0 r-dir 0 r-sec 0 r-entries 0
```
This _Unable to read partition as FAT_ problem is just a warm boot thing, cold boot is fine. So probably there is a leftover state somewhere around the PCIe bus that prevents rebooting.

lurch, 2021-06-22:
I've got no idea if it's the same thing or not, but the latest beta release seems to have some PCIe-related changes https://github.com/raspberrypi/rpi-eeprom/commit/cecc46f6878ff03ab87c1cdd3c17ca0b446546d2

KrisztianKende, 2021-06-22:
Yes, and no, the latest stable EEPROM image (2021-04-29) is also affected by this issue.

timg236, 2024-01-08:
This seems to be an issue with the VLI ROM and is not fixable.

closed 2024-01-08 by timg236