raspberrypi/rpi-eeprom #349: PI 4B can not enter the system normally and SD card is invalid to write EEPROM

jlhsr, 2021-06-27:
PI 4B can't be started, the specific performance is: the red light is always on, the green light is not on, but gipo can still supply power to the fan normally
I have tried to refresh boot and EEPROM through SD card, but raspberry pie can't normally enter the refresh process just like it can't recognize SD card [normally, the green light should flash until EEPROM is finished]??

pelwell, 2021-06-27:
If you power on the Pi with an HDMI monitor attached and no SD card, what do you see on the LEDs (I think I know this) and the monitor?

jlhsr, 2021-06-27:
The LED will display in red, and the green display will display no signal

pelwell, 2021-06-27:
Has the Pi ever worked? I think it is faulty and should be returned to your supplier.

jlhsr, 2021-06-27:
PI was working a few hours ago, but I accidentally cut off the power and it happened

seamusdemora, 2021-06-27:
@jlhsr :
> PI was working a few hours ago, but I accidentally cut off the power and it happened

Possible that the power cut off corrupted the SD card. Perhaps try to re-flash SD card & try again.

jlhsr, 2021-06-27:
I tried several SD cards, but PI couldn't recognize them. Is it possible that some hardware is damaged??

jlhsr, 2021-06-27:
[image]

timg236, 2021-06-27:
Unfortunately, it like a hardware issue so you'll need to contact the reseller.

* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

closed 2021-06-27 by timg236