raspberrypi/rpi-eeprom #352: rpi-eeprom-update wants to update VL805 EEPROM even on CM4

andrum99, 2021-07-08:
See https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=315293&p=1886835#p1886231

lurch, 2021-07-08:
I've just read that thread, and (I may not be understanding it fully, but) perhaps the "right" course of action here is simply for the rpi-eeprom-update tool to **not** display the VL805 version when being run on a CM4?

andrum99, 2021-07-08:
> I've just read that thread, and (I may not be understanding it fully, but) perhaps the "right" course of action here is simply for the rpi-eeprom-update tool to **not** display the VL805 version when being run on a CM4?

Yes, that would seem to be the solution. I've not read the code - I was just flagging this up. (My last attempt at understanding the code lead to me reaching the wrong conclusion, hence why I've just pointed straight to the discussion).

lurch, 2021-07-08:
> I've not read the code

IIRC the rpi-eeprom-update code is very "flexible" :wink: so that it works on as many different distros as possible.

andrum99, 2021-07-08:
> > I've not read the code
> 
> IIRC the rpi-eeprom-update code is very "flexible" 😉 so that it works on as many different distros as possible.

Indeed, but

andrum99, 2021-07-08:
Sorry - pressed the wrong button.

timg236, 2021-07-09:
To be honest. I don't think it's worth adding script complexity here so probably NGTF

lurch, 2021-07-10:
> A CM4 could have a VLI805 without an SPI flash and use embedded bootloader image to act like a Pi4B 1.4

Is that a supported / recommended way of doing things @dp111 ?

andrum99, 2021-07-10:
I think the issue with rpi-eeprom-update attempting to discover the firmware revision of the VL805 on CM4 is that it sets the expectation that this is something that should work on the CM4, which is not the case.

> (*) A CM4 could have a VLI805 without an SPI flash and use embedded bootloader image to act like a Pi4B 1.4

If you're saying that a third party could add a VL805 via a carrier board, then I don't see how that's relevant.

kbingham, 2021-09-15:
> (*) A CM4 could have a VLI805 without an SPI flash and use embedded bootloader image to act like a Pi4B 1.4

Hi Tim,

This is exactly what we're trying to achieve, but doesn't seem to work 'out of the box'.

We've added

```
# Request the VC core to load the VL805 firwmare directly
VL805=1
```
to our eeprom conf (that was read back from rpi-eeprom-config on our CM4 target), but we still don't appear to have a firmware loaded.

Is there any way to verify that the CM4 bootloader is also loading the VL805 fw, or any further documentation on how to enable this process?
It's not clear to me /where/ the boot process reads the vl805 firmware from to load it into the VL805 during boot.

peterharperuk, 2021-09-15:
Have your CM4 bootloader? Support for "VL805=1" was only added fairly "recently" I think.

kbingham, 2021-09-15:
> Have your CM4 bootloader? Support for "VL805=1" was only added fairly "recently" I think.

We're using the latest available commit from 
https://github.com/raspberrypi/usbboot/commit/8bcd5ce6f6a08c7f44222385228d04b40c036655
```
pi@raspberrypi:~ $ sudo vcgencmd bootloader_version
Jul  6 2021 11:44:53
version c258ef8fe1d2334a750078b17dab5e2c1a1787fc (release)
timestamp 1625568293
update-time 1631712166
capabilities 0x0000007f
```
(from the recovery/pieeprom.bin origin)

I see the pi400 has 
```
pi@raspberrypi:~/usbboot $ sudo vcgencmd bootloader_version
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
```

So I suspect the files I have from usbboot are outdated.
Perhaps I should move the conversation to an issue over there...

peterharperuk, 2021-09-15:
Yes that looks like the latest. It looks like "VL805=1" only does anything when doing an "USB MSD" boot if that's any help. Unfortunately I can't see any logging that can be used to confirm it's working.

kbingham, 2021-09-15:
Covering my head in shame, I can see I was referencing a "2020" binary as if it was "2021" ... Sorry ;-( Certainly not the latest ;-)

timg236, 2021-09-24:
> @kbingham Looks like your use-case is on the way to being fixed  #366

That will require updated firmware, kernel and probably need to check some sort of official reference design. Although, it did work for me with a PCIe VL805 card with a nobbled EEPROM

kbingham, 2021-09-24:
> > @kbingham Looks like your use-case is on the way to being fixed  #366
> 
> That will require updated firmware, kernel and probably need to check some sort of official reference design. Although, it did work for me with a PCIe VL805 card with a nobbled EEPROM

Thanks, we have also taken a PCIe VL805 and removed the eeprom to be able to validate this as closely as possible on a CM4 IO board.

Do you have a specific reference for which kernel version (/commit) is required to support this?

timg236, 2021-09-24:
> Thanks, we have also taken a PCIe VL805 and removed the eeprom to be able to validate this as closely as possible on a CM4 IO board.
>
> Do you have a specific reference for which kernel version [...]

rpi-update latest 5.10 is fine for the kernel. You just need to have the pcie0 XHCI reset controller defined in the CM4 DTB. That was added a while ago.

The bootloader is the latest beta release

The firmware isn't released yet but will be in the next rpi-update released now that it's on our internal release branch

timg236, 2021-09-30:
The firmware fix should now be available via rpi-update

kbingham, 2021-09-30:
Hi @timg236,

Thank you, we were able to validate this earlier today and see and use the VL805 connected to a CM4 with the EEPROM disabled on the VL805.

Next step is to figure out why it doesn't work through a PCI bridge, but that's way off topic for this issue #352

timg236, 2021-09-30:
Glad to hear that it’s able to find the VL805. Btw there’s no support for PCIe bridges for boot and VL805 firmware loading has never been tested with a bridge. Although so long as it can map config space from the device address it should just work.