raspberrypi/rpi-eeprom #354: SD Card killer?

ovhx, 2021-07-29:
This repository tracks bugs for the Raspberry Pi 4 bootloader EEPROM and Linux update scripts.

* If you suspect a hardware problem then please read the [Boot Problems](https://www.raspberrypi.org/forums/viewtopic.php?p=437084) post first before contacting the reseller.
* Support questions or should be posted on the Raspberry Pi [General Discussion](https://www.raspberrypi.org/forums/viewforum.php?f=63)**

**Mandatory information**
Raspbeery Pi 4 Model B - 4GB
* Board revision (cat /proc/cpuinfo | grep Revision) 
Ubuntu preinstalled server 21.04
A wireless usb keyboard
* Photo of the HDMI diagnostics screen, UART trace.

**Describe the bug**
after installing the bootloader it killed 2 sd cards. One was a 200 gb kingston sd card and the other a 512gb kingston sd card which both popped up the same error code "Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)
Failed to open device: 'sdcard' (cmd 371a0010 status 1fff0001)" after rebooting the system. Ive tried flashing a recovery image but it had the same error code after. Both cards were undetectable on my pc after even with special recovery software.

**To Reproduce**
Steps to reproduce the behavior: Install the bootloader then reboot

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Bootloader version and configuration**
If you have modified the default bootloader release or configuration then please attach the bootloader configuration vcgencmd bootloader_config and version (vcgencmd bootloader_version) Stock bootloader

**SD card boot (please complete the following information):**
 - SD card type 512GB kingston canvas Go! Plus V30
 - Partition information (sudo fdisk -l) if you are able to obtain this from another computer.

**Network boot (please complete the following information):**
Network boot bug normally require one or more of the following log types. [PiServer](https://github.com/raspberrypi/piserver) is the officially supported network boot server.

 - DHCP server configuration files e.g. dnsmasq.conf
 - Wireshark binary packet capture
 - UART logs

**USB boot (please complete the following information):**
Verify that the the USB device works correctly when hot-plugged under Linux and attache the output of 'lsusb -vvv'
Ive plugged in other cards that do not have ubuntu and they both worked correctly after reboots

**Additional context**
Add any other context about the problem here. 

The [Bootloader configuration](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) page describes how to enable UART or NETCONSOLE logs. For complex USB boot issues NETCONSOLE logs are recommended.

timg236, 2021-07-29:
If the bootloader has printed that message then it hasn't detected the card let alone written to it. Therefore, it's not going to be a software issue.

ovhx, 2021-07-29:
> If the bootloader has printed that message then it hasn't detected the card let alone written to it. Therefore, it's not going to be a software issue.

How ? It booted perfectly first try and even updated but after doing sudo reboot it popped up.

timg236, 2021-07-29:
It sounds like it failed before the reboot

lurch, 2021-07-29:
> Both cards were undetectable on my pc after even with special recovery software.

Are you sure they were genuine SD cards, and not fakes?
Windows sometimes has problems "seeing" badly-formatted SD-cards, can you try accessing them from a Linux computer?
See also https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives

ovhx, 2021-07-29:
> Are you sure they were genuine SD cards, and not fakes?
> Windows sometimes has problems "seeing" badly-formatted SD-cards, can you try accessing them from a Linux computer?
> See also [...]

The card is genuine yes but I'm worried that it's something about the bootloader itself. I've had the 1st card for 2 years straight running nonstop and then it got killed a day after the bootloader was installed. I'll try seeing if I can recover the sd card.

pelwell, 2021-07-29:
> 2 years straight running nonstop

If this is really the first time in 2 years that the Pi has had a chance to cool down then it is possible that this is the first opportunity it's had to notice a minor hardware problem - a joint that has gone "dry", perhaps. I'm not saying that you've done anything wrong or should have let it rest, but occasionally faults like this do develop - remember the X-Box "red ring of death" and the "fix" of wrapping it in a blanket to raise its temperature and reflow its own solder?

ovhx, 2021-07-29:
> If this is really the first time in 2 years that the Pi has had a chance to cool down then it is possible that this is the first opportunity it's had to notice a minor hardware problem - a joint [...]

yes i remember that so are you suggesting that the solder has gone dry ? If so ill take the cooler off and ill check if so

pelwell, 2021-07-29:
Yes, I'm suggesting that perhaps a solder joint was a bit marginal, and that over time it has become worse but wasn't detected because the device was kept on. However, I am clutching at straws...

pelwell, 2021-07-29:
One thing I'm not clear on - were these cards in two different Pi 4s which were both updated and then both failed? Can you list the steps involved?

pelwell, 2021-07-29:
And can you get the board revision numbers of the Pi(s)? The EEPROM will show that if booted without a card.

ovhx, 2021-07-29:
> One thing I'm not clear on - were these cards in two different Pi 4s which were both updated and then both failed? Can you list the steps involved?

all were on the same pi and the problem popped up after i installed the bootloader

timg236, 2021-07-29:
It's not really possible for the bootloader to kill a card. It just does read commands. We do 10,000s of reboot tests with a variety SD cards so we'd see a generic software issue

It's more likely be that the SD card interface has failed (mechanical or physical) and more cards will fail. It's hard to debug without using measuring the voltage of the SD card pins. 

As @pelwell mentioned a screenshot of the HDMI diagnostics screen showing the board information and software versions would be useful to track this issue.

ovhx, 2021-07-29:
> It's not really possible for the bootloader to kill a card. It just does read commands. We do 10,000s of reboot tests with a variety SD cards so we'd see a generic software issue
>
> It's more likely [...]

Okay I will. I also booted on a live linux system and the card doesn't even show up

ovhx, 2021-07-29:
> It's not really possible for the bootloader to kill a card. It just does read commands. We do 10,000s of reboot tests with a variety SD cards so we'd see a generic software issue
>
> It's more likely [...]

[image]

timg236, 2021-10-05:
This doesn't appear to be an issue with the bootloader so closing.

closed 2021-10-05 by timg236