raspberrypi/rpi-eeprom #355: Hangs after 1 hour on WD 18Tb

axlslak, 2021-08-04:
Hi. Am using several raspberry pi 4 (4Gb ram). Each of them has a pair of hard-drives attached, running a raid1 array. The usb2sata adapter I am using is german made gembird docking station, for the express purpose that it has a power adapter that feeds both 12V and 5V with enough power not to drain the pi of its own electricity. In fact I checked with one of those usb volt/amp meters and it seems the sata controller drains 0 power from usb. Just to be clear. Also according to manuals and stuff, the adapter feeds 2 amps, and disks only need about 0.8 amps.

I have 3 of these machines. One has a pair of WD reds of 4Tb. No issues whatsoever. One has a pair of WD reds 10Tb. No issues whatsoever. One has a pair of WD reds 18Tb. This one hangs after one hour and 3 minutes. more or less. I haven't managed to measure the exact time yet, but its consistent around one hour and some minutes. Never 20 minutes, never 2 hours. 1 hour and some minutes. Consistently. 

Other details.

I am using a custom version of gentoo. Same on all 3 machines. Identical in fact. 2 work, third, not so much. Same kernel, same userspace. Can't really blame the distro. Tried to switch that pair of disks to another pi, same issue. In fact I tried other sata controller, other usb cable, other power supply. other pi. other pi power supply. No matter what i throw at it, seems only those 2 discs hang the pi after an hour.

Discs aren't partitioned. Raid is made from complete block device /dev/sda and /dev/sdb. Therefor there is no boot. Using dhcp&tftp for that. again, all 3 have the same deal. Doesn't seem to be a kernel issue. More like a hardware firmware issue, I feel. On the pi, when it hangs, kernel cursor is still blinking. kernel does nothing, but it still blinks. i tried to connect / disconnect usb devices to see if kernel says something... but it doesn't. on the network that host seems down. no arp, no reply to ping, no activity on network, but leds do blink. Keyboard doesn't do anything. (maybe I should try one with cable to see if i can switch light on caps/nums). 

I have a feeling the huge capacity might be the issue. Something not thought of at the time as being possible. There was a fix a few months ago for hdds of 5Tb capacity. So I guess its not unreasonable to ask if 18Tb might be an issue. Also as I type this, I realize I didn't try to use a mmc to put rootfs on and simply trying to connect the hard-drives later, see if it still crashes. Fact the i use that raid1 as rootfs, might be the issue. The drives are ok though. In a pc at least. 

I tried to leave open the logs, top, cat /proc/mdadm, cat /proc/diskstats. pretty much all I can think of to catch last glimpse of what made the machine crash. Nothing out of the ordinary. Its like the hardware fails for some reason and kernel is just left hanging.

Any thoughts are ofc welcome. Any information I could provide... anything you need.

lurch, 2021-08-04:
If it gets past boot and only hangs after an hour of usage, I think this is an issue for https://github.com/raspberrypi/linux/issues rather than https://github.com/raspberrypi/rpi-eeprom/issues ?

> The drives are ok though. In a pc at least.

Is that with the drives directly-connected to the PC via SATA, or using the same docking station that you're using to connect the drives to the Pi?

timg236, 2021-08-05:
Since the bug report didn't provide any of the mandatory information this is auto-closed.

It sounds like it was able to boot Linux which means that this isn't a bootloader issue

closed 2021-08-05 by timg236